home-assistant/operating-system #1687: IPv6 Addr-gen-mode should be EUI64 and not stable-privacy

trunneml, 2021-12-21:
### Describe the issue you are experiencing

Home-Assistant IMHO is a server system and server system shouldn't change their IP address from time to time.

The default settings for NetworkManager under Home Assistant OS is `ipv6.addr-gen-mode=stable-privacy`. This settings helps to hide the mac-adress of the user. But it also prevents to configure the router firewall to access Home-Assistant over IPv6 from the Internet. 

To full support IPv6 remote access from the Internet a stable IPv6 address is needed, that doesn't change unpredictable. 

I updated the Network-Manager configuration with `ha login` and changed `addr-gen-mode` to `eui64`. With this change the public IPv6 address uses the same suffix as the link local and it's possible to configure the router firewall.

You can validate this running `ip addr` on `ha login`. EUI64 address have `valid_lft forever`, but `stabe-privacy` only has a only 7067sec lifetime .

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Setup IPv6
2. Check the Suffix of the registered IP-Addresses

They will differ between public and link-local addresses and their lifetime will be limited.

### Anything in the Supervisor logs that might be useful for us?

```txt
Not needed
```

### Anything in the Host logs that might be useful for us?

```txt
Not needed
```

agners, 2021-12-23:
> Home-Assistant IMHO is a server system and server system shouldn't change their IP address from time to time.

Agreed.

> The default settings for NetworkManager under Home Assistant OS is ipv6.addr-gen-mode=stable-privacy. This settings helps to hide the mac-adress of the user. 

According to [NetworkManager documentation](https://developer-old.gnome.org/NetworkManager/stable/settings-ipv6.html) this should enable [RFC7217](https://datatracker.ietf.org/doc/html/rfc7217) SLAAC, which as per abstract:

> This document specifies a method for generating IPv6 Interface
   Identifiers to be used with IPv6 Stateless Address Autoconfiguration
   (SLAAC), such that an IPv6 address configured using this method is
   stable within each subnet, but the corresponding Interface Identifier
   changes when the host moves from one network to another. 

So as long as you don't move your server into a different subnet, the address should not change.

> But it also prevents to configure the router firewall to access Home-Assistant over IPv6 from the Internet.

It shouldn't as the address should stay the same as long as you stay in your subnet.

> You can validate this in your router. Mine shows 10 different IPv6 addresses after some month.

Could it be that your router got a new IPv6 network address assigned? Or is the network part in all those 10 addresses the same?

trunneml, 2021-12-26:
There are two problems with stable-privacy.

1. As far as I can tell, it uses the internal UUID of the NetworkManager config to calculate the suffix. If I change the network settings in Home-Assistant this UUID changes and then I'm getting a new suffix which results in a new IPv6 address. And then I have to reconfigure my router. This also could happen when Home-Assistent internally change the settings file.
2. In home routers often you have to set the Interface ID (MAC-Address) to open the firewall.
[image]
The problem here is, that static-privacy generates a different suffix for local, link-local, an public IPv6 adresses. AVM automatically uses the suffix of the link local IP address to open the firewall. You can set the right suffix manually but it makes things more complicated and error prone. Especially when the suffix is changed because of first problem.

I don't like to push my mac address into public, but Home-Assistant I not a Laptop where a user is browsing in the internet. It is a machine that may be connects to some cloud services with clear credentials. So IMHO there is no privacy Issue with EUI64, but there are some issues with SLAAC and home routers.

agners, 2021-12-27:
> 1. As far as I can tell, it uses the internal UUID of the NetworkManager config to calculate the suffix. If I change the network settings in Home-Assistant this UUID changes and then I'm getting a [...]

Hm, is that only the first time editing settings or every time? I was under the assumption we reuse the connection setting. If not, we should probably fix this.

> 2. In home routers often you have to set the Interface ID (MAC-Address) to open the firewall.
>   [image]
>   The problem here is, that static-privacy generates a different suffix for local, link-local, [...]

Ugh, ok unfortunate. They probably should not rely on that. Privacy enabled IPv6 address have different suffix depending on network by design.

> I don't like to push my mac address into public, but Home-Assistant I not a Laptop where a user is browsing in the internet. It is a machine that may be connects to some cloud services with clear [...]

Personally, I think I wouldn't mind either in this case. However, we are a privacy focused project, so I'd rather prefer to left it enabled if possible. It seems that it can also be used to find attack vectors (see https://en.wikipedia.org/wiki/IPv6_address#Stable_privacy_addresses).

trunneml, 2021-12-27:
> Hm, is that only the first time editing settings or every time? I was under the assumption we reuse the connection setting. If not, we should probably fix this.

As far as I can tell it changes every time I edit the settings. Yes the important thing is that the IPv6 is stable. Best thing would be that it is also stable after an reinstallation from a backup.

> Ugh, ok unfortunate. They probably should not rely on that. Privacy enabled IPv6 address have different suffix depending on network by design.

Yes, and it not a no-name router. :(

> Personally, I think I wouldn't mind either in this case. However, we are a privacy focused project, so I'd rather prefer to left it enabled if possible. It seems that it can also be used to find [...]

The attack vector is a good point.

agners, 2021-12-28:
I did some testing on my end: Initially, OS uses the `default` connection profile with pre-determined UUID. When changing the network settings through Supervisor the first time, the Supervisor creates a new NetworkManager profile (`/etc/NetworkManager/system-connections/Supervisor\ eth0.nmconnection` in my case). The connection profile gets a new UUID, which leads to a change in the IPv6 address. After that, the Supervisor seems to keep reusing that profile with the same UUID (using `supervisor-2021.12.2`). Changing settings and/or rebooting did not change the IPv6 address for me.

We can't ship with a pre-defined connection profile which is picked up by Supervisor since the network interface name is device/environment dependent. Maybe the Supervisor should create a connection profile on first start. Doesn't seem ideal though.

trunneml, 2022-01-07:
Create a connection profile on start, sounds like a good idea. Because then the ip would change the first time the user saves it network settings.

What's about an option to disable stable-privacy and use eui64, that will also fix the broken ipv6 implementation of some routers?

agners, 2022-03-31:
I found one more wrinkle to the `ipv6.addr-gen-mode=stable-privacy` setting: It requires stable machine-id as well as a file `/var/lib/NetworkManager/secret_key` which contains a generated key (see [this NetworkManager documentation](https://developer-old.gnome.org/NetworkManager/stable/NetworkManager.html) about that file). Now I noticed in my virtual machine that this file was missing. It seems that due to lack of entropy at early boot NetworkManager was not able to generate the file. From the NetworkManager.service logs:

```
Mar 31 17:29:45 homeassistant NetworkManager[337]: <warn>  [1648747785.0433] secret-key: failure to generate good random data for secret-key (use non-persistent key)
```

This message appeared after every boot. Unfortunately, it seems that NetworkManager also does not retry to generate that file. This leads to the unfortunate situation that the system never creates that file. Due to lack of the file, NetworkManager always generates a new IPv6 address when using SLAAC. Restarting network manager manually at a later time after boot (`systemctl restart NetworkManager`) seems to generate the file, from which point on the SLAAC address got stable.

@trunneml can you check if the above file exists in your installation/if NetworkManager warns about secret_key? Also make sure that the machine ID file indeed stays static.

In my case, using the ova image on libvirt/Virtual Machine Manager, it was simply a matter of adding `RNG` from the hardware list, or in raw libvirt xml

```
<rng model="virtio">
  <backend model="random">/dev/urandom</backend>
</rng>
```
> Create a connection profile on start, sounds like a good idea. Because then the ip would change the first time the user saves it network settings.

With the changes https://github.com/home-assistant/operating-system/pull/1813 and https://github.com/home-assistant/supervisor/pull/3528 that will be the case: An ephemeral "Wired connection 1" connection will be generated, and on first start of Supervisor the connection (with the same UUID) will be stored. Even though the "Wired connection 1" is ephemeral, it seems that the UUID is stable still (I assume NetworkManager users a stable hash). In conclusion: The generated IPv6 address will stay the same, after Supervisor stores the network configuration or even when rebooting before Supervisor gets started.

closed 2022-07-06 by github-actions[bot]

dermoth, 2025-01-31:
I don't think this has been implemented, but it definitely shouldn't be. Stable-privacy creates unique addresses at regular intervals and deprecates old ones. Communications **from the host** use these IP addresses and since they're ephemeral they won't be usable later to try gaining access to the device. Default deny firewall policy should also block incoming requests.

What you need for permanent access (internal or external) is to configure an IPv6 suffix in your DNS server's static leases. That suffix is used for two things:

  * As a suffix to the internal ULA address, giving you _a static address for internal addressing_
  * As a suffix to the public prefix from your router, which is a _permanent IPv6 address for inbound connections_

A negative mask firewall entry can then be used to allow forwarding external traffic to the device. All that is needed when the dynamic IPv4 address or IPv6 prefix changes is to automatically update the A / AAAA records of your external domain to point at the new addresses, and the device can be accessed directly behind a router/firewall using both IPv4 and IPv6. That IP address won't be visible to external client your device connects **to** so it enhances privacy and makes it more difficult to look up externally accessible services when using IPv6.

dermoth, 2025-01-31:
Actually I was referring to IPv6 Privacy Extensions , apologies for my confusion. It looks like network-manager don't use them even when enabled if a static suffix is returned by the DHCP server, I'm trying to see if there's a way to force it. dhclient OTOH has no issues, this is what I'm using for my "appliances" that need external connectivity.