home-assistant/operating-system #1689: Black screen after boot up, graphics driver crash

bjschuit, 2021-12-23:
### Describe the issue you are experiencing

I've migrated HASS OS from a VM to a new dedicated computer (fresh install with a restore of a backup).
All works fine, except for the following:
On the VM in Virtual Box, I always had a terminal at my disposal.
However, on the new computer, the screen becomes blank after starting up Home Assistant. The computer first shows the screen and I can see Linux loading, but at some point (it goes to fast to see what is the last step) the screen becomes blank. At that point, Home Assistant web GUI is available. But I'd like to have the terminal on the pc itself visible as well of course. 

It is a Intel Celeron 5905 on a Gigabyte H410M motherboard. In the host logs I see something with VGA that seems to be crashing. 
Secure boot is off, UEFI is on. 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Fresh install on a new PC. 

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
[    4.256818] ------------[ cut here ]------------
[    4.257300] i915 0000:00:02.0: drm_WARN_ON(!IS_PLATFORM(dev_priv, INTEL_TIGERLAKE) && !IS_PLATFORM(dev_priv, INTEL_ROCKETLAKE))
[    4.257347] WARNING: CPU: 1 PID: 137 at drivers/gpu/drm/i915/intel_pch.c:123 intel_pch_type+0x1e7/0x9b0 [i915]
[    4.258793] Modules linked in: x86_pkg_temp_thermal i915(+)
[    4.259291] CPU: 1 PID: 137 Comm: systemd-udevd Not tainted 5.10.83 #1
[    4.259793] Hardware name: Gigabyte Technology Co., Ltd. H410M S2H V3/H410M S2H V3, BIOS F1 05/24/2021
[    4.260331] RIP: 0010:intel_pch_type+0x1e7/0x9b0 [i915]
[    4.260832] Code: 8b 67 50 4d 85 e4 0f 84 d8 07 00 00 e8 12 6c ac df 48 c7 c1 98 13 46 c0 4c 89 e2 48 c7 c7 e6 4c 48 c0 48 89 c6 e8 5a 5f f7 df <0f> 0b b8 09 00 00 00 e9 cc fe ff ff 66 81 fe 00 1c 0f 84 b2 04 00
[    4.261884] RSP: 0018:ffff9e878022fa18 EFLAGS: 00010286
[    4.262408] RAX: 0000000000000000 RBX: ffff8d8746af8000 RCX: 0000000000000027
[    4.262942] RDX: 0000000000000027 RSI: 00000000ffffdfff RDI: ffff8d88f651b8e8
[    4.263469] RBP: ffff8d8740b0a800 R08: ffff8d88f651b8e0 R09: ffff9e878022f848
[    4.264001] R10: 0000000000000001 R11: 0000000000000001 R12: ffff8d8740a7d960
[    4.264527] R13: 0000000000004380 R14: ffff8d8746af8000 R15: 0000000000000000
[    4.265046] FS:  00007f2480f1ce40(0000) GS:ffff8d88f6500000(0000) knlGS:0000000000000000
[    4.265575] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[    4.266094] CR2: 00007f2480ed338d CR3: 00000001050ae003 CR4: 00000000003706e0
[    4.266094] Call Trace:
[    4.266130]  intel_detect_pch+0x56/0x2f0 [i915]
[    4.267673]  i915_driver_probe+0x2c3/0xc20 [i915]
[    4.267677]  ? __kernfs_new_node+0x15b/0x1c0
[    4.268696]  ? __kernfs_new_node+0x15b/0x1c0
[    4.268721]  i915_pci_probe+0x3f/0x120 [i915]
[    4.268724]  local_pci_probe+0x3d/0x70
[    4.270244]  ? pci_assign_irq+0x22/0xd0
[    4.270246]  pci_device_probe+0xd0/0x150
[    4.271247]  really_probe+0x1eb/0x320
[    4.271248]  driver_probe_device+0x4a/0xa0
[    4.271250]  device_driver_attach+0x4e/0x60
[    4.271251]  __driver_attach+0x47/0xb0
[    4.271252]  ? device_driver_attach+0x60/0x60
[    4.271253]  bus_for_each_dev+0x73/0xb0
[    4.271254]  bus_add_driver+0x172/0x1c0
[    4.271255]  driver_register+0x67/0xb0
[    4.271256]  ? 0xffffffffc0531000
[... 299 lines cut ...]
[   40.299555] eth0: renamed from veth5f01821
[   40.304498] IPv6: ADDRCONF(NETDEV_CHANGE): veth0b51a7e: link becomes ready
[   41.898450] hassio: port 6(veth89e04cd) entered blocking state
[   41.898451] hassio: port 6(veth89e04cd) entered disabled state
[   41.898478] device veth89e04cd entered promiscuous mode
[   41.898496] kauditd_printk_skb: 111 callbacks suppressed
[   41.898497] audit: type=1700 audit(1640252154.373:125): dev=veth89e04cd prom=256 old_prom=0 auid=4294967295 uid=0 gid=0 ses=4294967295
[   42.028355] audit: type=1325 audit(1640252154.503:126): table=nat family=2 entries=0 op=xt_register pid=2712 subj==unconfined comm="iptables"
[   42.031601] audit: type=1325 audit(1640252154.506:127): table=filter family=2 entries=0 op=xt_register pid=2713 subj==unconfined comm="iptables"
[   42.034535] audit: type=1325 audit(1640252154.509:128): table=nat family=2 entries=5 op=xt_replace pid=2716 subj==unconfined comm="iptables"
[   42.035071] audit: type=1325 audit(1640252154.509:129): table=nat family=2 entries=7 op=xt_replace pid=2717 subj==unconfined comm="iptables"
[   42.036013] audit: type=1325 audit(1640252154.510:130): table=nat family=2 entries=8 op=xt_replace pid=2719 subj==unconfined comm="iptables"
[   42.036527] audit: type=1325 audit(1640252154.511:131): table=nat family=2 entries=10 op=xt_replace pid=2720 subj==unconfined comm="iptables"
[   42.037215] audit: type=1325 audit(1640252154.512:132): table=nat family=2 entries=11 op=xt_replace pid=2721 subj==unconfined comm="iptables"
[   42.037881] audit: type=1325 audit(1640252154.512:133): table=nat family=2 entries=12 op=xt_replace pid=2722 subj==unconfined comm="iptables"
[   42.038577] audit: type=1325 audit(1640252154.513:134): table=nat family=2 entries=13 op=xt_replace pid=2723 subj==unconfined comm="iptables"
[   42.051865] eth0: renamed from vethf623a5d
[   42.053521] IPv6: ADDRCONF(NETDEV_CHANGE): veth89e04cd: link becomes ready
[   42.053562] hassio: port 6(veth89e04cd) entered blocking state
[   42.053565] hassio: port 6(veth89e04cd) entered forwarding state
[   42.168809] hassio: port 7(veth76b234b) entered blocking state
[   42.168810] hassio: port 7(veth76b234b) entered disabled state
[   42.168864] device veth76b234b entered promiscuous mode
etc.
```

bjschuit, 2021-12-23:
Could this be a helpful direction for the solution? https://forums.gentoo.org/viewtopic-p-8616597.html?sid=a7247da3e24d589dbc2251f0253b5a31

agners, 2021-12-23:
Hm, it seems that [the patches which fix the problem](https://gitlab.freedesktop.org/drm/intel/-/issues/2742#note_803367) did not make it into 5.10.y stable kernel :cry: 

Ideally we should ask the Intel graphics team to backport them to 5.10.y.

I plan to move to 5.15 LTS kernel which should have those patches soon. So alternatively just wait for a development release and try with that once available.

agners, 2022-03-30:
@bjschuit the latest nightly builds use Linux 5.15 LTS. Can you give this a try and see if the problem is indeed fixed?

https://os-builds.home-assistant.io/8.0.dev20220328/

bjschuit, 2022-03-30:
Super, thanks for the effort! Can I live upgrade to this version? If so, can you perhaps point me in the direction how I can do this (running Generic x86-64 native directly on hardware). Or do I need to reinstall everything and restore a backup? For the latter I unfortunately lack the time at the moment, that I can do in a couple of weeks probably.

agners, 2022-04-05:
Whoops, sorry missed your message. You can upgrade to that version, use the following commands:

```
ha su options --channel=dev
ha su reload
ha os update
```

You might want to go back to stable channel afterwards. You can safely stay on the dev version of OS until the stable version comes out.

```
ha su options --channel=stable
```

bjschuit, 2022-04-07:
I can confirm the dev release with new Linux kernel 5.15 fixed the issue. No more errors in the logs and the expected terminal screen output is there. Thank you!

closed 2022-04-07 by bjschuit