home-assistant/operating-system #1690: Sound does not work on Wyse 3040 thin client

jcam, 2021-12-23:
### Describe the issue you are experiencing

The required kernel module for audio in my Wyse 3040 is not included. It uses the rt5670 driver (https://github.com/torvalds/linux/blob/master/sound/soc/codecs/rt5670.c).
This requires CONFIG_SND_SOC_RT5670, which in turn depends on CONFIG_I2C and likely also CONFIG_SND.

It would be great to try out things like Almond and Ada, but would need audio in/out to do that.

Should I attempt to make my own build to verify this is all that is required and to make a PR, or is this a simple thing the HAOS folks can handle?

Thank you!
Apologies if this is in the wrong section, I wasn't sure if this counts as a "bug" since sound is supported for other boards, but I don't see a way to provide my own modules without going in deep.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

No reproduction steps necessary, CONFIG_SND_SOC_RT5670 is not in the kernel config.

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

agners, 2022-01-17:
Additional audio drivers are now enabled in OS 7.2 (currently in beta channel).

closed 2022-01-17 by agners

x5500, 2024-09-07:
No audio driver on Wyse 3040 thin client. 
I'm using:
  OS Version:               Home Assistant OS 13.1
  Home Assistant Core:      2024.9.0
-----

Was it removed or I have to do something unusual to force it work?

jcam, 2024-09-09:
I had it working in my custom built kernel way back when I had requested this change, but I just checked and it is not working on the release build that's on the wyse 3040 at my ex's house right now. Perhaps there was an additional kernel option that was missed, but I don't have the config that I used anymore, and I'm putting together an N100 for my new place...