home-assistant/operating-system #1691: VMWARE ESXi 5.5 ova

mocallins, 2021-12-23:
### Describe the issue you are experiencing

The downloadable ova files for VMWare ESXi will not deploy with either the web ui for the ESXi host, or the Windows vmware client app.

Looks like its related to a usb / sata configuration.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.0 , 8.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Just follow the steps to deploy a new vm
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
ova did not deploy system does not exist
```

### Anything in the Host logs that might be useful for us?

```txt
ova did not deploy system does not exist
```

### System Health information

ova did not deploy system does not exist

### Additional information

ova did not deploy system does not exist

agners, 2021-12-27:
> The downloadable ova files for VMWare ESXi will not deploy with either the web ui for the ESXi host, or the Windows vmware client app.

What happens? Any error message or similar?

> 
> Looks like its related to a usb / sata configuration.

How do you come to that conclusion?

mocallins, 2022-01-10:
[image]

This is the inital problem.
So i extracted the .ova, and then looked through the .ovf to see what line it was referring too,.
There is also another incompatiability with the sound card definition,  so i just removed that altogether.

agners, 2022-01-12:
> So i extracted the .ova, and then looked through the .ovf to see what line it was referring too,.
> There is also another incompatiability with the sound card definition, so i just removed that altogether.

Ok, I hope you have been able to get it running with this changes. Updates will not affect the ovf obviously, so you should still be able to use HAOS in your installation.

ESXi got released in 2013/09/22, and according to this [blog post](https://kb.vmware.com/s/article/51491) its EOL at this point. So I'd say its not worth adjusting for ESXi 5.5, especially if we need to compromise on features for other more modern platforms. Hence I am closing this issue.

closed 2022-01-12 by agners

mocallins, 2022-01-16:
Cool, no problem, Yea, I know 5.5 is older.