home-assistant/operating-system #1692: Impossible to update from 5.13 to 7.0 version on Home Assistant Blu

trunktrunk, 2021-12-24:
### Describe the issue you are experiencing

When I try to update from 5.13 to 7.0 version on the Home Assistant Blue just bought, I receive the message "connection lost" and then the odroid reboot whit the old version

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

5.13

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1.Click on opdate button
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Fornitore del registro
Supervisor
[16:00:13] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[16:00:13] INFO: Starting local supervisor watchdog...
21-12-24 16:00:15 INFO (MainThread) [__main__] Initializing Supervisor setup
21-12-24 16:00:15 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
21-12-24 17:00:15 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: odroid-n2
21-12-24 17:00:15 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/aarch64-hassio-supervisor with version 2021.12.2
21-12-24 17:00:15 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
21-12-24 17:00:15 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
21-12-24 17:00:15 INFO (MainThread) [__main__] Setting up Supervisor
21-12-24 17:00:15 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
21-12-24 17:00:15 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
21-12-24 17:00:15 WARNING (MainThread) [supervisor.dbus.manager] Can't load dbus interface io.hass.os: The name io.hass.os was not provided by any .service files
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
21-12-24 17:00:15 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.info] Updating local host information
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.services] Updating service information
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.network] Updating local network information
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.manager] Host information reload completed
21-12-24 17:00:15 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
21-12-24 17:00:15 WARNING (MainThread) [supervisor.host.apparmor] AppArmor is not enabled on host
21-12-24 17:00:15 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2021.09.0
21-12-24 17:00:15 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
21-12-24 17:00:16 INFO (SyncWorker_1) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/aarch64-hassio-cli with version 2021.09.0 - 172.30.32.5
21-12-24 17:00:16 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2021.06.0
21-12-24 17:00:16 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
21-12-24 17:00:17 INFO (SyncWorker_0) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2021.06.0 - 172.30.32.3
21-12-24 17:00:17 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
21-12-24 17:00:17 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2021.07.0
21-12-24 17:00:17 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
21-12-24 17:00:17 INFO (SyncWorker_1) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2021.07.0 - 172.30.32.4
21-12-24 17:00:17 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
21-12-24 17:00:17 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2021.04.0
[... 539 lines cut ...]
[   19.562751] hassio: port 4(vethc701fec) entered blocking state
[   19.562756] hassio: port 4(vethc701fec) entered disabled state
[   19.562846] device vethc701fec entered promiscuous mode
[   19.563438] hassio: port 4(vethc701fec) entered blocking state
[   19.563443] hassio: port 4(vethc701fec) entered forwarding state
[   19.913014] eth0: renamed from vethc9b446d
[   19.948667] hassio: port 4(vethc701fec) entered disabled state
[   19.949041] IPv6: ADDRCONF(NETDEV_CHANGE): vethc701fec: link becomes ready
[   19.949093] hassio: port 4(vethc701fec) entered blocking state
[   19.949096] hassio: port 4(vethc701fec) entered forwarding state
[   20.234656] hassio: port 5(vethc3d2f64) entered blocking state
[   20.234661] hassio: port 5(vethc3d2f64) entered disabled state
[   20.234759] device vethc3d2f64 entered promiscuous mode
[   20.234975] hassio: port 5(vethc3d2f64) entered blocking state
[   20.234979] hassio: port 5(vethc3d2f64) entered forwarding state
[   20.597048] eth0: renamed from veth6ed07c8
[   20.620981] IPv6: ADDRCONF(NETDEV_CHANGE): vethc3d2f64: link becomes ready
[   31.716460] USB_PWR_EN: disabling
[   31.716466] TF_IO: disabling
[   76.128029] kauditd_printk_skb: 25 callbacks suppressed
[   76.128032] audit: type=1334 audit(1640361671.728:120): prog-id=12 op=UNLOAD
[   76.128040] audit: type=1334 audit(1640361671.728:121): prog-id=11 op=UNLOAD
[   76.215763] audit: type=1334 audit(1640361671.816:122): prog-id=10 op=UNLOAD
[   76.215773] audit: type=1334 audit(1640361671.816:123): prog-id=9 op=UNLOAD
```

trunktrunk, 2021-12-24:
Solved using command in SSH

agners, 2021-12-27:
> Solved using command in SSH

Did you update to a 6.x version first?

I realized there is a bug which prevents a direct update. The logs on the host OS reveal the following errors:

```
Dec 27 10:46:59 homeassistant rauc[357]: Updating /dev/disk/by-partlabel/hassos-boot with /run/rauc/bundle/spl.img
Dec 27 10:46:59 homeassistant rauc[357]: installing /mnt/data/supervisor/tmp/hassos-7.0.raucb: Updating slot spl.0
Dec 27 10:46:59 homeassistant rauc[357]: Running custom slot install hook for spl.0
Dec 27 10:46:59 homeassistant rauc[357]: Running slot hook slot-install for spl.0
Dec 27 10:46:59 homeassistant rauc[2957]: dd: invalid argument 'direct' to 'oflag'
Dec 27 10:46:59 homeassistant rauc[2958]: dd: invalid argument 'direct' to 'oflag'
Dec 27 10:46:59 homeassistant rauc[357]: installing /mnt/data/supervisor/tmp/hassos-7.0.raucb: Updating slot spl.0 status
Dec 27 10:46:59 homeassistant rauc[357]: Marking target slot kernel.1 as bootable...
Dec 27 10:46:59 homeassistant rauc[357]: installing /mnt/data/supervisor/tmp/hassos-7.0.raucb: Updating slot spl.0 done
Dec 27 10:46:59 homeassistant rauc[357]: installing /mnt/data/supervisor/tmp/hassos-7.0.raucb: All slots updated
Dec 27 10:46:59 homeassistant rauc[357]: installing /mnt/data/supervisor/tmp/hassos-7.0.raucb: finished
Dec 27 10:46:59 homeassistant rauc[357]: installing `/mnt/data/supervisor/tmp/hassos-7.0.raucb` succeeded
```

This leads to the boot loader not being updated.

It seems this has been introduced with #1408. The update script uses `dd` from the host OS, however, 5.13 still used Busybox which did not support those flags. To make the update work across two major versions we need some special logic in that script.

agners, 2021-12-27:
This is being addressed with #1692.

agners, 2021-12-29:
Direct update from 5.13 to 7.1 and newer will be possible.

closed 2021-12-29 by agners

agners, 2022-01-16:
Unfortunately I missed backporting the fix to 7.1.

It will be fixed in 7.2 :crossed_fingers: