home-assistant/operating-system #1697: HassOS unusable on Raspberry Pi 3A+

morgansimonsen, 2021-12-28:
### Describe the issue you are experiencing

After booting HassOS on a Raspberry Pi 3A+ the system quickly becomes unresponsive and the only way to regain contact it so disconnect the power supply and reboot. Any interaction with the Supervisor, like looking at logs, installing add-ons etc. seem to make it worse. Installing add-ons does not work at all. Therefore SSH access to retrieve more data is not possible. Retrieving logs might work half of the time. In the very short time the system is responsive I can see that total memory usage is around 30 %.

Have tried multiple SD cards and both Wifi and Ethernet (using a LAN7500 USB dongle). 

The 32-bit version of HassOS works better, but does not support all desired add-ons and according to the documentation; 64-bit is recommended.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Burn fresh SD card on Windows or Linux
2. Boot RPi 3A+ with SD card
3. Create user
4. Wait some minutes for the system to gradually become unusable
5. OR interact with Supervisor to look at info or logs. This aggravates the issue and provokes the issue more rapidly.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:119
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 11:19:01 PM (4 occurrences)
Last logged: 11:29:54 PM

Failed to to call /core/stats -
Failed to to call /supervisor/stats -
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 5.10.63-v8 (builder@bad4cce73fbc) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g907739ed) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Fri Dec 3 17:50:41 UTC 2021
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 3 Model A Plus Rev 1.0
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x0000000017c00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000001bffffff]
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   DMA zone: 1792 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 114688 pages, LIFO batch:31
[    0.000000] percpu: Embedded 31 pages/cpu s87448 r8192 d31336 u126976
[    0.000000] pcpu-alloc: s87448 r8192 d31336 u126976 alloc=31*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: ARM erratum 845719
[    0.000000] CPU features: kernel page table isolation forced ON by KASLR
[... 514 lines cut ...]

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 118.3 GB
disk_used | 4.3 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | 

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

agners, 2021-12-28:
Hm, seems the Raspberry Pi 3A+ has only 512 MB RAM, I'd guess this is memory related still. 

> In the very short time the system is responsive I can see that total memory usage is around 30 %.

Where do you read the total memory usage exactly?

Home Assistant OS with Supervisor and Core really push 512MB systems to the limits, especially 64-bit variants (since they use naturally a bit more memory). I'd recommend something with more memory, especially if you use Add-ons as well.

agners, 2021-12-29:
> I would expect to see out of memory errors if memory was the issue, not unresponsiveness and requests timing out.

Usually, Linux tries at first to free up memory before invoking OOM. This at times can take quite some time and make the system unresponsive. But yeah, eventually you should see OOM errors on the HDMI screen.

> Does HassOS use paging?

Not in the classic sense, but it uses zram, which essentially compresses memory pages the system would like to page out.

If you have a HDMI screen and keyboard connected to the system you can check by executing the command `free` periodically (use `login` to get access to the OS shell).

morgansimonsen, 2022-01-01:
Hi. Where exactly should the OOM messages appear? I have been running the system with an attached monitor for a while, periodically executing `free`. This is the result of a fresh boot without any add ons:
[image]
I guess this confirms the suspicion. ~40 MB free and 0 MB swap does not sound like a viable HA setup. Can you confirm the findings?

thiscantbeserious, 2022-01-04:
HassOS is using Docker - that itself is "lightweight" considered against VMs but then again it's not something you should use with 0.5GB RAM, even 1GB is to little - recommended is 4GB, 2GB is likely already stretching it.

That said I can totally recommend FHEM instead for the 3A+ - it is much more lightweight and can just run fine on it for most of what HA could do - minus the pretty UI - it is quite usablable tought. I'm using a 3A+ connected via Serial to my Central Heating System and have it setup to forward messages via MQTT and Notify me via Telegram if it goes down. 

I like how simple it is ... fits just well with the limited power that the 3A+ brings.

agners, 2022-01-04:
> Hi. Where exactly should the OOM messages appear?

They should appear on the console yes.

> I guess this confirms the suspicion. ~40 MB free and 0 MB swap does not sound like a viable HA setup. Can you confirm the findings?

Yeah that system seems almost at the limit at this point. I mean, if you don't load add-ons and does what it should, then ok, but definitely no room to spare here. You might consider `rpi3` image (32-bit), as that seems to use quite a bit less of memory due to 32-bit pointer size etc.

morgansimonsen, 2022-01-16:
The 32-bit image does not support some very important plugins like SSH. Maybe this should be added to the docs for HA? I thank you all for your knowledge and assistant. Will close this bug.

closed 2022-01-16 by morgansimonsen

agners, 2022-01-17:
SSH Add-on should work well on 32-bit (see https://github.com/home-assistant/addons/tree/master/ssh).

jerrychong25, 2022-03-14:
> HassOS is using Docker - that itself is "lightweight" considered against VMs but then again it's not something you should use with 0.5GB RAM, even 1GB is to little - recommended is 4GB, 2GB is likely already stretching it.
> 
> That said I can totally recommend FHEM instead for the 3A+ - it is much more lightweight and can just run fine on it for most of what HA could do - minus the pretty UI - it is quite useable tought once you got a grip on it.
> 
> I'm using a 3A+ connected via Serial to my Central Heating System and have it setup to forward messages via MQTT and Notify me via Telegram if there's an Error and I have to go downstairs to restart it f.e.
> 
> I like how simple it is ... fits just well with the limited power that the 3A+ brings.

Thank you @thiscantbeserious for your recommendation on at least 2GB RAM for Raspberry Pi 4