home-assistant/operating-system #1698: Kernel Parameters for PCI Dual Edge TPU

tortho, 2021-12-29:
### Describe the issue you are experiencing

PCI Dual Edge TPU does not work on HomeAssistant OS.
Troubleshooting and probably the solution requires kernel parameters to be set and I believe I cannot do that myself?

```
[2021-12-20 11:50:34] detector.coral_pci             INFO    : Starting detection process: 527
W driver/beagle/beagle_kernel_top_level_handler.cc:131] Could not set performance expectation : 122 (Inappropriate ioctl for device)
[2021-12-20 11:50:34] frigate.edgetpu                INFO    : Attempting to load TPU as pci
[2021-12-20 11:50:34] frigate.edgetpu                INFO    : TPU found
E driver/mmio_driver.cc:254] HIB Error. hib_error_status = 0000000000002200, hib_first_error_status = 0000000000000200
E driver/mmio_driver.cc:254] HIB Error. hib_error_status = 0000000000002200, hib_first_error_status = 0000000000000200
[2021-12-20 11:50:54] frigate.watchdog               INFO    : Detection appears to be stuck. Restarting detection process...
[2021-12-20 11:50:54] root                           INFO    : Waiting for detection process to exit gracefully...
```

This is the coral dual edge tpu
https://docs.rs-online.com/15f6/A700000007334657.pdf

This is probably the issue but I cant find any way to do this on HAOS
https://coral.ai/docs/m2/get-started/#troubleshooting-on-linux

This is the computer in use.
https://www.asus.com/Displays-Desktops/Mini-PCs/PN-PB-series/Mini-PC-PN51/techspec/

Others with same issue:
https://community.home-assistant.io/t/home-assistant-os-amd-cpu-coral-pcie-accelerator/342787

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Install Coral Dual Edge TPU
Install Frigate
Try to enable coral detector

### Anything in the Supervisor logs that might be useful for us?

```txt
No, not as I have seen
```

### Anything in the Host logs that might be useful for us?

```txt
No, not as I have seen
```

### System Health information

## System Health

version | core-2021.12.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.83
arch | x86_64
timezone | Europe/Oslo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.18.0
Stage | running
Available Repositories | 934
Installed Repositories | 21

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 7, 2022, 01:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-1.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | failed to load: timeout

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.0
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 465.0 GB
disk_used | 45.7 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.3.3), Mosquitto broker (6.0.1), Terminal & SSH (9.2.2), Home Assistant Google Drive Backup (0.105.2), Z-Wave JS to MQTT (0.32.1), Samba share (9.5.1), Studio Code Server (3.7.0), Double Take (1.7.0), Node-RED (10.2.2), Exadel CompreFace (0.6.1), Frigate NVR Beta (0.10.0) (3.2), Zigbee2mqtt (1.22.1-1), DeepStack (CPU) (2021.09.1)

</details>

<details><summary>Husqvarna Automower</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Lovelace</summary>

dashboards | 5
-- | --
resources | 9
views | 9
mode | storage

</details>

agners, 2021-12-29:
You can set custom kernel command line by editing `cmdline.txt` on the first partition of the boot media. If you have [ssh access to the underlying OS](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host) the file is located in `/mnt/boot/cmdline.txt`. Try appending the suggested kernel command line argument `gasket.dma_bit_mask=32` in that file.

tortho, 2021-12-29:
Thank you for the swift reply!
That files only content were:
console=tty1
Added gasket.dma_bit_mask=32 on a new line and rebooted.

It did not work unfortunately.

tortho, 2022-01-02:
May we have the possibly newer driver implemented in HomeAssistant OS?

From this post it seems like the module is working with the new driver.
https://github.com/blakeblackshear/frigate/discussions/1916#discussioncomment-1782231

tortho, 2022-01-18:
@agners will it be possible to get the new driver implemented or tested?

agners, 2022-04-15:
Google Coral out-of-tree drivers are now part of 8.0.rc2.

closed 2022-04-15 by agners

EticWeb, 2022-10-14:
I have HAOS version 9.2 newly installed on a new computer with google coral dual edge installed into a PCI-EX1 port with the Dual Edge TPU Adapter - PCIe x1 Low Profile.
I switch from a supervised installation on Debian 11 to try HAOS because I face problem to install frigate on that previous configuration.
So now I have Frigate installed and I hope coral driver are installed and my Coral dual edge device  up and running.
I can see an apex_0 into the parameter->system->material and all material.
But frigate always crash and report into the log : 
frigate.edgetpu                ERROR   : No EdgeTPU was detected. If you do not have a Coral device yet, you must configure CPU detectors.
What can I try now?