home-assistant/operating-system #1702: /mnt/overlay disk only supports dates up to 2038

christiaangoossens, 2022-01-02:
### Describe the issue you are experiencing

When booting, HASSOS reports: "ext4 filesystem being mounted at /mnt/overlay supports timestamps until 2038 (0x7fffffff)".
My setup does not seem unique, as this random Google issue: https://gitanswer.com/kvm-hassos-6-2-with-mysql-problem-shell-operating-system-982965894 also has it in the logs.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Install any older version of HassOS
2. Boot and view VNC logs

### Anything in the Supervisor logs that might be useful for us?

```txt
No, this is OS related.
```

### Anything in the Host logs that might be useful for us?

```txt
No, this is OS related.
```

### Additional information

I have updated my setup many times, it might be from the 5.5, 5.6 era. Thus, might not be reproducible in a new install.

christiaangoossens, 2022-01-02:
Many other tickets also include these lines, such as https://github.com/home-assistant/operating-system/issues/1673

agners, 2022-01-03:
You can safely ignore the error, unless you plan to run your instance beyond 2038 :smile: Background: Newer Linux kernels warn when a file system has only been created with 32-bit timestamp support. 

This is fixed for new installations with 7.0 or newer (see #1589). So if you plan to use this installation beyond 2038 (or you want to get rid of the message just because), you have to create a new installation using release 7 or newer. You can use the backup and restore feature to do that. Unfortunately, I don't have another solution currently.

closed 2022-01-03 by agners

agners, 2022-01-03:
Background: Earlier versions of mkfs.ext4 created file systems with Inode size 128 bytes on smaller partition (like the overlay partition). To support 64-bit time stamp inodes need to be 256 bytes. Unfortunately there is no way to change the inode size on file system already created. The only way would be to reformat the overlay partition, which is rather complex/risky doing as an update step.

OS 7.0 and newer have the partition formatted with 256 bytes inode size, so those warnings won't appear anymore.

christiaangoossens, 2022-01-08:
Thanks for your comments! Glad to see that it's already resolved. I hope that this ticket helps some people in the future with finding out why their installations are not working after 2037.