home-assistant/operating-system #1703: odroid C2: device descriptor read/64, error -32 in version 7.x

osegarra, 2022-01-02:
### Describe the issue you are experiencing

Hi, 

After upgrading the haos to version 7.0 my zwave (zwave.me) dongle does not work. When I pluged it in I get the following error:

`[ 341.572014] usb 1-1.2: device descriptor read/64, error -32`

I have tested it in 7.0 and 7.1 versions in order to check if the error is fixed in the latest available version but I get exactly the same error. In version 6.x it worked perfeclty.

Thanks a lot.
Óscar Segarra

### What operating system image do you use?

odroid-c2 (Hardkernel ODROID-C2)

### What version of Home Assistant Operating System is installed?

7.0

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Plug the zwave usb dongle.
2. The error appears: [ 341.572014] usb 1-1.2: device descriptor read/64, error -32

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
`[ 341.572014] usb 1-1.2: device descriptor read/64, error -32`
```

agners, 2022-01-04:
For me it seems that USB hotplug isn't working at all on C2 :see_no_evil: 

However, it seems that this problem is present on upstream Linux since a while: Also with 6.6 hotplug did not work. The problem seems known and there is a discussion on the mailing list about it (https://lore.kernel.org/linux-usb/0badab7c-f12e-e9ed-2f90-2cf5f25f4038@bluematt.me/).

It seems that Armbian [is using a patch](https://github.com/armbian/build/blob/master/patch/kernel/archive/meson64-5.10/general-dwc2-partial-powerdown-fix.patch) which might work around the bug. I'll give this patch a try.

agners, 2022-01-04:
Unfortunately that patch does nothing for ODROID-C2 really: `DWC2_POWER_DOWN_PARAM_NONE` gets already set in ODROID-C2 case, and disables power save modes.

It seems that USB not working properly is a known issue and discussed since quite a while in forums/mailing lists:
https://forum.odroid.com/viewtopic.php?f=135&t=22717&sid=295eb0be993cbfdbd23880bea574a0e3&start=1150

In my case, when I plug in the devices before booting, they work flawlessly.

Did you try rebooting/plug in different device/location?