home-assistant/operating-system #1705: can't log to HASS after 1-2 days

pacoso81, 2022-01-05:
### Describe the issue you are experiencing

I had a problem after i installed second docker (miner) on my PC.
After installing of the second docker my HASS didn't run any more.
I trued to restore my VM image to early point but that did not help, so i update my VirtualBox to latest version and HASS was working.
I then restore the HASS from my latest backup and updated HASS to 12.6/12.1/7.0 (before this latest version).
After everything started working again i noticed that after a day or two i can't log into HASS by web.
In the console i wound that it is frozen at "ha >" and have to do shutdown command and then gives me 
"rcu: INFO:  rcu_sched detected stalls on CPU/tasks: " and a lot of other info,
then i have to do another shutdown command to actually shut down the docker.

Strange thing is that after i start it again and go to log i can see that all the sensors and switches wore working wile i can't access the frontend.

i'm running HASS on VirtualBox 6.1.30 on Windows 10.
core-2021.12.7
supervisor-2021.12.2
Home Assistant OS 7.1

So now i can't tell is it the HASS or VirtualBox problem.
Can you help, please?

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. start the HASS
2. Works for 1-2 days
3. Freeze 
...

### Anything in the Supervisor logs that might be useful for us?

```txt
can't see older logs.
after i restart the HASS everything in the log is green except for 2 warning from Samba and Mosquito for unsupported commands.
```

### Anything in the Host logs that might be useful for us?

```txt
can't see older logs.
```

agners, 2022-01-05:
> "rcu: INFO: rcu_sched detected stalls on CPU/tasks: " and a lot of other info,

Can you post a screenshot of this info? Can you also post the system, storage and network settings?

It seems that the Linux kernel hangs on something (either disk read or network). In those situations its common that some part of the system still work while others don't.

Driekes, 2022-01-23:
I have the exact same thing
[image]

Today it also happened at start, during 'A start job is running for Docker Application Container Engine'. Now trying to get it working again. I don't know what happened, I tried to increase CPU cores, but that didn't help.

pacoso81, 2022-01-23:
i fix my problem whit reinstalling VirtualBox to earlier version and than updating to newest version.
Just make sure you backup your last screenshots, copy them to local drive.

agners, 2022-01-24:
> It is obvious that this is VM problem

What do you mean exactly by that, a problem of VirtualBox or the virtual machine image (HAOS ova)?

> You can find more info on my reply in HA community post.

Which community post? Can you add a link to that post?

pacoso81, 2022-02-13:
it happened again.
i disconnected the audio device in VirtualBox and changed the chipset in System tab to ICH9.
after that i had to select boot device in VM bios.
4-5days have passed and i have no problem.

PowZone, 2022-02-18:
Same issue here with latest versions of Home Assitant and VirtualBox :/

https://imgur.com/a/MEjxM01

Rebooting the VM sometime fix this sometime not

Driekes, 2022-02-18:
I have not seen it the last weeks, not sure what I did. I have updated all to latest. I did try some things with USB versions, disabling all items that have no relevancy (e.g. like above I removed the audio device I think). I also tested with amount of CPU's, but I don't know what exactly the cause it that it runs stable now.

PowZone, 2022-02-18:
Solved for me editing the VM and removing useless devices: floppy, optical, sounds, usb

ExceedingLife, 2022-02-25:
I have been getting this issue lately as well. Plus some others. I was thinking my hard drive is going bad. I can sometimes get my HASSIO to boot and keep it up for most the day but then randomly this happens or many other issues I have a big list of screenshot errors happening. like right now I tried reseting my vm like 7 times and still no success turning it on yet.
[image]

I might want to try a fsck on my diff partitions but im afraid itll freeze up in the middle of me running fsck.
I'll look at removing useless drives and audio stuff. I am also thinking about trying an older version of Virtualbox. or installing Virtualbox on a different SSD. I switched my VM from 1 HDD that was old to a diff SSD and i been getting the same issues.. so i'm confused about that happening.. Here are some other errors i received.
blk_update_request: I/O error, dev sda, sector 3423235...
Buffer I/O error on device sda8, logical block 1084234
systemd-coredump: failed to get COMM no such process
CIFS: VFS: No username specified
rcu: INFO: rcu_sched self-detected stall on CPU
systemd-resolved.service: watchdog timeout
rcy_sched kthread starved for 348343 jiffies!
Failed to start Network Time Sync
SQUASHFS error: Unable to read page, block 343423 size 7c6
EXT4-fs error (device sda8): ext4_journal)check_start:83 Detected aborted journal

and more than all these.. hopefully someone with more knowledge can point me in a direction to go.
Thank you

Driekes, 2022-06-02:
I have been running weeks without issue. I also reduced the amount of CPU's to just one.
I did notice some speed issues, so I tried to increase to two CPU's, but this creates a more unstable environment.
I assigned 2,5GB of memory and 2CPU's to Virtualbox, but it has a harder time starting than with one CPU and less memory.

In Windows I see high memory use (but not sure if this is just Virtualbox reservation), and I see continuous CPU usage. The CLI shows that it is stuck at starting hypervisor.
After a reboot it works, but I do get more 'rcu_sched' notifications and got an unresponsive HA, which I haven't had for weeks.

Reverting back to 1 CPU solves the issues.

pacoso81, 2022-06-02:
> I have been running weeks without issue. I also reduced the amount of CPU's to just one. I did notice some speed issues, so I tried to increase to two CPU's, but this creates a more unstable [...]

Same thing.
Currently i'm running one cpu for 1 week and no stalls, if i add 2 cpu's i get stall in 5-10min after starting.
This happened after update to 5.5, before that i got stalls whit 2 cpu's once a week.
If you get stall most of the times you don't need to restart, just do PAUSE and than UNPAUSE or CTRL+P two times and it will continue, but it might stall again if the process is not finished.

I have read that in linux it is possible that the 2 cpu's can have different timings (not synced to each other) and that can cause big problems and stalls.

pacoso81, 2022-06-02:
since the problem got much worse after upgrading HA i'm sure it is HA problem and not VB.

Driekes, 2022-06-03:
Yesterday I tried again with the latest and greatest versions of Virtualbox, HA, OS and all. Set it to two CPU's and I think within 30 minutes HA stalled, I got the rcu_sched notifications etc.

I'm not sure if the host hardware could be impacting this, or what is the cause, but it just doesn't work. Or maybe it is impossible to switch between cpu's and I should start fresh with 2 cpu's, but I haven't tested that yet.

pacoso81, 2022-06-03:
> Yesterday I tried again with the latest and greatest versions of Virtualbox, HA, OS and all. Set it to two CPU's and I think within 30 minutes HA stalled, I got the rcu_sched notifications etc.
>
> I'm [...]

Maybe it is cpu related problem.
Mine is AMD V1605B, what CPU do you have?

pacoso81, 2022-06-07:
well one cpu no problems, but when i opened task manager i noticed that only cpu1&2 are used, others are parked. i hate loadning only one cpu so i unpark the rest of the cpu's, now all of them are utilized.
we will se how it goes but so far the results are very good. 
i used program unpark cpu.

agners, 2022-06-07:
Maybe this tip works? https://www.virtualbox.org/ticket/20131#comment:2

> I found that "perf top" was good at stalling it out a bit, and doing a "vboxmanage modifyvm foo --hpet on" on the host made the problem occur virtually never or not at all for that VM, even while every other VM without that change was stalling.

Driekes, 2022-06-07:
Tried as well, also didn't work.

pacoso81, 2022-06-08:
here is a video whit explanations regarding cpu stalls.
i think what we are dealing is explained on 16:11

https://www.youtube.com/watch?v=23_GOr8Sz-E

agners, 2022-06-08:
Yeah it basically means the kernel doesn't get to run on a particular CPU within a certain time limit. That can have different causes:
- The kernel itself locks up (on a particular CPU) due to some code blocking (this can be bad/buggy drivers, actual kernel bug, or broken hardware). Unlikely to be the issue here.
- The kernel actually doesn't get to run on a CPU time within a certain timespan, either caused by timer not executed or different expectation of a timer speed. Both causes are likely on Virtual Machines.

IMHO, this is a VirtualBox bug. Maybe VirtualBox needs a particular kernel config to run fine, but if that is the case, it should be documented somewhere. Also, HAOS kernel configuration is not really special, it mainly enables a lot of virtualization drivers. It works fine on other Virtual Machines as well, so :man_shrugging:

Driekes, 2022-06-09:
So far up and running without issue since my previous post! Looks promising.

curtgrimes, 2022-06-12:
I noticed this week that I have been experiencing the same issue with a Home Assistant install I have running in VirtualBox on Windows 10. I may have been experiencing the issue for several weeks/months before this week, but a lot of variables on my end changed about a month ago (installed new networking hardware on computer, caught up on Windows 10 upgrades, reinstalled the latest version of Home Assistant VDI image), so I can't be certain before that point.

I'll try changing System > Accelleration > Paravirtualization interface to KVM per https://github.com/home-assistant/operating-system/issues/1737#issuecomment-1106855304 and report back.

## Specs

PC:
Edition	Windows 10 Pro
Version	21H2
OS build	19044.1706
Experience	Windows Feature Experience Pack 120.2212.4170.0
Processor	Intel(R) Core(TM) i7-9700K CPU @ 3.60GHz   3.60 GHz
Installed RAM	32.0 GB
System type	64-bit operating system, x64-based processor
Pen and touch	No pen or touch input is available for this display

VirtualBox
6.1.34 r150636 (Qt5.6.2)

Home Assistant VM
- Image: haos_ova-7.6.vdi
System
- Base Memory: 10240 MB
- EFI: Enabled
- Accelleration: VT-x/AMD-V, Nested Paging, KVM Paravirtualization
Display
- Video Memory: 16 MB
- Graphics Controller: VMSVGA
- Remote Desktop Server: Disabled
Audio
- Host Driver: Windows DirectSound
- Controller: Intel HD Audio
Storage
- Controller: IDE
- IDE Secondary Device 0 [Optical Drive]: Empty
- Controller: SATA
- SATA Port 0: haos_ova-7.6.vdi (Normal, 32 GB)
Network
- Adapter 1: Intel PRO/1000 MT Desktop (Bridged Adapter, Intel(R) Wi-Fi 6E AX210 160Mhz)
Serial ports: Disabled
USB:
- USB Controller: OHCI
- Device Filters: 1 (1 active)

Driekes, 2022-06-13:
@curtgrimes the comment below mentions KVM didn't work as I also believe that is equal to default. However, try it!

Mine is now stable with 2 cpus with the minimal setting!

pacoso81, 2022-06-13:
[image]
one cpu, but unparked all cores.
VB is sharing all cpu's whit windows.
in the time of snapshot i'm updating HA and nothing else is using the cpu.
i'm using this method for more than a week with not isuess so far.

mig8447, 2022-08-30:
I tested changing the para-virtualization setting to minimal, and although it lasted longer before freezing, it froze anyway. One curious thing I discovered is that if you enable the virtual keyboard from Virtual Box  (Input > Keyboard) and type something, the machine unfreezes, but that's not good either way.

I also noticed that before I saw the rcu errors on the VM screen but now it freezes without telling me anything, I'm quite sure, the issue is still the rcu_sched warnings though

I reduced the number of CPUs to 1 to see if that helps.

mig8447, 2022-09-02:
My setup has been running smoothly for 3 days by just reducing the number of CPUs to 1, even when leaving the default para virtualization parameters

jjcf89, 2022-10-10:
Same issue, I will set cpu's to 1 to see if that helps.

OGTK423, 2022-10-12:
I'm currently experiencing this issue but have also never been able to launch HA in a VM when using 1 CPU, it always results in a bootloop and won't successfully launch unless 2 CPUs is selected.

jjcf89, 2022-10-12:
> Same issue, I will set cpu's to 1 to see if that helps.

So far 1 CPU has been stable after 2 days, fingers crossed.

pacoso81, 2022-10-12:
mine is runing happy whit one cpu and unparked cores for months now whitout any stall so far.
i will continue to work on one cpu untill this problem is fixed.

yzlnew, 2022-10-28:
I'm also experiencing this issue on Proxmox VM.

lbouriez, 2023-01-01:
Hello, I am also experiencing this issue since yesterday on VirtualBox without any changes to my setup.
Did someone get to find a solution other than reducing the CPUs number ?

Driekes, 2023-01-02:
> I saw in another post [#1737 (comment)](https://github.com/home-assistant/operating-system/issues/1737#issuecomment-1108843382) that you can change the paravirtualization settings and that some worked. I now am running on 'minimal' and so far it is running for a few hours, more than the test yesterday. I however also increased memory so it could still be that has some impact. I'll keep you updated.

@lbouriez 
Yes, see my comment.

Driekes, 2023-01-03:
I have a couple of feedback points:

- I'm using Home Assistant as home automation and I want all my devices not
to be depended on cloud environments. They can have issues, can be shut
down. So please investigate the possibility to access the device locally.
- If not locally, please report from cloud a counter that just lists total
energy consumption. This way I'm not depended on constant reset to 0 and I
have a more stable idea on total consumption. Would be great if device
reports this, so in case of (long) wifi outage, the totals are still
correct.
- As I understand the total energy consumption is not very accurate,
especially when there are voltage fluctuations. Not validated yet, but
seems Wallbox is assuming 230V, which is of course not always the case.

Kind regards,

Ruud

jjcf89, 2023-01-03:
> I have a couple of feedback points:

I think you've posted in the wrong place

polter05, 2023-01-31:
Same issue on proxmox :(
[image]

yzlnew, 2023-01-31:
> Same issue on proxmox :( [image]

Updating the kernel fix the problem for me.

polter05, 2023-01-31:
Which version ? Mine is Linux 5.15

Benoit Waldmann

Tel. : +33 3 55 40 90 57

Mail: ***@***.******@***.***>

SIAGILUS

6 All. de la Forêt de la Reine

54500 Vandoeuvre-Lès-Nancy

Site: https://siagilus.fr/<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fsiagilus.fr%2F&data=05%7C01%7Cbenoit.waldmann%40siagilus.fr%7C7444ed178b044755126e08da8b5da6dc%7C6a8a7862481b4f6c8236058d5f89a731%7C0%7C0%7C637975531590266632%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=6xCpWHjMPv7PRuk85HOprXRVg%2B121ltQrXc%2ByQkO4E0%3D&reserved=0>

Diffusion restreinte

lbouriez, 2023-01-31:
FYI, what I noticed for me is that it was happening while shutting down the VM hardly, I mean like an electricity outtage.

yzlnew, 2023-02-01:
> Which version ? Mine is Linux 5.15 Benoit Waldmann Tel. : +33 3 55 40 90 57 Mail: ***@***.******@***.***> SIAGILUS 6 All. de la Forêt de la Reine 54500 Vandoeuvre-Lès-Nancy Site: [...]

Update to an Edge Kernel using scrips from https://tteck.github.io/Proxmox/

hprotzek, 2023-02-06:
I got the same issue on Proxmox.
[image]

Not sure if it works, but here is a suggestion for a workaround:
https://bugzilla.kernel.org/show_bug.cgi?id=199727#c18

Setting VirtIO SCSI Single / iothread=1 / aio=threads on all our KVM guests.

pacoso81, 2023-02-06:
i think it is time for an update.
i'm still running on 1 cpu and unparked cores (basically i'm running on all of the cpu's) since June 2022 and no stall whatsoever.
for me this is the best solution, even if i didn't had stalls i would run unparked cpu's because that way i can utilize all of my cpu's cores at once.

agners, 2023-02-06:
This issue is about rcu stalls on VirtualBox. I opened a new issue for rcu stalls on Proxmox, see #2342.

lobhater, 2023-04-11:
So I am running HA in virtual box on windows 11.  Everything has been running great and I have been doing monthly updates. This week I installed the add-on RTSPtoWeb - WebRTC and the system started freezing seemingly randomly with no logs except 
[image]
I uninstall WebRTC and everything is perfect again. Someone on FB of all places linked me here and I read the workaround of changing to 1 cpu which also fixed the issue. Seems this has been an ongoing issue with no solution, right?

madduck, 2023-05-18:
Just had this problem while running 2023.5.3 on libvirt/kvm/plain Debian.

jjcf89, 2023-08-16:
Seems unlikely that this problem went away.  Most people are probably just running with 1 core as the only known work around.

mlefe, 2023-09-06:
Same here, running Proxmox with HA and I got this error as soon as I moved from 2 to 4 CPUs. I also tried 3 CPUs and it fails so I'll put 1 CPU (just in case).

mlefe, 2023-09-06:
> Same here, running Proxmox with HA and I got this error as soon as I moved from 2 to 4 CPUs. I also tried 3 CPUs and it fails so I'll put 1 CPU (just in case).

An update, my HA instance really started to fail and I realized that an issue in my USB kept appearing: so I unmapped the bluetooth usb device from the HA instance. Looks like everything came back to normal...

closed 2023-12-13 by github-actions[bot]

captcurrent, 2024-01-04:
Ok just found this in my research.   I am running HA on a Linux Guest on a Mac OS host.   Most of my UNIX experience is on FreeBSD so I choose Linux because a) the examples said to and b) thought it was closest to my FreeBSD. experience.  I had been tweaking the Guest settings bur age strange thing is that I and running two installation in different  locations with pretty much the same configuration.   One throws me these CPU errors the other is rock solid.   Any suggestion on best approach such as erase the who thing and start from the beginning?  
[image]

jjcf89, 2024-01-04:
Try updating the VM to use one CPU core

captcurrent, 2024-01-06:
I update stop use one core.   I have two installations on two different machines on two different locations.. One work pretty rock steady this one does not.  I have match the VM settings.   One keeps freeing the other one seems to freeze every day

RaM555468, 2024-01-22:
Hey @jjcf89 , @captcurrent @mlefe 

I am not sure about the virtualbox options regarding CPUs and cores, but maybe you could check this. In VMWare Workstation you can configure "number of processors" and "number of cores per processor" and I had the mentioned error with option 1 set to 4 (number of processors) while setting this to 1 and setting "number of cores per processor" to 4 instead seems to help.

Grawl, 2024-03-03:
Just faced this issue on HAOS / Windows 11 / VirtualBox. Checked number of cores, it was 2. Changed to 1.

Also just disabled USB and IDE.

Will report here if it will not help.

ruppde, 2024-04-17:
Please check if your VM's vbox.log contains these lines:

```
{timestamp} HM: HMR3Init: Attempting fall back to NEM: VT-x is not available
{timestamp} NEM: WHvCapabilityCodeHypervisorPresent is TRUE, so this might work...
```
or
```
{timestamp} HM: HMR3Init: Attempting fall back to NEM: AMD-V is not available
{timestamp} NEM: WHvCapabilityCodeHypervisorPresent is TRUE, so this might work...
```

Then it's this problem with Windows activating Hyper-v vs. Virtualbox:
https://forums.virtualbox.org/viewtopic.php?p=506311&sid=ab3928de3126cacd81185751f59b141e#p506311