home-assistant/operating-system #1707: Parallels Desktop could not start Home Assistant after upgraded to Release-7 build 1

bsx1990, 2022-01-06:
### Describe the issue you are experiencing

After updated to the latest version, I'm blocked to start Home Assistant VM from Parallels Desktop.
Here is the error message screenshot:
[image]
Could anyone help to look at this issue?

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Not sure about that

### Anything in the Supervisor logs that might be useful for us?

```txt
Could not get the logs
```

### Anything in the Host logs that might be useful for us?

```txt
Could not get the logs
```

agners, 2022-01-06:
Uh, it seems that the boot partition got corrupted. What disk image did you use?

You probably can restore it by replacing that first partition with a genuine one from a freshly downloaded image. But I am not familiar with parallels, so I can't provide exact instructions.

bsx1990, 2022-01-08:
Previously, I downloaded the vdi disk image from the Home Assistant website and then converted it to the .hdd files for Parallel Desktops.

Yesterday, I tried to mount the disk image to a Ubuntu VM from Parallel Desktops. I tried to replace the /boot files with the latest vdi image from Home Assistant website. But I failed.

Then I copied the backup files and recovered my Home Assistant system. For now, it works well. But I lost the history data.

But it's not very important. It works now.

bsx1990, 2022-01-08:
@agners Thanks for your reply!

closed 2022-01-08 by bsx1990

tangentsoft, 2024-03-09:
I got it working with the VDI image, converted like so:

```
prl_convert haos_ova-12.0.vdi --dst=home-assistant.hdd --allow-no-os --stand-alone-disk
```

I then attached that to a blank Debian VM, with the default HDD removed and this one attached in its place.

The trick is to then enable EFI support on that newly-attached HDD. Follow [these instructions](https://kb.parallels.com/en/122735) except that you set the value to 1, not 0.