home-assistant/operating-system #1710: OS partially hangs

bettio, 2022-01-08:
### Describe the issue you are experiencing

The operating system on the RapsberryPi stops behaving properly randomly after some days:
- Most of the ha web UI fails to load except some random pages (such as the login page and the lovelace dashboard)
- the weather is displayed correctly on the lovelace dashboard
- the nanoleaf light integration still works (and it is possible to switch on and off the light from the lovelace dashboard)
- deconz doesn't work
- doing ssh on the same IP fails with "No route to host"
- the observer (4357 port) doesn't work 

In order to fix the issue the system must be power cycled since ssh is not working.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

This issue happens randomly without any external interaction

### Anything in the Supervisor logs that might be useful for us?

```txt
Log is lost after reboot.
```

### Anything in the Host logs that might be useful for us?

```txt
Log is lost after reboot.
```

### System Health information

## System Health

version | core-2021.12.8
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v8
arch | aarch64
timezone | Europe/Rome

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 219.4 GB
disk_used | 7.5 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), deCONZ (6.11.1), File editor (5.3.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

### Additional information

```
$ dmesg 
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.10.63-v8 (builder@77cfd3deba5a) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -g3cedcad6) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Wed Dec 29 18:55:14 UTC 2021
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[...]
```

agners, 2022-01-10:
Hm, this could be caused by a memory leak of some type, leading to processes getting killed. Can you enable [system monitoring integration](https://www.home-assistant.io/integrations/systemmonitor/) to see if memory usage is growing over time?

bettio, 2022-01-13:
today I found this on dmesg:
```
[138619.145040] sd 0:0:0:0: [sda] tag#28 uas_eh_abort_handler 0 uas-tag 10 inflight: CMD OUT 
[138619.145063] sd 0:0:0:0: [sda] tag#28 CDB: opcode=0x2a 2a 00 01 12 88 b8 00 00 08 00
[138619.145462] sd 0:0:0:0: [sda] tag#27 uas_eh_abort_handler 0 uas-tag 9 inflight: CMD OUT 
[138619.145476] sd 0:0:0:0: [sda] tag#27 CDB: opcode=0x2a 2a 00 01 12 47 90 00 00 08 00
[138619.145819] sd 0:0:0:0: [sda] tag#26 uas_eh_abort_handler 0 uas-tag 8 inflight: CMD OUT 
[138619.145832] sd 0:0:0:0: [sda] tag#26 CDB: opcode=0x2a 2a 00 01 12 46 68 00 00 08 00
[138619.146209] sd 0:0:0:0: [sda] tag#25 uas_eh_abort_handler 0 uas-tag 7 inflight: CMD OUT 
[138619.146222] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x2a 2a 00 01 12 45 18 00 00 08 00
[138619.146602] sd 0:0:0:0: [sda] tag#24 uas_eh_abort_handler 0 uas-tag 5 inflight: CMD OUT 
[138619.146614] sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x2a 2a 00 01 12 42 98 00 00 08 00
[138619.146991] sd 0:0:0:0: [sda] tag#23 uas_eh_abort_handler 0 uas-tag 3 inflight: CMD OUT 
[138619.147004] sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x2a 2a 00 01 12 3f 18 00 00 08 00
[138619.147377] sd 0:0:0:0: [sda] tag#22 uas_eh_abort_handler 0 uas-tag 6 inflight: CMD OUT 
[138619.147390] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x2a 2a 00 01 12 43 c0 00 00 08 00
[138619.147762] sd 0:0:0:0: [sda] tag#21 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD OUT 
[138619.147774] sd 0:0:0:0: [sda] tag#21 CDB: opcode=0x2a 2a 00 01 12 41 70 00 00 08 00
[138619.148161] sd 0:0:0:0: [sda] tag#20 uas_eh_abort_handler 0 uas-tag 4 inflight: CMD OUT 
[138619.148173] sd 0:0:0:0: [sda] tag#20 CDB: opcode=0x2a 2a 00 01 12 40 40 00 00 08 00
[138619.148567] sd 0:0:0:0: [sda] tag#10 uas_eh_abort_handler 0 uas-tag 20 inflight: CMD 
[138619.148580] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x85 85 06 20 00 00 00 00 00 00 00 00 00 00 00 e5 00
[138619.148619] sd 0:0:0:0: [sda] tag#9 uas_eh_abort_handler 0 uas-tag 19 inflight: CMD OUT 
[138619.148630] sd 0:0:0:0: [sda] tag#9 CDB: opcode=0x2a 2a 00 00 55 c9 98 00 00 08 00
[138619.148967] sd 0:0:0:0: [sda] tag#8 uas_eh_abort_handler 0 uas-tag 18 inflight: CMD OUT 
[138619.148980] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x2a 2a 00 0c 9d 9a f0 00 00 18 00
[138619.149377] sd 0:0:0:0: [sda] tag#7 uas_eh_abort_handler 0 uas-tag 17 inflight: CMD OUT 
[138619.149390] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x2a 2a 00 0c 9d 9a d0 00 00 08 00
[138619.149758] sd 0:0:0:0: [sda] tag#6 uas_eh_abort_handler 0 uas-tag 16 inflight: CMD OUT 
[138619.149771] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x2a 2a 00 00 a1 47 c0 00 00 08 00
[138619.150153] sd 0:0:0:0: [sda] tag#5 uas_eh_abort_handler 0 uas-tag 15 inflight: CMD OUT 
[138619.150165] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x2a 2a 00 0c 9d 9b 08 00 00 08 00
[138619.150552] sd 0:0:0:0: [sda] tag#4 uas_eh_abort_handler 0 uas-tag 14 inflight: CMD OUT 
[138619.150564] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x2a 2a 00 01 12 e1 28 00 00 18 00
[138619.150954] sd 0:0:0:0: [sda] tag#3 uas_eh_abort_handler 0 uas-tag 13 inflight: CMD OUT 
[138619.150966] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x2a 2a 00 01 12 8c 30 00 00 08 00
[138619.151344] sd 0:0:0:0: [sda] tag#2 uas_eh_abort_handler 0 uas-tag 12 inflight: CMD OUT 
[138619.151357] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x2a 2a 00 01 12 8b 08 00 00 08 00
[138619.151733] sd 0:0:0:0: [sda] tag#1 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD 
[138619.151745] sd 0:0:0:0: [sda] tag#1 CDB: opcode=0x2a 2a 00 00 26 4e 30 00 00 28 00
[138619.151781] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 11 inflight: CMD OUT 
[138619.151791] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x2a 2a 00 01 12 89 e0 00 00 08 00
[138643.721283] sd 0:0:0:0: [sda] tag#11 uas_eh_abort_handler 0 uas-tag 21 inflight: CMD OUT 
[138643.721307] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x2a 2a 00 00 13 0c 08 00 00 02 00
[138648.585355] sd 0:0:0:0: [sda] tag#14 uas_eh_abort_handler 0 uas-tag 22 inflight: CMD OUT 
[138648.585378] sd 0:0:0:0: [sda] tag#14 CDB: opcode=0x2a 2a 00 00 13 cb 18 00 00 0e 00
[138648.601344] scsi host0: uas_eh_device_reset_handler start
[138648.730734] usb 2-2: reset SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[138648.755080] scsi host0: uas_eh_device_reset_handler success
[138648.755744] sd 0:0:0:0: [sda] tag#10 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x06 cmd_age=37s
[138648.755767] sd 0:0:0:0: [sda] tag#10 CDB: opcode=0x85 85 06 20 00 00 00 00 00 00 00 00 00 00 00 e5 00
```

This raspberrypi has an external USB3 SSD disk, that is connected using a SATA-USB3 adapter. It worked flawlessly for more than one year, did anything change in the kernel configuration?

agners, 2022-01-13:
> did anything change in the kernel configuration?

Not really in the configuration, but we updated to the latest tagged release from the Raspberry Pi kernel team in #1696. However, its not a new kernel major version, and this should be mostly contain bug fixes etc.

That said, this kind of UAS issues are [quite well known problem](https://forums.raspberrypi.com/viewtopic.php?t=245931) with Rapberry Pi though. Typically adding `usb-storage.quirks=xxxx:yyyy:u` (replace xxxx and yyyy with PID/VID of the USB3 SSD adapter you are using) to the `cmdline.txt` file present on the first partition of the SSD should help. 

It is strange though that this now suddenly appears.

dallas6672, 2022-01-17:
I just had my second crash since upgrading to 7.1. Was previously running 5.2 (which ran great) since I was having the same issue as #1119. I am running off an SD Card, but I do have a USB device connected (Wyze Sense bridge). I enabled the Memory monitor like @agners requested of the OP. Free memory stayed around 3,347 megs when everything was working. After the crash it increased to about 3,390 megs.

I can still access a few of the menus from the browser like the Dashboards, Maps and Developer Tools. Under Configurations, I can access everything under Settings except for Info. I lose access to all my MQTT devices, but can still control my ZHA devices (Sonoff Zigbee bridge running Tasmota). I also lose access to SSH and the observer. 

Here are the only logs that show up under Configuration > Settings > Logs...

- Failed to to call /supervisor/available_updates -
- 8:22:31 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:18:10 PM and shows up 4 times
- Client error on /core/info request Cannot connect to host 172.30.32.2:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
- 8:22:31 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:06:48 PM and shows up 28 times
- Error handling request
- 8:22:22 PM – (ERROR) components/http/static.py - message first occurred at 8:18:13 PM and shows up 8 times
- Client error on api app/entrypoint.js request Cannot connect to host 172.30.32.2:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
- 8:22:21 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:18:13 PM and shows up 2 times
- Can't read Supervisor data:
- 8:21:58 PM – (WARNING) Home Assistant Supervisor - message first occurred at 8:06:48 PM and shows up 4 times
- Error doing job: Task exception was never retrieved
- 8:21:23 PM – (ERROR) helpers/storage.py - message first occurred at 8:11:23 PM and shows up 4 times
- Unhandled exception
- 8:18:39 PM – (ERROR) /usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py - message first occurred at 8:18:38 PM and shows up 10 times
- Error fetching openweathermap data: API call timeouted
- 8:07:52 PM – (ERROR) OpenWeatherMap - message first occurred at January 13, 2022, 12:00:18 AM and shows up 3 times
- Disconnected from MQTT server core-mosquitto:1883 (16)
- 8:06:10 PM – (WARNING) MQTT - message first occurred at 8:06:10 PM and shows up 2 times

agners, 2022-01-17:
@dallas6672 the problem the OP is describing is clearly USB SSD related (as indicated by the `uas_eh_abort_handler` errors). Despite similar symptoms, your case is likely unrelated. Can you open a new issue?

bettio, 2022-01-18:
> > did anything change in the kernel configuration?
> 
> Not really in the configuration, but we updated to the latest tagged release from the Raspberry Pi kernel team in #1696. However, its not a new kernel major version, and this should be mostly contain bug fixes etc.
> 
> Which HAOS version did you use before?

I can't remember, is it possible to retrieve any OS updates history? Anyway a number of versions since April 2021.

> That said, this kind of UAS issues are [quite well known problem](https://forums.raspberrypi.com/viewtopic.php?t=245931) with Rapberry Pi though. Typically adding `usb-storage.quirks=xxxx:yyyy:u` [...]

Indeed, it is quite strange since it worked for months, I think since April 2021, without any quirks option. Now suddenly the same problem happened 3 o 4 times in 2 weeks. So it looks like something changed somewhere.
If I am correct, this specific adapter was know to work without any kind of issue.

agners, 2022-01-24:
@bettio did storage.quirks help in your case?

> If I am correct, this specific adapter was know to work without any kind of issue.

Which one is that exactly?

With OS 7.1 we introduced a new Raspberry Pi firmware, so maybe that changed something. You can downgrade the OS via terminal command quite easily:

```
ha os update --version 7.0
```

It would be interesting if the problem disappears by just downgrading to that previous version. Note that 7.2 just released, but there are not changes to Raspberry Pi, so I don't expect that to make a difference compared to 7.1.

closed 2022-05-01 by github-actions[bot]