home-assistant/operating-system #1723: HA crashing about once a week after upgrading to 7.1

dallas6672, 2022-01-17:
### Describe the issue you are experiencing

I just had my second crash since upgrading to 7.1. Was previously running 5.2 for about a year (which ran great) since I was having the same issue as #1119. I am running off an SD Card, but I do have a USB device connected (Wyze Sense bridge). I enabled the Memory monitor after the first crash. Free memory stayed around 3,347 megs when everything was working. After the crash it increased to about 3,390 megs.

I can still access a few of the menus from the browser like the Dashboards, Maps and Developer Tools. Under Configurations, I can access everything under Settings except for Info. I lose access to all my MQTT devices, but can still control my ZHA devices (Sonoff Zigbee bridge running Tasmota). I also lose access to SSH and the observer.

Here are the only logs that show up under Configuration > Settings > Logs...

> Failed to to call /supervisor/available_updates -
> 8:22:31 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:18:10 PM and shows up 4 times
> Client error on /core/info request Cannot connect to host 172.30.32.2:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
> 8:22:31 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:06:48 PM and shows up 28 times
> Error handling request
> 8:22:22 PM – (ERROR) components/http/static.py - message first occurred at 8:18:13 PM and shows up 8 times
> Client error on api app/entrypoint.js request Cannot connect to host 172.30.32.2:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
> 8:22:21 PM – (ERROR) Home Assistant Supervisor - message first occurred at 8:18:13 PM and shows up 2 times
> Can't read Supervisor data:
> 8:21:58 PM – (WARNING) Home Assistant Supervisor - message first occurred at 8:06:48 PM and shows up 4 times
> Error doing job: Task exception was never retrieved
> 8:21:23 PM – (ERROR) helpers/storage.py - message first occurred at 8:11:23 PM and shows up 4 times
> Unhandled exception
> 8:18:39 PM – (ERROR) /usr/local/lib/python3.9/site-packages/aiohttp/web_protocol.py - message first occurred at 8:18:38 PM and shows up 10 times
> Error fetching openweathermap data: API call timeouted
> 8:07:52 PM – (ERROR) OpenWeatherMap - message first occurred at January 13, 2022, 12:00:18 AM and shows up 3 times
> Disconnected from MQTT server core-mosquitto:1883 (16)
> 8:06:10 PM – (WARNING) MQTT - message first occurred at 8:06:10 PM and shows up 2 times

### What operating system image do you use?

rpi4  (Raspberry Pi 4/400 32-bit OS)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Allow HA to run for about a week and eventually it crashes

### Anything in the Supervisor logs that might be useful for us?

```txt
Unable to access once it's crashed
```

### Anything in the Host logs that might be useful for us?

```txt
Unable to access once it's crashed
```

### System Health information

## System Health

version | core-2021.12.9
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v7l
arch | armv7l
timezone | America/New_York

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 56.6 GB
disk_used | 5.6 GB
healthy | true
supported | true
board | rpi4
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Glances (0.14.1), Mosquitto broker (6.0.1), ESPHome (2021.12.3), TasmoAdmin (0.16.0), TasmoBackup (1.04.26)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 6
mode | storage

</details>

### Additional information

Raspberry Pi 4 Model B Rev 1.2

agners, 2022-01-17:
> I can still access a few of the menus from the browser like the Dashboards, Maps and Developer Tools.

So to be clear crash means that only some part of the web frontend are unavailable? So this is unlikely the OS which crashes, but "only" some services then.

> I can still access a few of the menus from the browser like the Dashboards, Maps and Developer Tools. 

Is that also true when forcing a page refresh (Ctrl+F5)

> Here are the only logs that show up under Configuration > Settings > Logs...

Are those log when the system is partially unavailable?

dallas6672, 2022-01-17:
> So to be clear crash means that only some part of the web frontend are unavailable? So this is unlikely the OS which crashes, but "only" some services then.

Correct

>     I can still access a few of the menus from the browser like the Dashboards, Maps and Developer Tools.
> 
> Is that also true when forcing a page refresh (Ctrl+F5)

Don't recall if I tried that or not. I'll be sure test next time it happens.

> Are those log when the system is partially unavailable?

Correct

agners, 2022-01-17:
I'd also recommend to enable SSH access on OS level using the [method described here](https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host). This should allow you to connect to the system on OS level next time it goes partly down. From the OS shell, the logs of Supervisor would be particularly interesting. You can get the log using:

```
docker logs hassio_supervisor
```

dallas6672, 2022-01-24:
Just crashed again, but for some reason I get "connection refused" now when trying to access SSH at the OS level. I set it up using the method you mentioned and it was working earlier in the week. When trying to access SSH through the add-on the connection just times out as expected.

Basically the same logs as before from Configuration > Settings > Logs...

> Error doing job: Task exception was never retrieved
> 7:54:45 PM – (ERROR) helpers/storage.py - message first occurred at 7:49:35 PM and shows up 3 times

> Can't read Supervisor data:
> 7:53:46 PM – (WARNING) Home Assistant Supervisor

> Client error on /host/info request Cannot connect to host 172.30.32.2:80 ssl:default [Connect call failed ('172.30.32.2', 80)]
> 7:53:46 PM – (ERROR) Home Assistant Supervisor - message first occurred at 7:53:46 PM and shows up 6 times

> Error fetching openweathermap data: HTTPSConnectionPool(host='api.openweathermap.org', port=443): Max retries exceeded with url: /data/2.5/weather?APPID=4fe9b26f9fdbf16b99777702fbfda5e5&lang=en&lon=-86.032356&lat=37.72696 (Caused by NewConnectionError('<urllib3.connection.HTTPSConnection object at 0xa7847598>: Failed to establish a new connection: [Errno -3] Try again'))
> 7:53:24 PM – (ERROR) OpenWeatherMap

> Disconnected from MQTT server core-mosquitto:1883 (16)
> 7:50:37 PM – (WARNING) MQTT - message first occurred at 7:50:37 PM and shows up 2 times

dallas6672, 2022-01-24:
After rebooting, these are the logs from "docker logs hassio_supervisor" shortly before crashing and the beginning of the boot after the crash...

```
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.CORE
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.FREE_SPACE/ContextType.SYSTEM
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.PLUGIN
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.check] System checks complete
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-01-23 19:13:28 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-01-23 19:18:28 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-01-23 19:20:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/danmed/TasmoBackupV1 repository
22-01-23 19:20:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/sjthespian/addon-tuyagateway repository
22-01-23 19:20:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/home-assistant/addons repository
22-01-23 19:20:25 INFO (MainThread) [supervisor.store.git] Update add-on https://github.com/hassio-addons/repository repository
22-01-23 19:20:32 INFO (MainThread) [supervisor.store] Loading add-ons from store: 64 all - 0 new - 0 remove
22-01-23 19:20:32 INFO (MainThread) [supervisor.store] Loading add-ons from store: 64 all - 0 new - 0 remove
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[03:41:33] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[03:41:34] INFO: Starting local supervisor watchdog...
22-01-24 03:41:37 INFO (MainThread) [__main__] Initializing Supervisor setup
22-01-24 03:41:37 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-01-23 22:41:37 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: raspberrypi4
22-01-23 22:41:37 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/armv7-hassio-supervisor with version 2021.12.2
```

agners, 2022-01-24:
Hm, it seems as is the Supervisor just disappeared, without even shutting down. That is strange.

Can you check all the logs from the time window around there, e.g. using this command:

```
journalctl --since "22-01-23 19:20:00" --until "22-01-23 19:30:00"
```

(you might want to adjust/extend the time range, depending on what shows up in the log exactly.

dallas6672, 2022-01-24:
journalctl --since "22-01-23 19:20:00" --until "22-01-23 19:30:00"
```
-- Journal begins at Fri 2021-04-23 09:22:20 UTC, ends at Mon 2022-01-24 13:24:47 UTC. --
Jan 23 19:20:45 HomeAssistant homeassistant/armv7-hassio-supervisor:latest/hassio_supervisor[453]:
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965359: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965423: Client <unknown> has exceeded timeout, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965434: Client DVES_F6XXXX has exceeded timeout, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965435: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965463: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965463: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: Client DVES_F6XXXX already connected, closing old connection.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: Socket error on client DVES_F6XXXX, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965496: Saving in-memory database to /data/mosquitto.db.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965867: New connection from 192.168.4.170 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965867: Client DVES_02XXXX already connected, closing old connection.
Jan 23 19:25:45 HomeAssistant homeassistant/armv7-hassio-supervisor:latest/hassio_supervisor[453]:
```

I then told it to show out to 19:50 and got this...

journalctl --since "22-01-23 19:20:00" --until "22-01-23 19:50:00"
```
-- Journal begins at Fri 2021-04-23 09:22:20 UTC, ends at Mon 2022-01-24 14:07:06 UTC. --
Jan 23 19:20:45 HomeAssistant homeassistant/armv7-hassio-supervisor:latest/hassio_supervisor[453]:
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965359: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965423: Client <unknown> has exceeded timeout, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965434: Client DVES_F6XXXX has exceeded timeout, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965435: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965463: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965463: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: Client DVES_F6XXXX already connected, closing old connection.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: Socket error on client DVES_F6XXXX, disconnecting.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New connection from 192.168.4.148 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965480: New client connected from 192.168.4.148 as DVES_F6XXXX (p2, c1, k30, u'addons').
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965496: Saving in-memory database to /data/mosquitto.db.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965867: New connection from 192.168.4.170 on port 1883.
Jan 23 19:24:27 HomeAssistant homeassistant/armv7-addon-mosquitto:6.0.1/addon_core_mosquitto[453]: 1642965867: Client DVES_02XXXX already connected, closing old connection.
[... cut ...]
Jan 23 19:43:50 HomeAssistant kernel: audit: type=1334 audit(1642967030.106:816): prog-id=322 op=UNLOAD
Jan 23 19:43:50 HomeAssistant kernel: audit: type=1334 audit(1642967030.106:817): prog-id=321 op=UNLOAD
Jan 23 19:45:46 HomeAssistant homeassistant/armv7-hassio-supervisor:latest/hassio_supervisor[453]:
Jan 23 19:48:20 HomeAssistant homeassistant/armv7-hassio-supervisor:latest/hassio_supervisor[453]: 22-01-23 14:48:20 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
```

agners, 2022-01-24:
It seems the Supervisor is up and doing well. The logs show no indication that something is wrong.

So at ~19:50, the system was not available?

Since you claim its caused by the OS upgrade, can you try a OS downgrade to see if it indeed disappears?

dallas6672, 2022-01-24:
Correct. At 19:50 I was having the same issue listed in the original post. This is the same issue I had over a year ago when I found the issue #1119. That is when I downgraded to 5.2 and the issue went away. 

I will downgrade it back to 5.2 unless there's another version you'd like me to try.

dallas6672, 2022-01-31:
A little over a week now and it hasn't crashed yet on 5.2.

dallas6672, 2022-02-08:
Is there anything else you'd like for me to try?

agners, 2022-02-08:
I don't really have a clue from the logs what could be happening. Either the logs are incomplete, or the anomaly is not caused by the Raspberry Pi itself.

Maybe its network related somehow?

Have you tried with another SD card? Also a clean install of OS 7.1 on a new SD card running on the same hardware would be interesting.

How do you downgrade? We use the A/B system, so it could be that 7.1 is on bad part of the SD card (wild guess).

dallas6672, 2022-02-08:
Haven't tried wiping the SD card and starting over yet but when I downgrade I just use the "ha os update --version x.x" command

dallas6672, 2022-02-14:
Wiped/Reimaged the SD card yesterday and it just crashed again. Also switched to 64 bit since it's now the recommended version. Was 32-bit when I originally installed it. 

Weird that it keeps happening on Sunday nights around this time. 7-8 PM on Sundays. 

SSH still refuses the connection. I was able to access the "Info" section this time (See below). I the past it would always error out.

I will try a different SD card now

## System Health

version | core-2022.2.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | America/New_York

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 56.6 GB
disk_used | 3.0 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | failed to load: unreachable
version_api | pending
installed_addons | ESPHome (2022.1.4), Glances (0.14.2), Mosquitto broker (6.0.1), TasmoAdmin (0.16.0), TasmoBackup (1.04.26), Terminal & SSH (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 1
views | 6
mode | storage

</details>

dallas6672, 2022-02-28:
Almost 2 weeks on a different microSD card and haven't seen the issue yet.

Could it be possible that there's something with this type/model of microSD? Or, if that were the case, would it not work at all?

agners, 2022-04-05:
Unfortunately SD cards are not that reliable. Could be a bad SD card.

Assuming this was an hardware issue then, closing.

closed 2022-04-05 by agners