home-assistant/operating-system #1724: 7.2 qcow2 doesn't boot. startup.nsh message

jmcollin78, 2022-01-22:
### Describe the issue you are experiencing

When trying to install the latest HA OS image on my Ubuntu 20 lts with KVM box.
The boot process stop immediatly with a message:

[image]

event if I try to change boot order, this error appears. The image seems corrupted. 

The image I try is: haos_ova-7.2.qcow2.xz

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Follow this installation step: https://www.home-assistant.io/installation/linux (7.1 image)
2. try with 7.2 image with same problems.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor logs are not available. Image won't start
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System Health information

-

### Additional information

-

jmcollin78, 2022-01-22:
See more info here: https://community.home-assistant.io/t/system-update-fail/218508/6

agners, 2022-01-24:
Did you follow the installation steps on the website? https://www.home-assistant.io/installation/linux#hypervisor-specific-configuration

The disk image is compressed using xz, make sure to uncompress it using `unxz` first.

dinomite, 2022-01-24:
I think I'm seeing this in the Proxmox VM that I use for Home Assistant. The console in Proxmox never loads, so I'm not sure how to confirm that this is the same as what @jmcollin78 has reported, but after the VM reboots it pegs a CPU core and never finishes booting.

jmcollin78, 2022-01-25:
@agners  yes I follow the installation on the website. I have send you another post showing that I'm not the only one with this problem.

You talk about image compression using unxz: no I don't uncompress it before trying to install it. I was not aware of xz compression because it is not explain and xz is very not familliar. I will try this and I let you know.

jmcollin78, 2022-01-25:
Hello, good news it seems to work normally with uncompress before installation. So sorry for this mistake. May be a note in the installation documentation should be a good point because that is not so obvious.

Anyway thanks for your valuable and quick answer.

jmcollin78, 2022-01-25:
I just don't have the Supervisor icon on the menu bar, but I will look forward for a solution.

closed 2022-01-25 by jmcollin78

agners, 2022-01-25:
@jmcollin78 the supervisor panel moved, see https://www.home-assistant.io/blog/2021/12/11/release-202112/#brand-new-configuration-panel

dinomite, 2022-01-25:
Is there a way I can hand-uncompress for supervisor installation?

agners, 2022-01-25:
@dinomite not sure what you mean, on Linux you can uncompress a xz file using:

```
unxz haos_ova-7.2.qcow2.xz
```

Then you have the uncompressed `haos_ova-7.2.qcow2` in the same directory.

I am sure there are tools to do the same on other operating systems like [Windows](https://stackoverflow.com/questions/42545028/open-xz-file-from-the-command-line-in-windows-10).

agners, 2022-01-25:
If you experience a problem after using the update functionality, this is not the same problem. The original poster had issues with a new installation. Please open a new issue with all details.

dinomite, 2022-01-25:
👍 I'll do some more digging and if I can't figure it out I'll make a new issue. Thanks for the kind messages!