home-assistant/operating-system #1726: Repeated remount of file systems

borpin, 2022-01-25:
### Describe the issue you are experiencing

Repeated remount of filesystem

HA Blue.

Note this template needs updating - system info is under Configuration -> Settings - > Info (took me a while to find it) 😀 

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. View Log
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
none
```

### Anything in the Host logs that might be useful for us?

```txt
[  133.552859] ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
[  133.553232] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
[  133.553506] ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
[  133.554172] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
[  133.554449] ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
[  133.554840] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
[  133.554893] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
[  133.554930] ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
[  133.555009] ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
[  133.585580] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
[  133.585622] ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
[  133.585644] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
[  133.585688] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
[  133.585708] ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
[  133.585725] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
[  133.585742] ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
[  133.585763] ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
[  133.585785] ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
[  133.880410] audit: type=1334 audit(1643118869.648:251): prog-id=17 op=LOAD
[  133.880523] audit: type=1334 audit(1643118869.648:252): prog-id=18 op=LOAD
[  134.483253] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
[  134.483290] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
[  134.483363] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
[  134.483392] ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
[  134.483667] ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
[  134.483978] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
[  134.484371] ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
[  134.484481] ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
[  134.484721] ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
[  134.515964] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
[  134.516015] ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
[  134.516037] ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
[  134.516061] ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
[  134.516079] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
[  134.516094] ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
[  134.516113] ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
[  134.516132] ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
[  134.516165] ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
```
```

### System Health information

## System Health

version | core-2021.12.10
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.91
arch | aarch64
timezone | Europe/London

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.18.0
Stage | running
Available Repositories | 1010
Installed Repositories | 23

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | 15 February 2022, 00:00
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | false
google_enabled | true
remote_server | eu-west-2-1.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | failed to load: timeout

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.2
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 113.9 GB
disk_used | 22.6 GB
healthy | true
supported | true
board | odroid-n2
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Home Assistant Google Drive Backup (0.105.2), Samba share (9.5.1), AppDaemon 4 (0.8.0), NGINX Home Assistant SSL proxy (3.1.0), Node-RED (10.3.1), Let's Encrypt (4.12.0), Mosquitto broker (5.1.1), Grafana (7.4.0), Glances (0.14.1), AdGuard Home (4.4.2), Log Viewer (0.12.1), Home Panel (2.3.3), ESPHome (2022.1.1), Emoncms (0.2.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 12
views | 9
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2022-01-25:
Those messages are triggered whenever an Add-on or the Core gets restartet or the time date service gets queries. They are not harmful, they just warn that some file systems don't have full 64-bit time stamp support (see #1702).

I'll update the issue template, thanks for mentioning!

borpin, 2022-01-26:
> Those messages are triggered whenever an Add-on or the Core gets restartet

Ok, thanks.

> I'll update the issue template, thanks for mentioning!

No bother 😄

closed 2022-01-26 by borpin