home-assistant/operating-system #1728: update to Home Assistant OS 7.2 stopped at "Waiting for Supervisor to startup..."

glc650, 2022-01-25:
### Describe the issue you are experiencing

After upgrading to Home Assistant OS 7.2 (from 7.1) my system hung at:
[image]

I waited for ~15 minutes and then did a hard reboot.  The update appears to have been successful but this is the first issue I've ever had with an OS update.

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Click on the update button.

### Anything in the Supervisor logs that might be useful for us?

```txt
couldn't check logs at the time
```

### Anything in the Host logs that might be useful for us?

```txt
couldn't check logs at the time
```

agners, 2022-01-26:
Did you have internet connection at that time? Was the second  boot successful?

glc650, 2022-01-26:
Yes to both.

agners, 2022-01-27:
I need the debug logs from the underlying OS (e.g. obtained via direct SSH connection using https://developers.home-assistant.io/docs/operating-system/debugging#ssh-access-to-the-host). To get all logs from the previous boot use `journalctl -b -1`.

glc650, 2022-01-27:
I've rebooted a few times since then to make sure all was good.  Can I still get the logs you need?

agners, 2022-01-27:
So the `-b -1` says "last reboot". You can use higher values, until you get the logs of the problematic boot. There is also `journalctl --list-boots` which lists boot. The log buffer is quite large, it should contain logs of a week+.

closed 2022-05-04 by github-actions[bot]