home-assistant/operating-system #1733: `ha os datadisk list` reports other disk than `ha os info`

mdeweerd, 2022-01-31:
### Describe the issue you are experiencing

`ha os datadisk list` reports other disk than `ha os info`.

See results of those commands:

```
[core-ssh ~]$ ha os datadisk list
devices:
- /dev/mmcblk0
[core-ssh ~]$ ha os info
board: generic-x86-64
boot: A
data_disk: /dev/sda8
update_available: true
version: "7.1"
version_latest: "7.2"
```

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

Access the system using ssh.

Execute:

- `ha os datadisk list` 
- `ha os info`.

Compare the results.

Must possibly have (unused) internal disk like /dev/mmcblk* + sata-disk like setup.

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

### System Health information

## System Health

version | core-2021.12.10
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.88
arch | x86_64
timezone | Europe/Paris

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4489
Installed Version | 1.20.0
Stage | running
Available Repositories | 970
Downloaded Repositories | 28

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.1
-- | --
update_channel | stable
supervisor_version | supervisor-2021.12.2
docker_version | 20.10.9
disk_total | 292.7 GB
disk_used | 12.1 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Check Home Assistant configuration (3.9.0), Mosquitto broker (6.0.1), File editor (5.3.3), Custom deps deployment (1.3.1), SQLite Web (3.2.1), Grafana (7.4.1), Samba Backup (5.0.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 9
-- | --
resources | 10
views | 21
mode | storage

</details>

agners, 2022-02-02:
`ha os datadisk list` lists the disks which can be used as a data disk (so potential data disk candidates).

`ha os info` shows the disk (and partition) currently in use as data disk.

closed 2022-02-02 by agners

mdeweerd, 2022-02-02:
The disk that is currently used is not in the potential list of disks ... .

agners, 2022-02-02:
Yes, because you can't move to it. Think of it as "potential new targets".

mdeweerd, 2022-02-02:
I understood the justification you want to give to it, but it's not the logic that I expect as a user/technical person.

Anyway, this leaves an opening to adapt the documentation to the reality and at the same time fix 'getting' into 'get' and replace part of the sentence with 'potential new targets': https://www.home-assistant.io/common-tasks/os/#using-cli-to-move-the-data-partition .

agners, 2022-02-04:
https://github.com/home-assistant/home-assistant.io/pull/21490