home-assistant/operating-system #1734: Conbee II Stick disconnecting

Prior99, 2022-01-31:
### Describe the issue you are experiencing

I am using the Conbee II Stick on Home Assistant OS on a normal x86_64 machine with a decent CPU und a decent PSU.

Sometimes the stick will run smoothely for days, sometimes only for minutes, until it will disconnect and then reconnect periodically.

Yes, I am using an extension cord.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Connect Conbee II Stick To Home Assistant OS
2. Use official Deconz Addon
3. Wait
...

### Anything in the Supervisor logs that might be useful for us?

```txt
22-01-31 11:04:09 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/010 - None
22-01-31 11:04:18 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/011 - None
22-01-31 11:04:18 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/011 - None
22-01-31 11:04:25 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/012 - None
22-01-31 11:04:29 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/012 - None
22-01-31 11:04:35 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/013 - None
22-01-31 11:04:39 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/013 - None
22-01-31 11:04:49 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/014 - None
22-01-31 11:04:49 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/014 - None
22-01-31 11:04:56 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/015 - None
22-01-31 11:04:59 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/003/015 - None
22-01-31 11:05:10 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.ADD hardware /dev/bus/usb/003/016 - None
```

### Anything in the Host logs that might be useful for us?

```txt
[    0.000000] Linux version 5.10.88 (builder@63b64195e937) (x86_64-buildroot-linux-gnu-gcc.br_real (Buildroot -g3cedcad6) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP Wed Dec 29 17:18:51 UTC 2021
[    0.000000] Command line: apparmor=1 security=apparmor console=tty1 bootchooser.active=A  root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rootwait systemd.machine_id=d5af13ea309f435fb470dc6b4fa035de zram.enabled=1 zram.num_devices=3
[    0.000000] x86/fpu: Supporting XSAVE feature 0x001: 'x87 floating point registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x002: 'SSE registers'
[    0.000000] x86/fpu: Supporting XSAVE feature 0x004: 'AVX registers'
[    0.000000] x86/fpu: xstate_offset[2]:  576, xstate_sizes[2]:  256
[    0.000000] x86/fpu: Enabled xstate features 0x7, context size is 832 bytes, using 'compacted' format.
[    0.000000] BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000009ffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000000a0000-0x00000000000fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x0000000009e01fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000009e02000-0x0000000009ffffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000a000000-0x000000000a1fffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000a200000-0x000000000a20afff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x000000000a20b000-0x000000000affffff] usable
[    0.000000] BIOS-e820: [mem 0x000000000b000000-0x000000000b01ffff] reserved
[    0.000000] BIOS-e820: [mem 0x000000000b020000-0x00000000eb614fff] usable
[    0.000000] BIOS-e820: [mem 0x00000000eb615000-0x00000000eb72dfff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000eb72e000-0x00000000eb8aefff] usable
[    0.000000] BIOS-e820: [mem 0x00000000eb8af000-0x00000000ebcbffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000ebcc0000-0x00000000ecba8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ecba9000-0x00000000eeffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000ef000000-0x00000000efffffff] reserved
[... 1251 lines cut ...]
version_api | ok
installed_addons | deCONZ (6.11.1), File editor (5.3.3), Terminal & SSH (9.3.0), BorgBackup (0.56), NGINX Home Assistant SSL proxy (3.1.1), Let's Encrypt (4.12.0), Portainer (2.0.0), Mosquitto broker (6.0.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 13
-- | --
resources | 19
views | 26
mode | yaml

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

Please note that the Host Logs are from a different boot.

Prior99, 2022-01-31:
I think this might be related to this issue: https://github.com/home-assistant/operating-system/issues/1633

Prior99, 2022-01-31:
Here is a chunk of relevant log from the Host after this happened again:

```
[ 7558.718488] usb 3-4: USB disconnect, device number 3
[ 7558.983570] usb 3-4: new full-speed USB device number 4 using xhci_hcd
[ 7559.110829] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[ 7559.110831] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 7559.110832] usb 3-4: Product: ConBee II
[ 7559.110833] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[ 7559.110834] usb 3-4: SerialNumber: DE2405775
[ 7569.121432] xhci_hcd 0000:0a:00.3: Abort failed to stop command ring: -110
[ 7569.141643] xhci_hcd 0000:0a:00.3: xHCI host controller not responding, assume dead
[ 7569.141656] xhci_hcd 0000:0a:00.3: HC died; cleaning up
[ 7569.141763] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[ 7569.141772] usb 3-4: can't set config #1, error -62
[ 7569.141881] usb 3-1: USB disconnect, device number 2
[ 7569.141979] ftdi_sio ttyUSB0: error from flowcontrol urb
[ 7569.142240] ftdi_sio ttyUSB0: FTDI USB Serial Device converter now disconnected from ttyUSB0
[ 7569.142254] ftdi_sio 3-1:1.0: device disconnected
[ 7569.142438] usb 3-4: USB disconnect, device number 4
[ 7715.047386] kauditd_printk_skb: 16 callbacks suppressed
```

gieljnssns, 2022-02-01:
I'm still seeing the same even after disabling BT as suggested in #1633, I even disabled wifi.

```
[   64.125673] eth0: renamed from veth27218aa
[   64.138943] IPv6: ADDRCONF(NETDEV_CHANGE): veth5710211: link becomes ready
[   64.138970] hassio: port 8(veth5710211) entered blocking state
[   64.138972] hassio: port 8(veth5710211) entered forwarding state
[   66.225519] usb 1-3: USB disconnect, device number 5
[   66.529410] usb 1-3: new full-speed USB device number 6 using xhci_hcd
[   66.658372] usb 1-3: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[   66.658376] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   66.658379] usb 1-3: Product: ConBee II
[   66.658381] usb 1-3: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[   66.658383] usb 1-3: SerialNumber: DE2215018
[   66.659558] cdc_acm 1-3:1.0: ttyACM0: USB ACM device
[   83.413973] kauditd_printk_skb: 1 callbacks suppressed
[   83.413974] audit: type=1334 audit(1643639822.963:178): prog-id=14 op=UNLOAD
[   83.413978] audit: type=1334 audit(1643639822.963:179): prog-id=13 op=UNLOAD
[   85.725137] audit: type=1334 audit(1643639825.274:180): prog-id=12 op=UNLOAD
[   85.725143] audit: type=1334 audit(1643639825.274:181): prog-id=11 op=UNLOAD
[   95.784499] hassio: port 9(veth6a32fd8) entered blocking state
[   95.784502] hassio: port 9(veth6a32fd8) entered disabled state
[   95.784628] device veth6a32fd8 entered promiscuous mode
[   95.784848] hassio: port 9(veth6a32fd8) entered blocking state
[   95.784850] hassio: port 9(veth6a32fd8) entered forwarding state
```

Prior99, 2022-02-01:
I think this kernel bug could be the reason: https://bugzilla.kernel.org/show_bug.cgi?id=202541

Prior99, 2022-02-02:
Here is another chunk of kernel log:

```
[142420.753693] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142420.753709] usb 3-4: can't set config #1, error -62
[142421.218564] usb 3-4: USB disconnect, device number 21
[142421.482455] usb 3-4: new full-speed USB device number 22 using xhci_hcd
[142421.609794] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142421.609799] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142421.609802] usb 3-4: Product: ConBee II
[142421.609804] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142421.609806] usb 3-4: SerialNumber: DE2405775
[142431.521691] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142431.521705] usb 3-4: can't set config #1, error -62
[142431.522762] usb 3-4: USB disconnect, device number 22
[142431.759246] usb 3-4: new full-speed USB device number 23 using xhci_hcd
[142431.887000] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142431.887005] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142431.887007] usb 3-4: Product: ConBee II
[142431.887009] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142431.887011] usb 3-4: SerialNumber: DE2405775
[142441.089699] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142441.089712] usb 3-4: can't set config #1, error -62
[142441.610563] usb 3-4: USB disconnect, device number 23
[142441.866371] usb 3-4: new full-speed USB device number 24 using xhci_hcd
[142441.993383] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142441.993387] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142441.993390] usb 3-4: Product: ConBee II
[142441.993392] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142441.993394] usb 3-4: SerialNumber: DE2405775
[142451.889703] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142451.889714] usb 3-4: can't set config #1, error -62
[142451.889816] usb 3-4: USB disconnect, device number 24
[142452.126837] usb 3-4: new full-speed USB device number 25 using xhci_hcd
[142452.254110] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142452.254112] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142452.254113] usb 3-4: Product: ConBee II
[142452.254114] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142452.254115] usb 3-4: SerialNumber: DE2405775
[142461.441723] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142461.441738] usb 3-4: can't set config #1, error -62
[142462.013124] usb 3-4: USB disconnect, device number 25
[142462.266672] usb 3-4: new full-speed USB device number 26 using xhci_hcd
[142462.393944] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142462.393948] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142462.393951] usb 3-4: Product: ConBee II
[142462.393953] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142462.393954] usb 3-4: SerialNumber: DE2405775
[142472.305704] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142472.305723] usb 3-4: can't set config #1, error -62
[142472.305889] usb 3-4: USB disconnect, device number 26
[142472.544440] usb 3-4: new full-speed USB device number 27 using xhci_hcd
[142472.671827] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142472.671832] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142472.671835] usb 3-4: Product: ConBee II
[142472.671837] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142472.671838] usb 3-4: SerialNumber: DE2405775
[142481.905725] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142481.905740] usb 3-4: can't set config #1, error -62
[142482.410314] usb 3-4: USB disconnect, device number 27
[142482.667269] usb 3-4: new full-speed USB device number 28 using xhci_hcd
[142482.794607] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142482.794612] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142482.794614] usb 3-4: Product: ConBee II
[142482.794616] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142482.794618] usb 3-4: SerialNumber: DE2405775
[142492.737743] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142492.737757] usb 3-4: can't set config #1, error -62
[142492.737886] usb 3-4: USB disconnect, device number 28
[142492.975035] usb 3-4: new full-speed USB device number 29 using xhci_hcd
[... 143 lines cut ...]
[142656.222902] usb 3-4: new full-speed USB device number 45 using xhci_hcd
[142656.350639] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142656.350644] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142656.350646] usb 3-4: Product: ConBee II
[142656.350648] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142656.350650] usb 3-4: SerialNumber: DE2405775
[142665.537846] xhci_hcd 0000:0a:00.3: Timeout while waiting for configure endpoint command
[142665.537861] usb 3-4: can't set config #1, error -62
[142666.122843] usb 3-4: USB disconnect, device number 45
[142666.378742] usb 3-4: new full-speed USB device number 46 using xhci_hcd
[... cut ...]
[142676.433983] usb 3-4: can't set config #1, error -62
[142676.434125] usb 3-4: USB disconnect, device number 46
[142676.671515] usb 3-4: new full-speed USB device number 47 using xhci_hcd
[142676.799181] usb 3-4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[142676.799186] usb 3-4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[142676.799188] usb 3-4: Product: ConBee II
[142676.799190] usb 3-4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[142676.799192] usb 3-4: SerialNumber: DE2405775
```

agners, 2022-02-02:
> I think this kernel bug could be the reason: https://bugzilla.kernel.org/show_bug.cgi?id=202541

Maybe, it seems to me though that more errors appear in their case?

To me it seems that `xHCI host controller not responding, assume dead` is the main culprit, but maybe there is more to it. Googling a bit it seems that others have a very similar issue with AMD/Ryzen based systems:
https://unix.stackexchange.com/questions/669015/xhci-restart-needed

I think I've also heard from other places where some AMD systems had USB issues. Can you try a BIOS update? Also just to make sure, did you try use a different extension cord?

Have you tried other Linux distributions with the same system?

I plan to update the Linux kernel to the latest stable 5.10 version with 7.3, and then also update to the latest LTS on dev. Both variants probably worth testing, maybe something upstream fixes it.

Prior99, 2022-02-07:
You are right. I changed to a port with another controller and the problem hasn't happened since. Closing this. Thanks :)

closed 2022-02-07 by Prior99

gieljnssns, 2022-02-07:
@Prior99 
Can you exactly tell me what you did? I have tried all my USB ports. I'm using an Intel® nUC KIt nUC8¡7BeH