home-assistant/operating-system #1737: Home assistant 7.2 not working in Virtual box

DJAMIRSAM, 2022-02-03:
### The problem

i cant get home assistant to work in  a windows VM after the update
i keep getting the error https://cdn.discordapp.com/attachments/705503647135432754/937918802132357130/unknown.png
I have had HA working in this computer before with the same method, i have restarted it and re installed it but nothing works
i have a backup so im fine with deleting stuff
pls ping me when responding i have this server set to ping only

### What version of Home Assistant Core has the issue?

The one off the website (7.2)

### What was the last working version of Home Assistant Core?

7.1

### What type of installation are you running?

Home Assistant OS

### Integration causing the issue

None

agners, 2022-02-03:
@DJAMIRSAM can you try OS 7.3 on beta channel? It comes with a new stable release Linux kernel, maybe something was buggy with the kernel in OS 7.2.

To enable beta channel, use

```
ha su options --channel beta
ha su reload
ha os update
```

DJAMIRSAM, 2022-02-04:
Is there anyone with the beta .vdi file?

agners, 2022-02-04:
@DJAMIRSAM you can find images of the latest beta here: https://github.com/home-assistant/operating-system/releases/tag/7.3

DJAMIRSAM, 2022-02-06:
The boot up worked fine but after boot up everything when back to the way it was (7.3) and there is no backup

DJAMIRSAM, 2022-02-07:
[image]
This is what it shows after leaving it on for 5 hours

agners, 2022-02-07:
What VirtualBox version are you using? Can you also share the virtual machine settings?

malosaa, 2022-02-07:
i had this problem one time aswell, it was my hdd and changed it and worked again.
but for you it could be something else, but it looks like its to slow to load, like i had.

@DJAMIRSAM try on a other PC/Server and see if that works.
Or try loading the vdi on a other hdd.

Be sure u have this enabled in virtualbox:
Enable EFI
and use atleast 3gb of memory.
enable 2x cpu or higher
Execution cap 100%
Just for now disable usb controller in virtualbox

and try again, that helped me before

DJAMIRSAM, 2022-02-07:
> What VirtualBox version are you using? Can you also share the virtual machine settings?

It’s 6.1.32
[image]

DJAMIRSAM, 2022-02-07:
> i had this problem one time aswell, it was my hdd and changed it and worked again. but for you it could be something else, but it looks like its to slow to load, like i had.
>
> @DJAMIRSAM try on a [...]

how can it be a HDD issue? it says cpu stall 
[image]
im gonna try it anyways

agners, 2022-02-08:
@DJAMIRSAM RCU callbacks are executed periodically, stalls are reported when those callbacks are not executed for quite a while for some reason. The culprit can be many things, e.g. software bugs but also very slow/stuck hardware (e.g. SD cards/HDDs). In virtualization it could also be that the timing/timers are off/not working correctly.

This is a bit a shot in the dark, but can you try different "Paravirtualization Interface" settings (in System -> Acceleration)?

There have been other reports of rcu_sched stalls like this issue https://www.virtualbox.org/ticket/20131. Probably worth posting in the VirtualBox forums (ideally attach the VirtualBox logs), they might have some hints what could be the culprit.

malosaa, 2022-02-08:
I found the issue, a guy had exactly the same problem with hass on virtualbox

https://github.com/home-assistant/operating-system/issues/1705

read this and it should be fixed.
@DJAMIRSAM

DJAMIRSAM, 2022-02-08:
ok so it kinda worked but after a restart its back to freezing but i had to restart due to freezing

DJAMIRSAM, 2022-02-08:
its stuck on 
[image]

malosaa, 2022-02-09:
> its stuck on [image]

Did u try on an other machine ? if i look at the error you describe i see many people with hdd errors/memory errors.
Otherwise download an older version and see if its gone..

Its odd this also doesnt work on a other machine, i have many versions running, none of them have that issue, but i have installed the extra additions to it (the addons)

1 machine is a 8gb quadcore  thin client (ssd 120gb) with windows 10 pro  (virtualbox with 7.0 Hass)
1 machine a 16gb Quadcore Laptop with ssd on windows 10 pro   (virtualbox 7.1 HASS)
1 machine 16gb laptop quadcore  with ssd (250gb) windows 11. (virtualbox 7.3 HASS)

all no problems, i wish i could help u but i have no idea how

regards

DJAMIRSAM, 2022-02-09:
So for now it seems to be working, I fixed it by deleting docker desktop from the pc an setting the accelerator to kvm but if it continues I will try that. I currently have an issue which is a VMBOX issue but since this exists I might as swell put it out there. The usb connects keeps disconnecting (I have 2 and they both do it at the same time)

degreal, 2022-02-13:
I updated windows and virtualbox 6.1.30 -> 6.1.32 at same time.
Zigbee USB dongle stopped working. "self-detected stall on CPU" error spamming. Virtualbox taking 30 min to get up, "Journal" errors.

I did fresh install but still getting stall errors.
I downgraded VirtualBox back to 6.1.30 and everything is working now!

DJAMIRSAM, 2022-02-13:
So fun fact the issue came back today so this reply came in clutch, I’m gonna downgrade vbox

degreal, 2022-02-14:
> So fun fact the issue came back today so this reply came in clutch, I’m gonna downgrade vbox

Nevermind, Today I started getting stall errors again. Downgrading to 6.1.30 does not seem to help perfectly.

DJAMIRSAM, 2022-02-14:
> > So fun fact the issue came back today so this reply came in clutch, I’m gonna downgrade vbox
> 
> Nevermind, Today I started getting stall errors again. Downgrading to 6.1.30 does not seem to help perfectly.

Try setting the paravirtualization as mentioned above to KVM it seems to work the best right now

frederikrooms, 2022-02-17:
> Try setting the paravirtualization as mentioned above to KVM it seems to work the best right now

@degreal 
Did the paravirtualization change to KVM work for you? I'm having the same problem with the exact setup, and I do hope the downgrade to 6.30 works. I'm sceptical because earlier today I ran into an USB problem (OHI-PCI problem) while running 6.30. So neither of the options seems to provide me a smooth working Home Assistant. It all seems very messy with the update strategy this year, I hope my monthly fee will provide a more stable platform in the future...

degreal, 2022-02-18:
> @degreal Did the paravirtualization change to KVM work for you? I'm having the same problem with the exact setup, and I do hope the downgrade to 6.30 works. I'm sceptical because earlier today I ran [...]

No it didn't. "6.30+KVM" combo fails once every 3 days. I would test "6.32+KVM" but I cannot have downtime right now, lot is riding on this setup.

DJAMIRSAM, 2022-02-18:
Let me clarify something, I does not need to be kvm, you can change it to something else, kvm is the one what worked for me but every computer is different

SilentT-FR, 2022-04-22:
Same issue in 6.1.32

DJAMIRSAM, 2022-04-22:
> Same issue in 6.1.32

Try setting the paravirtualization as mentioned above to KVM it seems to work the best right now.
Let me clarify something, I does not need to be kvm, you can change it to something else, kvm is the one what worked for me but every computer is different

SilentT-FR, 2022-04-25:
Thanks for the tips
For me the KVM not work but the minimum works

closed 2022-07-31 by github-actions[bot]