home-assistant/operating-system #1739: HassOS 7.3 does not boot on Intel NUC (i5-10210U)

nta, 2022-02-04:
### Describe the issue you are experiencing

After upgrading to HassOS 7.3 on the i5-10210U NUC, boot would hang at a random point each time without taking keyboard input (once upon loading `i915` frame buffer, once at a `Mounted systemd ...`).

A few failed boots in, it rolled itself back to 7.2, and it boots again.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Use a 10th Gen Intel NUC with 7.2 installed.
2. Upgrade to 7.3.
3. Notice the device fails to boot.

### Anything in the Supervisor logs that might be useful for us?

```txt
N/A
```

### Anything in the Host logs that might be useful for us?

```txt
N/A
```

### Additional information

Found one Google result regarding newer Linux 5.10 LTS kernels hanging during boot on Intel systems: https://forum.manjaro.org/t/after-updating-to-kernel-5-10-93-from-5-10-89-the-computer-freezes-at-lenovo-logo/100852

agners, 2022-02-04:
> A few failed boots in, it rolled itself back to 7.2, and it boots again. Yay for A/B installs!

Good to hear that this helped here!

Hm, I have a `NUC7i3DNHE` which I think is a 7th Gen NUC, that successfully updated. That link seems  not to crash randomly though.

So it would completely hang? Did you had to reset the system manually then?

nta, 2022-02-04:
> So it would completely hang? Did you had to reset the system manually then?

Correct. A bit annoying indeed to not have a watchdog.

agners, 2022-02-05:
Does that system happen to have a BNX2/BNX2X firmware? Besides the move from 5.10.91 to 5.10.96 stable kernel release that is the only change.

Do you have external devices connected? Disconnecting everything and see if it comes up without external devices might help isolate in case the problem is triggered by one of those devices.

When you are in OS 7.2, and trigger an OS update again, it will reapply 7.3 to the other partition. So it should be safe to retry as you will be able to go back again with three reboots. In any case. probably worth taking a backup and downloading it before experimenting :sweat_smile:

nta, 2022-02-05:
I can check again later this weekend, hopefully. No Broadcom network cards
attached, no, which indeed would lead to a bad patch in the Linux kernel,
but I find it weird that a LTS kernel would get a breaking change
backported like that.

agners, 2022-02-07:
> but I find it weird that a LTS kernel would get a breaking change
> backported like that.

Most likely it wouldn't be a intentional break. Unfortunately we had breakages before in LTS releases... (USB serial adapters broke a couple of releases ago.

[v5.10.98](https://cdn.kernel.org/pub/linux/kernel/v5.x/ChangeLog-5.10.98) released on Saturday, but not fixes in that direction.

Since revert seems to work in that case, I am going ahead an promote 7.3 to stable still. We can selectively revert if more reports like this come in.

agners, 2022-02-07:
@nta what exact model is that?

JackEVE, 2022-02-07:
Same problem on a nuc NUC10i7FNH running proxmox.
Seems to be boot looping and maxing out one of 2 cpus

Espressoshooter, 2022-02-07:
Same problem for me generic-x86-64.

agners, 2022-02-07:
> Same problem on a nuc NUC10i7FNH running proxmox. Seems to be boot looping and maxing out one of 2 cpus.
> 
> I've been letting in run for a good 10 minutes now without recovering.

I assume HAOS is the virtual guest here? That is somewhat surprising. Are you passing through lots of devices?

JackEVE, 2022-02-07:
> > Same problem on a nuc NUC10i7FNH running proxmox. Seems to be boot looping and maxing out one of 2 cpus.
> > I've been letting in run for a good 10 minutes now without recovering.
> 
> I assume HAOS is the virtual guest here? That is somewhat surprising. Are you passing through lots of devices?

Yes, HAOS is the guest. 
I'm passing a Z-wave stick (Aeotec gen5) and an Arduino (as a MySensors Gateway) through to homeassistant.
I installed proxmox using the standard debian iso and then created a HA VM using this script here: [https://github.com/tteck/Proxmox#-select-a-proxmox-helper-below-](https://github.com/tteck/Proxmox#-select-a-proxmox-helper-below-)
I haven't tried running that script with the new HAOS version, yet.

robertboccia, 2022-02-07:
getting the same issue on my nuc

agners, 2022-02-07:
@robertboccia is it also 10th gen?

robertboccia, 2022-02-07:
> @robertboccia is it also 10th gen?

bxnuc10i7fnh

agners, 2022-02-07:
I removed 7.3 from the stable stream for `generic-x86-64` for now.

timborino, 2022-02-07:
BXNUC10i3FNH1 wouldn't boot after update from 7.2 to 7.3.

Nortek HUSBZB-1, Google Coral, and APC UPS are the only devices connected to the NUC.

I ran the update from the GUI this morning and the system never came back online. Screenshot 1  is how I found the system after attaching a monitor. Keyboard was unresponsive and I had to hold power button down to cycle power. [image]

System locked up again (screenshot 2).[image]

Powered up again with all USB devices disconnected except the keyboard. System locked up again (screenshot 3).[image]

Cycled power again and system came back online with version 7.2.

tomlut, 2022-02-07:
Thanks timborino. I was stuck at the same NTP service. I disconnected my UPS (only USB device other than the keyboard) and after two restarts finally got past this:

[image]

Not sure if it was the number of restarts or disconnecting the UPS but my system rolled back to 7.2 as well. 

NUC11PAHi7

nevir, 2022-02-08:
Similar here with a 10i5FNK1.  Was able to revert back to 7.2 after a few failed boots

agners, 2022-02-08:
@JackEVE do you pass through the WiFi card by chance? If so, your problem is not related to this issue and warrants a separate issue.

Does using 7.2 fixes the problem for you?

JackEVE, 2022-02-08:
> @JackEVE do you pass through the WiFi card by chance? If so, your problem is not related to this issue and warrants a separate issue.
> 
> Does using 7.2 fixes the problem for you?

I'm using a network bridge. 
7.2 works fine.  The issue only showed up when I updated from 7.2 to 7.3

agners, 2022-02-08:
This issue is now fixed with HAOS 7.4 (currently on beta channel). If some of you can confirm that this version works, I'll promote it to stable on Thursday or so.

JackEVE, 2022-02-09:
For me 7.4 is still freezing during boot. Hast to be a different issue in my case, since I am not passing through the wifi card so the fix shouldn't work anyway.

[image]
[image]
[image]

robertboccia, 2022-02-09:
I was not using Wi-Fi either, mine is on lan via ethernet

[hassos74_freeze]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F18036705%2F153139743-88e6d094-c0f8-4ce1-b351-cc63c80e1b74.PNG&data=04%7C01%7C%7C3d14ccb83caa4a2581ff08d9eb9bc04a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637799876440527994%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=%2BWzL9ZQLE9LAPXmNKctYuNatfz08CiUfyZTW0FwjvHU%3D&reserved=0>
[hassos74_freeze_hardware]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F18036705%2F153140089-5c27a1ce-c797-45c2-9265-c3064a231816.PNG&data=04%7C01%7C%7C3d14ccb83caa4a2581ff08d9eb9bc04a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637799876440527994%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=Vcg%2FrTafSgYy3ijhLkMeqx1awto%2B27UZLUvG1e2tHkU%3D&reserved=0>
[hassos74_freeze_options]<https://emea01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F18036705%2F153140094-fb6816fa-69ce-4e98-a2c7-afd22742babf.PNG&data=04%7C01%7C%7C3d14ccb83caa4a2581ff08d9eb9bc04a%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637799876440527994%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000&sdata=dRK7qlP2jbplhtCSYXU8eHh8EQ7LLXp2aEtmph%2FMzmg%3D&reserved=0>

-

nta, 2022-02-09:
7.4 boots fine for me again, yes.

agners, 2022-02-10:
The problem was caused by WiFi even when not in use. Thanks for the positive reports.

@JackEVE your problem still seems different, but I also bumped the stable kernel version, so maybe something else fixed it :man_shrugging:

closed 2022-02-10 by agners