home-assistant/operating-system #1740: HASS Becomes Unresponsive Nearly Every Day

Kackerlacka, 2022-02-06:
### Describe the issue you are experiencing

Nearly once a day (but sometimes doesn't happen for a week+), HASS will become completely unresponsive locally and remotely through Nabu Casa. It usually comes back on its own after a couple of hours more often than not, but sometimes it never does and I have to restart it manually by unplugging/replugging my RPi4 back in. 

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

It happens randomly with no known cause or a specific time.

### Anything in the Supervisor logs that might be useful for us?

```txt
Supervisor logs will not load at all, "Bad Gateway 502" error.
```

### Anything in the Host logs that might be useful for us?

```txt
https://pastebin.com/nCmXEGfw
```

### System Health information

## System Health

version | core-2021.12.10
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.63-v8
arch | aarch64
timezone | America/Los_Angeles

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4526
Installed Version | 1.21.0
Stage | running
Available Repositories | 988
Downloaded Repositories | 40

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | February 27, 2022, 4:00 PM
relayer_connected | true
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | false
remote_server | us-west-2-0.ui.nabu.casa
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 219.4 GB
disk_used | 19.6 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.105.2), File editor (5.3.3), Samba share (9.5.1), motionEye (0.17.0), Terminal & SSH (9.3.0), Photons Interactor (0.12.0), Grafana (7.4.1), InfluxDB (4.3.0), Z-Wave JS (0.1.52), Mosquitto broker (6.0.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 25
views | 14
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

### Additional information

My system has become unresponsive every day the past 3 days. The logs attached unfortunately only show today's "crash" at around 20:00 on 02/05/22, however I've attached a pastebin link below for the crash that happened yesterday around 14:00. 

https://pastebin.com/qwdenNcV

I am running HASS.io 64 bit on an RPi4b 2GB with a 3.5A power supply, an unpowered SSD, and a Zigbee/Zwave dongle. I've had this issue for over a year now, long before I had an SSD or Zigbee dongle (I had a plain SD card back then), I just don't know exactly which version of Home Assistant this started.

agners, 2022-02-07:
It could be related to memory, can you enable system monitoring (specifically memory): https://www.home-assistant.io/integrations/systemmonitor/

Kackerlacka, 2022-02-07:
I do have System Monitor enabled with both Memory Percent and Memory Use. Here's a Grafana graph I have that shows my memory usage percent over the last 7 days. 

[image]

agners, 2022-02-07:
There is a drop around 14:00 on February 4th, so it could be that the OOM killer kicked in at that point.

You can get the logs from the previous boot on the OS level shell. You need to use the terminal and login as root, then type login to get to the shell, or use SSH access (see https://developers.home-assistant.io/docs/operating-system/debugging how to enable this).

From there, you can use `journalctl`, e.g. `journalctl -b -1` gets all the logs of the previous boot, or `journalctl --since "22-02-04 14:00:00" --until "22-02-04 15:30:00"` gets the logs from the problematic time window.

Kackerlacka, 2022-02-07:
Unfortunately I'm getting a "Can't Read Supervisor Data" error when I try to look at my addons, so I can't access the Terminal & SSH addon at all. I've also restarted HA many times since the last crash, so if it's only the "previous boot" logs then they're probably gone. I will keep trying to access them though and I will try to post them if/when I can.

agners, 2022-02-08:
@Kackerlacka you can't run this command on the Terminal add-on, you need to get access on the underlying operating system as explained in the link posted above.

Kackerlacka, 2022-02-08:
I followed all the steps in the above link with the authorized_codes file and imported it into HA, but all I get is an "Error: Connection Refused" when trying to SSH into HA through port 22222 using PuTTY. I've been at this for quie a few hours now but I will keep poking at it to try to get it working...

agners, 2022-02-08:
@Kackerlacka you need to set the private key in the authentication settings (see [https://devops.ionos.com/tutorials/use-ssh-keys-with-putty-on-windows/#connect-to-server-with-private-key](https://devops.ionos.com/tutorials/use-ssh-keys-with-putty-on-windows/#connect-to-server-with-private-key)).

Kackerlacka, 2022-02-09:
@agners It works! Thank you, I was struggling with that for many, many hours. Now that I'm in root, here are the logs specifically between 14:00:00 and 15:30:00 on 22-02-04:

```
# journalctl --since "22-02-04 14:00:00" --until "22-02-04 15:30:00"
-- Journal begins at Wed 2022-01-19 04:14:37 UTC, ends at Wed 2022-02-09 02:56:0  3 UTC. --
Feb 04 14:24:01 homeassistant dbus-daemon[116]: [system] Activating via systemd:   service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.s  ervice' requested by ':1.803' (uid=0 pid=560516 comm="python3 -m supervisor ")
Feb 04 14:24:03 homeassistant dbus-daemon[116]: [system] Successfully activated   service 'org.freedesktop.hostname1'
Feb 04 14:24:03 homeassistant dbus-daemon[116]: [system] Activating via systemd:   service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.s  ervice' requested by ':1.805' (uid=0 pid=560516 comm="python3 -m supervisor ")
Feb 04 14:24:04 homeassistant dbus-daemon[116]: [system] Successfully activated   service 'org.freedesktop.timedate1'

```

I'm not sure if this affects anything, but HA is saying these journal entries are in UTC, but I'm physically in the PST timezone, so I imagine if something happened at 2:00 PM PST then I should be scanning for 10:00 PM UTC no? Either way, there's "no entries" found at the problematic (but timezone-corrected) time.

However, fortunately (and unfortunately...) my system crashed again today so here are more logs from today's crash at 6:00 PM PST:

```
# journalctl --since "22-02-08 18:00:00" --until "22-02-08 19:00:00"
-- Journal begins at Wed 2022-01-19 04:14:37 UTC, ends at Wed 2022-02-09 03:02:05 UTC. --
Feb 08 18:43:31 homeassistant dbus-daemon[120]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.24' (uid=0 pid=1293 comm="python3 -m supervisor ")
Feb 08 18:43:32 homeassistant dbus-daemon[120]: [system] Successfully activated service 'org.freedesktop.hostname1'
Feb 08 18:43:32 homeassistant dbus-daemon[120]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.26' (uid=0 pid=1293 comm="python3 -m supervisor ")
Feb 08 18:43:33 homeassistant dbus-daemon[120]: [system] Successfully activated service 'org.freedesktop.timedate1'

```
But once again, still not sure about the timezone so here's the PST to UTC corrected version (18:00:00 PST to 02:00:00 UTC the next day):

```
# journalctl --since "22-02-09 00:00:00" --until "22-02-09 02:45:00"
-- Journal begins at Wed 2022-01-19 04:14:37 UTC, ends at Wed 2022-02-09 03:13:06 UTC. --
Feb 09 01:03:40 homeassistant dbus-daemon[120]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.24' (uid=0 pid=1293 comm="python3 -m supervisor ")
Feb 09 01:03:42 homeassistant dbus-daemon[120]: [system] Successfully activated service 'org.freedesktop.hostname1'
Feb 09 01:03:42 homeassistant dbus-daemon[120]: [system] Activating via systemd: service name='org.freedesktop.timedate1' unit='dbus-org.freedesktop.timedate1.service' requested by ':1.26' (uid=0 pid=1293 comm="python3 -m supervisor ")
Feb 09 01:03:43 homeassistant dbus-daemon[120]: [system] Successfully activated service 'org.freedesktop.timedate1'
-- Boot 68545b36f5104993b8d4e7feaaa10680 --
Oct 12 15:24:07 homeassistant systemd-timesyncd[356]: System clock time unset or jumped backwards, restoring from recorded timestamp: Wed 2022-02-09 01:26:01 UTC
Feb 09 01:26:01 homeassistant systemd[1]: Started Network Time Synchronization.
Feb 09 01:26:01 homeassistant kernel: audit: type=1334 audit(1644369961.071:4): prog-id=7 op=LOAD
Feb 09 01:26:01 homeassistant kernel: audit: type=1334 audit(1644369961.071:5): prog-id=8 op=LOAD
Feb 09 01:26:01 homeassistant audit: BPF prog-id=7 op=LOAD
Feb 09 01:26:01 homeassistant audit: BPF prog-id=8 op=LOAD
Feb 09 01:26:01 homeassistant systemd-time-wait-sync[123]: adjtime state 5 status 40 time Wed 2022-02-09 01:26:01.000853 UTC
Feb 09 01:26:01 homeassistant systemd[1]: Reached target System Initialization.
Feb 09 01:26:01 homeassistant systemd[1]: Started Daily Cleanup of Temporary Directories.
Feb 09 01:26:01 homeassistant systemd[1]: Reached target System Time Set.
Feb 09 01:26:01 homeassistant systemd[1]: Starting Docker Socket for the API...
Feb 09 01:26:01 homeassistant hassos-persists[361]: [INFO] machine-id is okay
Feb 09 01:26:01 homeassistant systemd[1]: Listening on Journal Gateway Service Socket.
Feb 09 01:26:01 homeassistant systemd[1]: Listening on Docker Socket for the API.
Feb 09 01:26:01 homeassistant systemd[1]: Reached target Socket Units.
Feb 09 01:26:01 homeassistant rngd[363]: Initializing available sources
Feb 09 01:26:01 homeassistant rngd[363]: [hwrng ]: Initialized
Feb 09 01:26:01 homeassistant rngd[363]: [rndr  ]: No HW SUPPORT
Feb 09 01:26:01 homeassistant rngd[363]: [rndr  ]: Initialization Failed
Feb 09 01:26:01 homeassistant systemd[1]: Reached target Basic System.
Feb 09 01:26:01 homeassistant systemd[1]: System is tainted: cgroupsv1
Feb 09 01:26:01 homeassistant systemd[1]: Starting Network Manager...
Feb 09 01:26:01 homeassistant systemd[1]: Starting Bluetooth for BCM43xx...
Feb 09 01:26:01 homeassistant systemd[1]: Starting HassOS persists data on bootstate...
Feb 09 01:26:01 homeassistant systemd[1]: Starting Rauc Update Service...
[... cut ...]
Feb 09 01:26:01 homeassistant systemd[1]: Starting WPA supplicant...
Feb 09 01:26:01 homeassistant systemd-resolved[355]: Positive Trust Anchors:
Feb 09 01:26:01 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
Feb 09 01:26:01 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
Feb 09 01:26:01 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root

```

jus101, 2022-02-09:
Just chiming in here to say I've been running into similar issues in the past 48 hours.
This was with 6.4, but I've since updated to 7.3 in the hopes that it might stop the problem.

A couple of things:

1. System seems completely unresponsive. It doesn't respond to ping or keyboard.
2. When I plugged in a monitor, it showed the HAOS prompt, and the RPI4 temperature icon was flashing. Heatsink was hot.
3. I tried to review the host logs, but I'm getting `zsh: command not found: journalctl`. Is there something I need to do to install journalctl?

Kackerlacka, 2022-02-11:
Had another crash yesterday but unfortunately the logs were empty. Unsurprisingly, I had another crash today as well, and it looks like there may be logs that could help:
```
# journalctl --since "22-02-11 07:00:00" --until "22-02-11 09:15:00"
-- Journal begins at Sun 2022-01-23 02:28:33 UTC, ends at Fri 2022-02-11 09:23:24 UTC. --
Oct 12 15:24:08 homeassistant systemd-timesyncd[366]: System clock time unset or jumped backwards, restoring from recorded timestamp: Fri 2022-02-11 07:59:36 UTC
Feb 11 07:59:36 homeassistant systemd[1]: Started Network Time Synchronization.
Feb 11 07:59:36 homeassistant kernel: audit: type=1334 audit(1644566376.059:4): prog-id=7 op=LOAD
Feb 11 07:59:36 homeassistant kernel: audit: type=1334 audit(1644566376.059:5): prog-id=8 op=LOAD
Feb 11 07:59:36 homeassistant kernel: uart-pl011 fe201000.serial: no DMA platform data
Feb 11 07:59:36 homeassistant audit: BPF prog-id=7 op=LOAD
Feb 11 07:59:36 homeassistant audit: BPF prog-id=8 op=LOAD
Feb 11 07:59:36 homeassistant systemd-time-wait-sync[128]: adjtime state 5 status 40 time Fri 2022-02-11 07:59:36.000262 UTC
Feb 11 07:59:36 homeassistant systemd[1]: Reached target System Initialization.
Feb 11 07:59:36 homeassistant systemd[1]: Started Daily Cleanup of Temporary Directories.
Feb 11 07:59:36 homeassistant systemd[1]: Reached target System Time Set.
Feb 11 07:59:36 homeassistant hassos-persists[371]: [INFO] machine-id is okay
Feb 11 07:59:36 homeassistant systemd[1]: Starting Docker Socket for the API...
Feb 11 07:59:36 homeassistant systemd[1]: Listening on Journal Gateway Service Socket.
Feb 11 07:59:36 homeassistant rngd[375]: Initializing available sources
Feb 11 07:59:36 homeassistant rngd[375]: [hwrng ]: Initialized
Feb 11 07:59:36 homeassistant rngd[375]: [rndr  ]: No HW SUPPORT
Feb 11 07:59:36 homeassistant rngd[375]: [rndr  ]: Initialization Failed
Feb 11 07:59:36 homeassistant rngd[375]: [jitter]: Initializing AES buffer
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/var/lib/systemd/linger supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
Feb 11 07:59:36 homeassistant systemd[1]: Listening on Docker Socket for the API.
Feb 11 07:59:36 homeassistant systemd[1]: Reached target Socket Units.
Feb 11 07:59:36 homeassistant wpa_supplicant[383]: Successfully initialized wpa_supplicant
Feb 11 07:59:36 homeassistant systemd[1]: Reached target Basic System.
Feb 11 07:59:36 homeassistant systemd[1]: System is tainted: cgroupsv1
Feb 11 07:59:36 homeassistant systemd[1]: Starting Network Mana
```

Kackerlacka, 2022-02-11:
> Just chiming in here to say I've been running into similar issues in the past 48 hours. This was with 6.4, but I've since updated to 7.3 in the hopes that it might stop the problem.
>
> A couple of [...]

I don't think you can use `journalctl` through zsh, you need to get access to the root system. You gotta follow the link above to create a public key and private key, put the public key on your system using the USB method, and log into your system as root using the private key through PuTTY.

Kackerlacka, 2022-02-17:
Another crash...
```
# journalctl --since "22-02-17 03:00:00" --until "22-02-17 04:00:00"
-- Journal begins at Sat 2022-01-29 18:34:25 UTC, ends at Thu 2022-02-17 04:17:25 UTC. --
Oct 12 15:24:08 homeassistant systemd-timesyncd[365]: System clock time unset or jumped backwards, restoring from recorded timestamp: Thu 2022-02-17 03:53:14 UTC
Feb 17 03:53:14 homeassistant systemd[1]: Started Network Time Synchronization.
Feb 17 03:53:14 homeassistant kernel: audit: type=1334 audit(1645069994.059:4): prog-id=7 op=LOAD
Feb 17 03:53:14 homeassistant kernel: audit: type=1334 audit(1645069994.059:5): prog-id=8 op=LOAD
Feb 17 03:53:14 homeassistant kernel: uart-pl011 fe201000.serial: no DMA platform data
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/var/lib/systemd/linger supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/NetworkManager/system-connections supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/docker supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/hostname supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/udev/rules.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/hosts supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modprobe.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/dropbear supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/systemd/timesyncd.conf supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/etc/modules-load.d supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant kernel: audit: type=1334 audit(1645069994.315:6): prog-id=9 op=LOAD
Feb 17 03:53:14 homeassistant kernel: audit: type=1334 audit(1645069994.315:7): prog-id=10 op=LOAD
Feb 17 03:53:14 homeassistant audit: BPF prog-id=7 op=LOAD
Feb 17 03:53:14 homeassistant audit: BPF prog-id=8 op=LOAD
Feb 17 03:53:14 homeassistant audit: BPF prog-id=9 op=LOAD
Feb 17 03:53:14 homeassistant audit: BPF prog-id=10 op=LOAD
Feb 17 03:53:14 homeassistant systemd-time-wait-sync[129]: adjtime state 5 status 40 time Thu 2022-02-17 03:53:14.000142 UTC
Feb 17 03:53:14 homeassistant dbus-daemon[120]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.7' (uid=0 pid=368 comm="/usr/sbin/NetworkManager --no-daemon ")
Feb 17 03:53:14 homeassistant systemd[1]: Reached target System Initialization.
Feb 17 03:53:14 homeassistant kernel: ext4 filesystem being remounted at /run/systemd/unit-root/var/lib/systemd/linger supports timestamps until 2038 (0x7fffffff)
Feb 17 03:53:14 homeassistant systemd[1]: Started Daily Cleanup of Temporary Directories.
Feb 17 03:53:14 homeassistant systemd[1]: Reached target System Time Set.
Feb 17 03:53:14 homeassistant hassos-persists[370]: [INFO] machine-id is okay
Feb 17 03:53:14 homeassistant systemd[1]: Starting Docker Socket for the API...
Feb 17 03:53:14 homeassistant systemd[1]: Listening on Journal Gateway Service Socket.
Feb 17 03:53:14 homeassistant rngd[375]: Initializing available sources
Feb 17 03:53:14 homeassistant rngd[375]: [hwrng ]: Initialized
Feb 17 03:53:14 homeassistant rngd[375]: [rndr  ]: No HW SUPPORT
Feb 17 03:53:14 homeassistant rngd[375]: [rndr  ]: Initialization Failed
[... cut ...]
Feb 17 03:53:28 homeassistant systemd[1]: Finished Network Manager Wait Online.
Feb 17 03:53:28 homeassistant systemd[1]: Reached target Network is Online.
Feb 17 03:53:28 homeassistant NetworkManager[368]: <info>  [1645070008.9040] manager: NetworkManager state is now CONNECTED_GLOBAL
Feb 17 03:53:38 homeassistant systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Feb 17 03:53:39 homeassistant kernel: cam-dummy-reg: disabling
Feb 17 03:53:39 homeassistant kernel: vcc-sd: disabling
Feb 17 03:53:39 homeassistant kernel: cam1-reg: disabling
```

agners, 2022-02-18:
The logs are not really helpful unfortunately. It doesn't show anything special.

Can you try if upgrading to 7.4 helps? It comes with new Kernel and Raspberry Pi firmware.

Did you try connecting an HDMI monitor? It might show something, especially if you run into the situation where it gets stuck.

Your monitoring shows a drop in memory usage quite frequently (once or even twice a day). Is this a reboot? E.g. if you check uptime or logs of your system, does it reboots without you noticing it?

gregersn, 2022-02-20:
I have the same problem, upgradeded to 7.4, the problem persist.

agners, 2022-02-21:
Yeah I'd guess its just newer versions of software which use more memory, making the system now running under so much memory pressure that it crashes sometimes. Can you try disabling some add-ons to see if it gets stable with that?

closed 2022-05-29 by github-actions[bot]