home-assistant/operating-system #1741: System becoming unresponsive - often requires full reboot of PI4 Host - 7.2

Trebacz, 2022-02-08:
### Describe the issue you are experiencing

Recently My Home Assistant seems to get very high processor usage and some controls in the UI become unavailable (and some work fine -maybe a bit slow). Eventually my supervisor becomes unhealthy. Typically getting the host to reboot clears things up. This a couple year installation that just started behaving badly.

I just learned how to pull the docker logs using - ssh root@homeassistant.local -p 22222  and docker logs hassio_supervisor

In the UI Supervisor logs are very slow to load - maybe 2 minutes

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
These do not load in the UI.

In Configuration > Settings > Logs I see an error like below:

Logger: homeassistant.components.hassio
Source: components/hassio/websocket_api.py:120
Integration: Home Assistant Supervisor (documentation, issues)
First occurred: 6:51:25 PM (4 occurrences)
Last logged: 7:29:58 PM

Failed to to call /core/stats -
Failed to to call /supervisor/stats -
```

### Anything in the Host logs that might be useful for us?

```txt
22-02-07 18:35:42 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/sdb - /dev/disk/by-id/usb-SanDisk_Cruzer_Glide_4C531001510814109160-0:0                                                                                                                                       
22-02-07 18:35:42 INFO (MainThread) [supervisor.hardware.monitor] Detecting HardwareAction.REMOVE hardware /dev/bus/usb/001/004 - None
22-02-07 18:45:19 WARNING (SyncWorker_2) [supervisor.docker.interface] Can't grep logs from homeassistant/aarch64-hassio-supervisor: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out.                                                                                                                     
22-02-07 18:45:19 WARNING (SyncWorker_1) [supervisor.docker.interface] Can't grep logs from homeassistant/aarch64-hassio-supervisor: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out.                                                                                                                     
22-02-07 18:45:24 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-02-07 18:45:39 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-155049' coro=<Scheduler._run_task.<locals>._wrap_task() done, defined at /usr/src/supervisor/supervisor/misc/scheduler.py:58> exception=DockerRequestError()>                                                                                                                               
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/local/lib/python3.9/http/client.py", line 1377, in getresponse
    response.begin()
  File "/usr/local/lib/python3.9/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.9/http/client.py", line 281, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out
During handling of the above exception, another exception occurred:
[... 443 lines cut ...]
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), Terminal & SSH (9.3.0), InfluxDB (4.3.0), Home Assistant Google Drive Backup (0.105.2), Spotify Connect (0.10.0), ESPHome (2022.1.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 9
views | 7
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

Trebacz, 2022-02-08:
So running - hdparm -t /dev/sda

/dev/sda:
Timing buffered disk reads:  271 MB in 3.13 seconds = 88633 kB/s

which seems very slow for an SSD... may try suggestions here - https://community.home-assistant.io/t/supervisor-logs-extremely-slow-to-open/344375/20

agners, 2022-02-08:
Are the kernel logs showing anything (`dmesg`)?

agners, 2022-02-08:
Nothing which really stands out from the log files. Have you had the issue again after disabling UAS?

jeroenvdheuvel, 2022-02-08:
I'm having similar issues without disabling UAS

Supervisor log:
```
Home assistant error system freeze

22-02-08 19:24:08 INFO (SyncWorker_0) [supervisor.docker.interface] Stopping hassio_observer application
22-02-08 19:24:10 INFO (MainThread) [supervisor.homeassistant.api] Updated Home Assistant API token
22-02-08 19:24:14 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_observer application
22-02-08 19:24:14 INFO (MainThread) [supervisor.plugins.observer] Starting observer plugin
22-02-08 19:24:18 INFO (SyncWorker_1) [supervisor.docker.observer] Starting Observer ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0 - 172.30.32.6
22-02-08 19:24:21 ERROR (MainThread) [asyncio] Task exception was never retrieved
future: <Task finished name='Task-3589' coro=<Scheduler._run_task.<locals>._wrap_task() done, defined at /usr/src/supervisor/supervisor/misc/scheduler.py:58> exception=DockerError()>
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 449, in _make_request
    six.raise_from(e, None)
  File "<string>", line 3, in raise_from
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 444, in _make_request
    httplib_response = conn.getresponse()
  File "/usr/local/lib/python3.9/http/client.py", line 1377, in getresponse
    response.begin()
  File "/usr/local/lib/python3.9/http/client.py", line 320, in begin
    version, status, reason = self._read_status()
  File "/usr/local/lib/python3.9/http/client.py", line 281, in _read_status
    line = str(self.fp.readline(_MAXLINE + 1), "iso-8859-1")
  File "/usr/local/lib/python3.9/socket.py", line 704, in readinto
    return self._sock.recv_into(b)
socket.timeout: timed out
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 440, in send
    resp = conn.urlopen(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 785, in urlopen
    retries = retries.increment(
  File "/usr/local/lib/python3.9/site-packages/urllib3/util/retry.py", line 550, in increment
    raise six.reraise(type(error), error, _stacktrace)
  File "/usr/local/lib/python3.9/site-packages/urllib3/packages/six.py", line 770, in reraise
    raise value
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 703, in urlopen
    httplib_response = self._make_request(
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 451, in _make_request
    self._raise_timeout(err=e, url=url, timeout_value=read_timeout)
  File "/usr/local/lib/python3.9/site-packages/urllib3/connectionpool.py", line 340, in _raise_timeout
    raise ReadTimeoutError(
urllib3.exceptions.ReadTimeoutError: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out. (read timeout=900)
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/supervisor/supervisor/docker/interface.py", line 556, in _is_failed
    docker_container = self.sys_docker.containers.get(self.name)
  File "/usr/local/lib/python3.9/site-packages/docker/models/containers.py", line 897, in get
    resp = self.client.api.inspect_container(container_id)
  File "/usr/local/lib/python3.9/site-packages/docker/utils/decorators.py", line 19, in wrapped
    return f(self, resource_id, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/docker/api/container.py", line 772, in inspect_container
    self._get(self._url("/containers/{0}/json", container)), True
  File "/usr/local/lib/python3.9/site-packages/docker/utils/decorators.py", line 46, in inner
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/docker/api/client.py", line 237, in _get
    return self.get(url, **self._set_request_timeout(kwargs))
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 542, in get
    return self.request('GET', url, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 529, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/sessions.py", line 645, in send
    r = adapter.send(request, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/requests/adapters.py", line 532, in send
    raise ReadTimeout(e, request=request)
requests.exceptions.ReadTimeout: UnixHTTPConnectionPool(host='localhost', port=None): Read timed out. (read timeout=900)
The above exception was the direct cause of the following exception:
Traceback (most recent call last):
[... 5 lines cut ...]
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/supervisor/supervisor/docker/interface.py", line 560, in _is_failed
    raise DockerError() from err
supervisor.exceptions.DockerError
22-02-08 19:44:09 INFO (MainThread) [supervisor.resolution.check] Starting system checks with state CoreState.RUNNING
22-02-08 19:44:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.SECURITY/ContextType.CORE
22-02-08 19:44:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.PWNED/ContextType.ADDON
22-02-08 19:44:09 INFO (MainThread) [supervisor.resolution.checks.base] Run check for IssueType.TRUST/ContextType.SUPERVISOR
[... cut ...]
22-02-08 19:44:09 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-02-08 19:45:43 INFO (MainThread) [supervisor.api.middleware.security] /supervisor/info access from cebe7a76_hassio_google_drive_backup
22-02-08 19:45:43 INFO (MainThread) [supervisor.api.middleware.security] /backups access from cebe7a76_hassio_google_drive_backup
22-02-08 19:45:47 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
```

dmesg logs:
```

```

Trebacz, 2022-02-09:
> Nothing which really stands out from the log files. Have you had the issue again after disabling UAS?

So far it has not reoccurred. That said it was unpredictable. What I have noticed are my CPU temps are up a couple degrees F, since disabling UAS, but y system logs in the UI load lightning fast.

jeroenvdheuvel, 2022-02-11:
Although the errors that happened on my Pi looked very similar. They were related to an addon writing too much data making the whole Pi stall. The timing of my Hass Os update and the addon were very unfortunate. I disabled the addon and haven't had an unresponsive system since.

Trebacz, 2022-02-13:
> > Nothing which really stands out from the log files. Have you had the issue again after disabling UAS?
> 
> So far it has not reoccurred. That said it was unpredictable. What I have noticed are my CPU temps are up a couple degrees F, since disabling UAS, but the system logs in the UI load lightning fast.

Been almost 1 week after disabling UAS, CPU temps are up a couple degrees F, system logs in the UI load lightning fast and no system freezes.

agners, 2022-02-25:
Ok, thanks for the update, so it seems that disabling UAS has helped then. `174c:55aa:u` will be in the list of drives disabled by default in the future (see #1743).

closed 2022-02-25 by agners