home-assistant/operating-system #1743: Adding the UAS module to kern module blacklist?

philippecrowdsec, 2022-02-08:
### Describe the issue you are experiencing

UAS mode is bringing marginal performance boost to USB disk R/W but the tradeoff is huge since most disks will generate errors on an RPI4 if plugged on the USB bus without patching the kernel command line. Adding the UAS to the module blacklist could maybe help to avoid those errors in the first place.

Actually you have to mount your USB drive boot partition, edit the cmdline and add something like:
usb-storage.quirks="152d:0579:u" dwc_otg.lpm_enable=0 console=tty1
(152d:0579 coming from a lsusb)

While this works, it's still an extra headache for the users, and using an SSD drive is way safer than an SDcard.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.1

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Plug a USB drive
2. Half of times, you'll have issue, seeing in your logs squashfs complaining or lines like 
3. . . . [sda] tag#27 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD IN
  . . . [sda] tag#27 CDB: opcode=0x28 28 00 00 03 b0 3a 00 00 b0 00
4. Disabling UAS on the given drive, by editing the cmdline.txt will get rid of this problem

### Anything in the Supervisor logs that might be useful for us?

```txt
nope, the kern.log is the most relevant source
```

### Anything in the Host logs that might be useful for us?

```txt
mostly the kern.log / console
```

philippecrowdsec, 2022-02-08:
(also relates to this post I made in the community forums for deeper explanations : https://community.home-assistant.io/t/raspberri-pi-4-9-steps-howto-get-both-hassio-boot-data-run-over-an-ssd/372548)

philippecrowdsec, 2022-02-08:
https://github.com/raspberrypi/linux/issues/3855

philippecrowdsec, 2022-02-08:
Yeah indeed, there are some setups that work. But, this is dependent on a combination of PI version, cable, hub, disk. So a lot can go wrong. 

The drawback of a generic ban is to lower the potential max throughput of a disk of about 50%, but given the speed of the USB3 bus of a PI, rather 20%. But anyway, there is no need for 50 Mb/s instead of 30 Mb/S on a PI using HA. I'm not even sure a large setup would be able to use 10 Mb/s of throughput.

On the other hand, for the people that have issues (not a marginal amount I'd guess), the process of mounting the drive, editing the command line and all is pretty daunting. Beyond the fact that having both boot & data is better, even just migrating the data to an external drive (offered by default by HA) is often failing because of this.

So even if its a trade-off, it's lowering unused max perf, in favor of more simplicity (no complicated command to issue for the user to migrate to an SSD or impossible to find combo of cable, hub, drive, pi to make it work) and security (more people using a SSD, vastly more stable and durable than an SDcard)

Then if the user wants to get the most out of its disk, it's easier to remove the UAS from the module blacklist for some, than editing the cmdline for most.

agners, 2022-02-08:
Valid points.

What is Raspberry Pi doing on that front? Do they blacklist by default? Did they consider blacklist it?

philippecrowdsec, 2022-02-08:
Excellent question and I've no good answer to help here.
My guess here is that distro-makers (or containers bakers) are all having their own receipt here. You could use a PI to make a FreeNAS and this patch would then greatly harm the perf of this NAS so adding it by default to Raspbian would likely be counterproductive.

Most use cases of the PI I know about are related to running it on an SDcard. I never really dug into the HDD/SSD problem before falling in love with HA and wanting to have a stable environment to run it.

I found a thread here though: https://github.com/raspberrypi/linux/issues/3855

Googling "github: raspberry pi SSD UAS module to kernel blacklist" returned a few results.

tmm1, 2022-02-08:
Seems like the wrong approach to me.

> since most disks will generate errors on an RPI4 if plugged on the USB bus without patching the kernel command line.

We work with a lot of consumer HDDs and my experience is the opposite. Most of them work just fine, and take advantage of UAS.

The problems seems to occur more with older DIY enclosures that people are using for SSDs.

> I'd rather prefer to have a set of `usb-storage.quirks`.

Agree with this, if the goal is to make things easier and work out of the box then we should bundle the list of quirks so most setups work.

agners, 2022-02-08:
Let's do that then. It seems that the JMicron already make a rather big share of the trouble makers. I found those in the Raspberry Pi Forum where there is also confirmation that those also helped:

- 152d:0579:u
- 152d:0578:u
- 152d:1561:u e.g. [Sabrent USB 3.0 to SATA-III adapter cable](https://www.amazon.com/dp/B011M8YACM).

philippecrowdsec, 2022-02-08:
Thanks guys for considering the request, I'd think your evaluation is very
fair and call it a day then.
If some others are running into the problem and they are the minority, then
they can follow a tutorial indeed.

Keep up the great work team!
(btw we're porting CrowdSec to HA as I speak, more security for the masses)

tmm1, 2022-02-08:
Here are some we are including: https://github.com/fancybits/operating-system/commit/72b2fde1bb5d0e224491fbb9f0e5b1afc3944d32

philippecrowdsec, 2022-02-08:
hummm indeed, I didn't see those (yet) in my own installation, so I was not
aware you had a manual list, sorry about that.
Well done!

axlthorell, 2022-02-10:
I have the [StarTech USB-SATA 3.0](https://www.amazon.com/StarTech-com-SATA-USB-Cable-USB3S2SAT3CB/dp/B00HJZJI84/ref=sr_1_3?crid=23YHHDE6PDZ3D&keywords=startech+usb+sata&qid=1644503912&sprefix=startech+usb+sata%2Caps%2C189&sr=8-3) adapter and use a [Kingston A400 120gb SSD](https://www.amazon.com/Kingston-120GB-Solid-SA400S37-120G/dp/B01N6JQS8C/ref=sr_1_1?crid=3QVQL9WZE9S0Q&keywords=kingston+a400+120gb&qid=1644503948&sprefix=kingston+a400+120gb%2Caps%2C165&sr=8-1). It works with UAS but after a while the system log in HA is slow to load and with UAS turned off it's now much more responsive.

- 14b0:0206:u

wokkeltje13, 2022-05-04:
> I have the [StarTech USB-SATA [...]

For me this fixed also made a huge difference with displaying the logs. Seems to be the same StarTech adapter.

Disk: Crucial BX500 with StarTech.com SATA to USB Cable - USB 3.0 to 2.5" SATA III Hard Drive Adapter

* 174c:55aa

agners, 2022-05-05:
Added `174c:55aa:u` to the list as well. With that we should have a pretty good default blacklist as part of HAOS 8.0.

closed 2022-05-05 by agners

carsten-h, 2022-05-16:
> Added `174c:55aa:u` to the list as well.

This is interesting. I had to use this quirk in my Argon One M.2 case since HA OS 5.10 I think, because my logs have been really slow to show: [https://github.com/home-assistant/supervisor/issues/3014](url).
This is the 64bit version of HA OS.

I also have a Pi 4 4GB inside the Argon One M.2 case but I am using Raspberry Pi OS Bullseye in 32bit.
There I have no problems when using this controller. UAS is working fine without any problem.

Is is possible that is is a 64bit only problem? Maybe there is somewhere an error in the 64 bit implementation?

philippecrowdsec, 2022-05-16:
It looks to me like a USB3 power and/or implementation problem.

Raspberry is not known to implement the most "by the book" USB hardware and there is also often a power shortage problem. Now to add to this mess, USB3 hubs are usually cheap and poorly designed, here again not very respectful of any norm. Combining the two, you get a very shaky gateway between an exigent SSD drive and an exigent OS. What could possibly go wrong? Well the UAS mode for a start, which is a combo of the old (but good) SCSI norm, adapted to the USB3 bus. It's way better & faster than the older BOT module that was used for this, but it's stricter. And when you get this unlikely teaming between RPI/USB hubs & varying quality of SSD drives, you can caught in this problem.

Now honestly deactivating the UAS for an SSD serving HA on an RPi, I'd bet there are very few corner cases where you would feel any performance difference I think. But I totally get the decision of the maintainers to support the majority case and not lose performances over complying even with the unlucky ones but sacrificing perfs. Anyway, this thread (and the one on HA forum) are there to hint the minority users having this problem to a pragmatic solution.

dafunkydan, 2023-01-22:
Another incompatible Adapter:
idVendor:174c
idProduct:225c

Sold as _"UGREEN USB 3.0 SATA Adapter für 2.5 Zoll SSD und HDD Festplatten SATA auf USB Adapter unterstützt SATA III UASP Ruhemodus SMART Trim usw."_, bought 07/2022 has the nasty UAS-Incompatibility as well.
Appending the Quirk manually did the Job.

agners, 2023-01-23:
Thanks for your report, this PR adds that device as well: https://github.com/home-assistant/operating-system/pull/2310

Ideally new devices should be reported in a new issue, so we have them tracked separately.