home-assistant/operating-system #1746: bluetoothctl won't keep devices connected

shawly, 2022-02-09:
### Describe the issue you are experiencing

I can't connect any bluetooth device to HA OS through bluetoothctl. Tried a bluetooth remote camera shutter which registers as input-keyboard and a Logitech MX Master.

I can reproduce the issue on two different HA OS instances on two different hosts with two different BT devices (a BT 5.0 USB Dongle and the internal BT module from an Intel NUC).

I cannot reproduce the issue on standard Linux machines or on the host machine itself, only within HA OS this problem exists. BT and BLE does work fine though in HA.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.3

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. In a terminal execute `bluetoothctl`
2. Execute `scan on`
3. Copy the mac address of the device
4. Execute `pair <mac>`
5. Execute `trust <mac>`
6. Execute `connect <mac>`

Example: 

```
[bluetooth]# pair 11:22:33:FF:5F:AB
Attempting to pair with 11:22:33:FF:5F:AB
[CHG] Device 11:22:33:FF:5F:AB Connected: yes
[CHG] Device 11:22:33:FF:5F:AB Modalias: usb:v05ACp022Cd011B
[CHG] Device 11:22:33:FF:5F:AB UUIDs: 00001124-0000-1000-8000-00805f9b34fb
[CHG] Device 11:22:33:FF:5F:AB UUIDs: 00001200-0000-1000-8000-00805f9b34fb
[CHG] Device 11:22:33:FF:5F:AB ServicesResolved: yes
[CHG] Device 11:22:33:FF:5F:AB Paired: yes
Pairing successful
[CHG] Device 11:22:33:FF:5F:AB ServicesResolved: no
[CHG] Device 11:22:33:FF:5F:AB Connected: no
bluetooth]# trust 11:22:33:FF:5F:AB
[CHG] Device 11:22:33:FF:5F:AB Trusted: yes
Changing 11:22:33:FF:5F:AB trust succeeded
[bluetooth]# connect 11:22:33:FF:5F:AB
Attempting to connect to 11:22:33:FF:5F:AB
[CHG] Device 11:22:33:FF:5F:AB ServicesResolved: yes
[CHG] Device 98:D3:31:FB:6F:F1 Connected: yes
Failed to connect: org.bluez.Error.NotAvailable
[CHG] Device 11:22:33:FF:5F:AB ServicesResolved: no
[CHG] Device 11:22:33:FF:5F:AB Connected: no
```

### Anything in the Supervisor logs that might be useful for us?

```txt
Nothing apparent seems to be logged that relates to this
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing apparent, the bluetooth device just connects and thats it

[10188.610952] usb 2-1: new full-speed USB device number 3 using uhci_hcd
[10188.769742] usb 2-1: New USB device found, idVendor=8087, idProduct=0a2a, bcdDevice= 0.01
[10188.769784] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[10188.796889] Bluetooth: hci1: read Intel version: 370810011003110e32
[10188.796892] Bluetooth: hci1: Intel device is already patched. patch num: 32
```

### System Health information

## System Health

version | core-2022.2.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.91
arch | x86_64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 5000
Installed Version | 1.22.0
Stage | running
Available Repositories | 1049
Downloaded Repositories | 24

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.2
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 31.3 GB
disk_used | 5.7 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Let's Encrypt (4.12.0), ESPHome (2022.1.3), TasmoAdmin (0.16.0), Node-RED (10.4.0), Mosquitto broker (6.0.1), Studio Code Server (4.1.0), MariaDB (2.4.0), InfluxDB (4.3.0), Terminal & SSH (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 14
views | 11
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

agners, 2022-02-25:
Did 7.4 maybe solve this issue?

It seems that the USB device got disconnected/reconnected somehow?

shawly, 2022-02-25:
Nope this issue persists since the first time I tested this with 6.x.

> It seems that the USB device got disconnected/reconnected somehow?

Where?

shawly, 2022-02-25:
This is from when I attached the device to the virtual machine, so it's completely normal for the entry to show up

agners, 2022-04-05:
Can you try if this is reproducible in 8.0 builds? It comes with newer Bluetooth stack which hopefully resolves the problem:

https://os-builds.home-assistant.io/8.0.dev20220404/

shawly, 2022-04-05:
I'll try thanks!

yada75, 2022-04-10:
Not the same issue exactly, I'm using Odroid N2+ and TP-link UB400 with integration [Fjäråskupan](https://www.home-assistant.io/integrations/fjaraskupan/) and last working OS was 7.0.

Fjäråskupan or Bleak is little buggy with many errors and disconnects but it would always recover in OS 7.0. With OS 7.1 to 7.6 timeouts or errors within 1-5 min and won't recover.

Installed 8.0.dev220404 yesterday and Bluetooth is actually working better now than OS 7.0 with less errors and disconnects.

shawly, 2022-07-09:
So I tried it with 8.2 now and it behaves the same, the issue is still present and the bt connection drops after connecting.

shawly, 2022-07-13:
> Both are input devices. How do you plan to use them with Home Assistant?

With this integration https://www.home-assistant.io/integrations/keyboard_remote/

Yes the behavior is the same in 8.2 and it appears immediately. I can do the same under a normal Ubuntu Server installation on a NUC or on my desktop Linux Mint installation and the camera shutter connects fine and stays connected, it's registered as keyboard.

closed 2022-10-20 by github-actions[bot]

jan-the-maker, 2025-01-18:
I have a similar issue but with Antix Linux which compared to a current ubuntu VM shows the same behavior. It seems to be related to the underlying stack and not the specific USB devices  or a specific kernel.