home-assistant/operating-system #1748: HassOS 7.3 zigbee USB and Zwave USB devices gone

herwarth, 2022-02-10:
### Describe the issue you are experiencing

When updating to HassOS 7.3 I lost my Sonoff Zigbee 3 USB stick and my Aeotec Zwave usb stick. 
The device files were missing, so rolled back to 7.2 and the device files where there again

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

Install 7.2 and update the Is to 7.3
Notice /dev/serial/by-id directory is missing files

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

agners, 2022-02-14:
@bruvv I don't think that the OS upgrade is responsible for your case. The `ch341-uart` has always been enumerated as `ttyUSBX` devices. `ttyACM0` are devices which use the USB Communication Class drivers, which are mostly devices which support USB directly (don't use a USB to UART adapater like the CH341). Also, the fact the OS downgrade did not change the device name also shows that its unlikely related to the OS.

Note that Linux does not guarantee that the number after ttyUSB stays the same! You should always use the `by-id` paths, in your case `/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0`.

agners, 2022-02-22:
@herwarth is that still reproducible for you?

Just tried with an Aeotec Z-Wave stick on my end, and it seems to appear in the Hardware list just fine:

[image]

herwarth, 2022-02-22:
Hmmmmm it looks like my stick stopped responding. A reboot solved the issue. I will check if this happens more often in the future!

closed 2022-02-22 by herwarth