home-assistant/operating-system #1750: Update to version 7.4 failed

mariuske, 2022-02-11:
### Describe the issue you are experiencing

When trying to update to the OS to version 7.4 I'm getting the following error message in the UI:
`Unknown error, see supervisor`

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.2

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Open Configuration tab in Home Assistant
2. Click on the available update message
[image]
3. Click on the update button
[image]
4. Error message recevied in UI
[image]

### Anything in the Supervisor logs that might be useful for us?

```txt
22-02-11 11:04:43 INFO (MainThread) [supervisor.os.manager] Fetch OTA update from https://github.com/home-assistant/operating-system/releases/download/7.4/haos_ova-7.4.raucb
22-02-11 11:04:46 INFO (MainThread) [supervisor.os.manager] Completed download of OTA update file /data/tmp/hassos-7.4.raucb
22-02-11 11:04:47 ERROR (MainThread) [supervisor.os.manager] Home Assistant Operating System update failed with: Installation error: Destination device '/dev/disk/by-partlabel/hassos-kernel0' not found
```

### Anything in the Host logs that might be useful for us?

```txt
[1339783.326452] audit: type=1334 audit(1644567725.221:1828): prog-id=804 op=UNLOAD
[1339783.326455] audit: type=1334 audit(1644567725.221:1829): prog-id=803 op=UNLOAD
```

### Additional information

The OS is running as a virtual machine in Proxmox VE 7.1-10

agners, 2022-02-12:
Seems like your disk image corrupted. The update system rauc tries to find the "other" (currently not active) kernel partition, which seems no longer present (`/dev/disk/by-partlabel/hassos-kernel0`).

mariuske, 2022-02-20:
Yes, you are correct @agners! The vm disk had corrupt file system.. Solved now, thank you for pointing me in the right direction.

closed 2022-02-20 by mariuske

timdonovanuk, 2022-03-03:
I have the same issue btw, with exactly the same upgrade path (Proxmox  and 7.2 -> 7.4). Never had an issue with OS upgrades before. 

Coincidence? 

@agners @mariuske care to share how to resolve this please?

timdonovanuk, 2022-03-03:
Ugh, I'm getting this again, when I reboot:

[image]

This is 100% a HA issue that has persisted for years now running on Proxmox. There are several bug reports on the Proxmox support forms with the same error (all of them linked to people running HA in proxmox).

timdonovanuk, 2022-03-03:
Some more details. 

I restored to a snapshot I took literally a few mins before starting the OS update and it appears "/dev/disk/by-partlabel/hassos-kernel0" still exists at this point in time.

So would it be fair to jump to the conclusion that it is the OS update process that is removing/corrupting hassos-kernel0 ?

timdonovanuk, 2022-03-03:
Finally found a fix on one of the several Proxmox forum threads about this. It would appear somehow HA OS corrupts the partition map on the EFI drive. So doing the following fixes it:

```
Run commands on the PVE host
# Shut down VM
# Make sure VM is disabled (only needed if under Proxmox HA control)
ha-manager set vm:<VMID> --state disabled
# Open GDISK to modify disk partition map
gdisk /dev/zvol/rpool/vm-<VMID>-disk-<DISKID#> (has to be the same disk that maps to the HA sata disk, for me thats 1)
# Once GDISK opens, then just use the W command to re-write the partition map
# Re-enable (start) VM to verify the VM boots using the disk
ha-manager set vm:<VMID> --state enabled
```

Finally a fix for this!

agners, 2022-03-03:
> ```
> # Once GDISK opens, then just use the W command to re-write the partition map
> ```

Hm, so gdisk can recover the partition table from the second entry then?

> Finally a fix for this!

Thanks for sharing the fix.

That said, I wonder what the root of the problem is here. I can't really see why HAOS would corrupt that partition table. While we write to the boot partition and other partition, HAOS does nott touch the partition table...

timdonovanuk, 2022-03-03:
> Hm, so gdisk can recover the partition table from the second entry then?

Yep! It rewrote the partition map with no issues and then I could successfully boot and upgrade HA.

> That said, I wonder what the root of the problem is here. I can't really see why HAOS would corrupt that partition table.

Not sure I'm afraid - all I know is this has been an ongoing issue for years with HAOS running on Proxmox (v5, 6 & 7 now) It's quite unnerving because HA works fine until you shutdown or reboot it, and it's only then you find out your VM is hosed, which means it's hard to know which backups are uncorrupted. 

Thank goodness for live snapshots (to get back to at least a hot running HA instance) and my UPS which generally means I never have to boot HA from cold! 

If there is anything I can do to help debug please let me know - I still have the snapshot from the VM prior to fixing the partition map.

mariuske, 2022-03-03:
> Finally found a fix on one of the several Proxmox forum threads about this. It would appear somehow HA OS corrupts the partition map on the sata drive. So doing the following fixes it:
> 
> ```
> Run commands on the PVE host
> # Shut down VM
> # Make sure VM is disabled (only needed if under Proxmox HA control)
> ha-manager set vm:<VMID> --state disabled
> # Open GDISK to modify disk partition map
> gdisk /dev/zvol/rpool/vm-<VMID>-disk-<DISKID#> (has to be the same disk that maps to the HA sata disk, for me thats 1)
> # Once GDISK opens, then just use the W command to re-write the partition map
> # Re-enable (start) VM to verify the VM boots using the disk
> ha-manager set vm:<VMID> --state enabled
> ```
> 
> Finally a fix for this!

I can confirm that I used the same steps to resolve the issue.

Kamil-u, 2022-04-03:
> Finally found a fix on one of the several Proxmox forum threads about this. It would appear somehow HA OS corrupts the partition map on the sata drive. So doing the following fixes it:
>
> ```
> Run [...]

Also worked for me.

timdonovanuk, 2023-11-24:
@agners still getting this :(

Core
2023.11.2
Supervisor
2023.11.3
Operating System
11.0
Frontend
20231030.2