home-assistant/operating-system #1751: Raspberry pi 4 with deconz zigbee upgrade to 7.4 fails

p71-yes, 2022-02-11:
### Describe the issue you are experiencing

Did an upgrade to 7.4 the other day and my raspberry pi refused to start again.
After a few hours of troubleshooting by installing a new system and configuring it, I came to the conclusion that one row in config.txt destroys it all. By downgrading to 7.2 all works fine again.

this is the line:
dtoverlay=miniuart-bt

The attached log extract is when i start it without this option.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. upgrade to 7.4 with dtoverlay=miniuart-bt in the config.txt

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.components.zha.core.gateway
Source: components/zha/core/gateway.py:152
Integration: Zigbee Home Automation (documentation, issues)
First occurred: 13:17:29 (2 occurrences)
Last logged: 13:17:36

Couldn't start deCONZ = dresden elektronik deCONZ protocol: ConBee I/II, RaspBee I/II coordinator
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 492, in wait_for
    fut.result()
asyncio.exceptions.CancelledError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/zigbee/application.py", line 67, in startup
    self.version = await self._api.version()
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 463, in version
    (self._proto_ver,) = await self[NetworkParameter.protocol_version]
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 428, in read_parameter
    r = await self._command(Command.read_parameter, 1 + len(data), param, data)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 313, in _command
    return await asyncio.wait_for(fut, timeout=COMMAND_TIMEOUT)
  File "/usr/local/lib/python3.9/asyncio/tasks.py", line 494, in wait_for
    raise exceptions.TimeoutError() from exc
asyncio.exceptions.TimeoutError
```

### Anything in the Host logs that might be useful for us?

```txt
system refuses to start
```

deviantintegral, 2022-02-11:
I haven't been able to test this (I was checking if there were known issues before upgrading from 6.6), but I'm curious where that line could come from, as I see it commented out at https://github.com/home-assistant/operating-system/blob/7.4/buildroot-external/board/raspberrypi/config.txt#L66

p71-yes, 2022-02-11:
Sorry, my bad. I was having that on since before to have both bt and decon
raspbee working, and everything broke with the upgrade to 7.4.

agners, 2022-02-12:
I am confused. The line `dtoverlay=miniuart-bt` hasn't changed. It was never enabled by default. If it is enabled in your installation, then it has been enabled by you.

So 7.2 works then again? Does 7.4 works with `dtoverlay=miniuart-bt` disabled?

p71-yes, 2022-02-12:
Sorry for creating the confusion. Yes, that has been enabled on my system
since before, but when upgrading to 7.4 with it enabled the system never
started. I need that to have both bt and deconz raspbee working. The system
starts up with having it disabled, but then the raspbee is not working.

Rolling back to 7.2 makes everything work.

Answer to your questions: yes, yes.

agners, 2022-02-16:
What serial port path do you use to connect to DeCONZ USB stick (e.g. what is shown as title on the ZHA integration card)?

p71-yes, 2022-02-16:
ttyAMA0

The raspbee II is not a USB stick...

GitPetri, 2022-03-10:
For me the 7.4 upgrade fails too. Running HASS on RasPi4 and having RazBerry Z-Wave hat. [As I need also bluetooth](https://community.home-assistant.io/t/using-bluetooth-and-dtoverlay-pi3-miniuart-bt/63971/8) my /boot/config.txt includes
```
dtoverlay=miniuart-bt
core_freq=250
```
2 Attempts to upgrade from 7.3 to 7.4 failed so that the RasPi itself boots up but doesn't get network access. Normal setup is that the RasPi is accessing NW via WLAN but after the 1st botched upgrade I connected it to LAN cable and the same result after the 2nd upgrade attempt, Ethernet port LEDs light up for a second and then drop off i.e the device attempts to configure network interface. So something between 7.3 and 7.4 breaks network connection really bad (I assume it might be related to the corresponding config.txt lines.

enjikaka, 2022-03-15:
I think I'm seeing this issue too. Also using RPi4 with deConz (RaspBee II). I cannot access it now but I think I also have `dtoverlay=miniuart-bt`. 

I pressed OS update in the UI and now it doesn't boot. Never gets connected to my router.

GitPetri, 2022-03-16:
> I think I'm seeing this issue too. Also using RPi4 with deConz (RaspBee II). I cannot access it now but I think I also have `dtoverlay=miniuart-bt`.
> 
> I pressed OS update in the UI and now it doesn't boot. Never gets connected to my router.

@enjikaka Now that you have non-working setup (unless you did full re-install already), would you mind testing this:
- Remove deConz
- Comment out `dtoverlay` (and `core_freq` if you have it) from `/boot/config.txt`
- Boot
to see what happens?

I did full re-install and backup restore so I have up and running system now and I'm not willing to botch my setup as of now. I have an vague memory that I had similar problem with another non-HASS setup where I had the similar boot options and it was solved via commenting out the corresponding lines but I'm not sure.

If the system boots up, then it might be a semi-decent workaround and after the upgrades then just uncomment the lines and connect the deConz and reboot...

enjikaka, 2022-03-16:
@GitPetri Still haven't gotten anything to work. Just back uped the entire SD card to a DMG file so I can mess around. I'm not able to mount any partition from the SD card that includes /boot/config.txt :( I only manage to boot s8 with a supervisor directory...

What's the steps to downgrade or restore? I see some backup folder under supervisor/ that has som tgz files.

p71-yes, 2022-03-16:
@enjikaka What i did was:
mount the sd card in another computer, as i recall it the config.txt was on a fat partition
commented out the dtoverlay=miniuart-bt
booted up rpi with the sd card
ran in a shell (via ssh in my case): ha os update --version 7.2 
mounted the sd card in another computer again, changed the config.txt to include dtoverlay again
booted up rpi with the sd card again

enjikaka, 2022-03-16:
@p71-yes I have these available:

```
/dev/disk2 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:      GUID_partition_scheme                        *31.0 GB    disk2
   1:         Microsoft Reserved                         33.6 MB    disk2s1
   2:           Linux Filesystem                         25.2 MB    disk2s2
   3:           Linux Filesystem                         268.4 MB   disk2s3
   4:           Linux Filesystem                         25.2 MB    disk2s4
   5:           Linux Filesystem                         268.4 MB   disk2s5
   6:           Linux Filesystem                         8.4 MB     disk2s6
   7:           Linux Filesystem                         100.7 MB   disk2s7
   8:           Linux Filesystem                         30.3 GB    disk2s8
```

`diskutil mount disk2s1` just says:

```
Volume on disk2s1 failed to mount
Perhaps the operation is not supported (kDAReturnUnsupported)
If you think the volume is supported but damaged, try the "readOnly" option
```

🤔

enjikaka, 2022-03-16:
Ok managed to get in now. Had to do the following to get to the boot parition on mac (if someone else has this problem and end up here via Google):

- Find identifier in `diskutil list` with type "Microsoft Reserved", probably `disk2s1`
- Create mount point `sudo mkdir /Volumes/data`
- Mount `sudo mount -w -t msdos /dev/disk2s1 /Volumes/data`

--

My config.txt includes 

```
enable_uart=1
dtoverlay=pi3-miniuart-bt
```

I remember now I added these to get the the RaspBee II to be detected. I'll comment them out and see if it boots. But I don't wanna loost the zigbee radio...

enjikaka, 2022-03-16:
Only commented out `dtoverlay=pi3-miniuart-bt`. Not booting. Guess I'll try also commenting out `enable_uart=1`...

enjikaka, 2022-03-16:
Also commented out `enable_uart=1` now and it finally booted again! Going to try a downgrade via `ha os update --version 7.2` now...

GitPetri, 2022-03-17:
> Also commented out `enable_uart=1` now and it finally booted again!
> 
> When I tried dowgrading with `ha os update --version 7.2` I got
> 
> ```
> [core-ssh ~]$ ha os update --version 7.2
> Processing... Done.
> 
> Error: 'OSManager.update' blocked from execution, system is not running - CoreState.STARTUP
> ```

@enjikaka I think you might not need to downgrade the system (the errormessage above seems to be of a system still initializing itself anyways). Run all remaining OS upgrades (if any) and uncomment `dtoverlay `and `enable_uart `again. It _should_ work.

enjikaka, 2022-03-17:
`ha os update --version 7.2` worked randomly after 4-5 tries. Maybe Mars aligned with the moon or something.

When I commented in `dtoverlay` and `enable_uart` it didn't boot on 7.2 either... :/

enjikaka, 2022-03-17:
Just changed `dtoverlay=pi3-miniuart-bt` to `dtoverlay=pi3-disable-bt` and it botted and the RaspBee II works and discovers zigbee devices.

GitPetri, 2022-03-17:
> Just changed `dtoverlay=pi3-miniuart-bt` to `dtoverlay=pi3-disable-bt` and it booted and the RaspBee II works and discovers zigbee devices.

Caveat there is that if one wants to use onboard Bluetooth for for example RuuviTag collection, the `disable-bt` is not of course suitable.

GitPetri, 2022-03-17:
> 

I managed to find some time for testing, here is what I did (I do have the System Information data on both pre-and post Home Assistant Operating System update if it is of any interest to anyone):

Related setup
	• Raspberry Pi 4 Model B Rev 1.1
	• RaZberry Z-Wave Hat
	• ConBee II USB Zigbee stick
	• On-board Bluetooth utilized for sensor data (mainly RuuviTags)
	• `dtoverlay=miniuart-bt` and `core_freq=250` in `/boot/config.txt` for BT and CPIO co-existence

1. Take backup and store it to external destination

2. Shutdown
- Comment out dtoverlay=miniuart-bt from /boot/config.txt
- Comment out core_freq=250 from /boot/config.txt

3. Reboot
- -> RazBerry hat stops working (expected)
- -> Bluetooth stops working (expected)

4. Run Update Home Assistant Operating System 7.3->7.5, wait for the system to boot up
- -> RazBerry hat still doesn't working (expected)
- -> Bluetooth still doesn't  working (expected)

5. Shutdown
- Comment in dtoverlay=miniuart-bt from /boot/config.txt
- Comment in core_freq=250 from /boot/config.txt

6. Reboot
- -> RazBerry hat works (happiness)
- -> Bluetooth works (happiness)

7. Take backup and store it to external destination

enjikaka, 2022-03-18:
So commenting out dtoverlay during os update seems to do the trick then? I can try that as well. I don't have any `core_freq` though. What does that do? Should I add it?

GitPetri, 2022-03-18:
> So commenting out dtoverlay during os update seems to do the trick then? I can try that as well. I don't have any `core_freq` though. What does that do? Should I add it?
https://forums.raspberrypi.com/viewtopic.php?t=213499

GitPetri, 2022-03-19:
> dtoverlay=miniuart-bt

The `core_freq=250`/`dtoverlay=miniuart-bt` -combination is to ensure that a HAT/shield (in my case [RaZberry](https://z-wave.me/products/razberry/)) and the on-board Bluetooth work at the same time. If I've understood correctly, you have deConz USB-stick and in that case you probably don't need `core_freq`

GitPetri, 2022-04-08:
I re-tested this with Home Assistant OS 7.5->7.6 upgrade i.e

- Shutdown HA
- comment out the 2 corresponding lines in config.txt
- Start up HA
- Run OS upgrade
- Shutdown HA
- comment in the 2 corresponding lines in config.txt
- Start up HA
- -> works

I didn't even try to do the OS upgrade without commenting the config.txt lines

p71-yes, 2022-04-14:
> I re-tested this with Home Assistant OS 7.5->7.6 upgrade i.e
>
> * Shutdown HA
> * comment out the 2 corresponding lines in config.txt
> * Start up HA
> * Run OS upgrade
> * Shutdown HA
> * comment in the 2 [...]

This is some unexplainable black magic happening.
I did try "your way" of doing this, it didnt work. It got stuck in some boot loop.
I got myself a hdmi cable to be able to see whats going on, plugged it in and everything suddenly started up..

So, I've managed to successfully upgrade to 7.6 with everything seemingly working.

Thanks a bunch GitPetri for the solution!

Closing this.

closed 2022-04-14 by p71-yes