home-assistant/operating-system #1752: Doesn't boot on Intel Atom Baytrail

ytruplyt53, 2022-02-12:
### Describe the issue you are experiencing

After writing the image (7.4) to the internal eMMC from my Intel Atom Z3735F machine it's not able to boot. With a external ssd connected via usb it is the same result.

Looks like their is no `bootia32.efi` but only a `bootx64.efi` in the boot partition which could explain this.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Get yourself a device with a Intel Atom Z3735F (or probably any other Bay-Trail or Cherry-Trail based CPU) and internal eMMC
2. Boot a live system of your choice
3. Write and verify the HaOS image
4. Try to boot it

### Anything in the Supervisor logs that might be useful for us?

```txt
No, because it doesn't boot
```

### Anything in the Host logs that might be useful for us?

```txt
No, because it doesn't boot
```

agners, 2022-02-12:
Hm, 32-bit UEFI boot wasn't really a thing in PC world. The CPU supports 64-bit, so it should support `bootx64.efi`. Maybe a BIOS update helps?

Just to be sure: Did you make sure to enable UEFI boot and disable secure boot (probably not available on that machine yet)

ytruplyt53, 2022-02-12:
> Hm, 32-bit UEFI boot wasn't really a thing in PC world.

Indeed, but Intel still released a total of 46 baytrail cpus. It might also be that the cherry trail series is affected to by this awkwardness.

> enable UEFI boot

actually this is the only way to boot this as there is no "legacy boot"

> disable secure boot

included but disabled

ytruplyt53, 2022-02-12:
Did "some" experiments.

When I add a `bootia32.efi` in the `EFI/BOOT` directory my machine can detect it as a boot device.

I grabbed a random from github and it booted into a grub shell. Just copying the `bootx64.efi` and rename to `bootia32.efi` results in efi shell error.

The idea came after I read this here: https://www.barebox.org/doc/latest/boards/efi.html

This machine actually runs home assistant supervisor since years but on debian. I need to choose the multi-arch debian iso for installation.

agners, 2022-02-14:
> This machine actually runs home assistant supervisor since years but on debian. I need to choose the multi-arch debian iso for installation.

Hm, not familiar with that, is it using a 64-bit kernel then (check `uname -a`)?

I guess technically it should be possible to build and distribute a second build of Barebox for UEFI 32-bit. [This post](https://qastack.com.de/unix/155557/installing-linux-on-an-32bit-uefi-only-machine) claims that it is possible to boot a 64-bit kernel from a 32-bit EFI using Grub, quite likely it works also with Barebox. It certainly needs the `CONFIG_EFI_MIXED=y` kernel config. 

All that said, it seems this is a rather exotic platform at this point. I am not opposed to support it, but since I have no hardware and there is no widespread request for it, its unlikely I am going to work on it.

ytruplyt53, 2022-02-19:
> is it using a 64-bit kernel then

Indeed. This works since debian 8

https://wiki.debian.org/UEFI#Support_for_mixed-mode_systems:_64-bit_system_with_32-bit_UEFI

ytruplyt53, 2022-02-26:
@agners I just saw in another issue that HaOS (8) switches to grub. :muscle: 

Could a 32 bit UEFI support maybe added then? It should be as easy as install `grub-efi-ia32` `grub-efi-ia32-bin` packages as grub has full support booting 64bit os with 32bit efi :tada: 

If testing is necessary please count me in!

agners, 2022-02-28:
Actually, it seems that Buildroot we are using currently (2021.08) only supports one EFI GRUB configuration at a time. But it seems that this changed in upstream Buildroot with [this patch](https://github.com/buildroot/buildroot/commit/b68810e70cbda70ec4e1a267a7d1d3e935c5718b).

Updating Buildroot to 2022.02 is on my todo list. I'll give this a try after updating to it.

uros76, 2022-03-23:
> Actually, it seems that Buildroot we are using currently (2021.08) only supports one EFI GRUB configuration at a time. But it seems that this changed in upstream Buildroot with [this [...]

Hi. Any update supporting 32bit UEFI? I have a similar device like ytruplyt53 and cant get HA used beyond grub bootup.

agners, 2022-04-04:
@uros76 @ytruplyt53  the latest OS 8.0 development builds have the 32-bit UEFI version enabled. Unfortunately, I can't test it on my end as I don't have such hardware. Can one of you give this image a try?

https://os-builds.home-assistant.io/8.0.dev20220331/

uros76, 2022-04-04:
> @uros76 @ytruplyt53 the latest OS 8.0 development builds have the 32-bit UEFI version enabled. Unfortunately, I can't test it on my end as I don't have such hardware. Can one of you give this image a try?
> 
> https://os-builds.home-assistant.io/8.0.dev20220331/

This is good news, thanks. In the meantime, I managed to find 64bit bios for my devices which helped me get current HA OS installed without a problem. 

Perhaps @ytruplyt53 can test?

amonhk, 2022-04-04:
@agners it works perfectly
thank you very much you saved my old air-vision nvr

closed 2022-04-04 by agners