home-assistant/operating-system #1753: 1-Wire with USB master broken in kernel version 5.10?

EvTheFuture, 2022-02-15:
### Describe the issue you are experiencing

Trying to setup 1-wire sensors using an Intel nuc and a DS9490R USB adapter.

The devices show up under /sys/bus/w1/devices but the w1_slave file is missing as well as temperature.

In Home Assistent the sensors show up but the state is set to "unknown".

Tried this on kernel 5.14 on a laptop and it works fine. If I boot the same laptop with kernel version 5.11 the w1_slave file is missing and I can't read any temperature data from the attached sensors.

[I found this information](https://bugzilla.redhat.com/show_bug.cgi?id=1873694) which indicate that this issue has been fixed in kernel version 5.12 and later.

Are there any plans for bumping the kernel version any time soon?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Attach a DS9490 USB 1-wire adapter
2. Attach at least one 1-wire sensor
3. Try to read any data from the sensor

### Anything in the Supervisor logs that might be useful for us?

```txt
Logger: homeassistant.components.onewire.sensor
Source: components/onewire/sensor.py:520
Integration: 1-Wire (documentation, issues)
First occurred: 9:38:22 PM (230 occurrences)
Last logged: 10:35:23 PM

Cannot read from sensor /sys/bus/w1/devices/10-000801605acc/w1_slave: [Errno 2] No such file or directory: '/sys/bus/w1/devices//10-000801605acc/w1_slave'
Cannot read from sensor /sys/bus/w1/devices/28-00000399453f/w1_slave: [Errno 2] No such file or directory: '/sys/bus/w1/devices//28-00000399453f/w1_slave'
```

### Anything in the Host logs that might be useful for us?

```txt
Nothing of interest...
```

### System Health information

## System Health

version | core-2022.2.6
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Stockholm

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | beta
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 228.5 GB
disk_used | 7.5 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Z-Wave JS (0.1.53), Snapcast (0.5.1), SSH & Web Terminal (10.0.2), File editor (5.2.0), motionEye (0.15.1)

</details>

<details><summary>Lovelace</summary>

dashboards | 3
-- | --
resources | 0
views | 7
mode | storage

</details>

agners, 2022-02-16:
Hm, yeah looks like a problem in 5.10 (and hence HAOS 6.x/7.x). I plan to update to the latest LTS kernel 5.15 for 8.x. I let you know once development builds for testing are available.

EvTheFuture, 2022-02-17:
> I let you know once development builds for testing are available.

Perfect, do you know an approximate time for when those builds can start to show up?

Is there anything I can do to help?

closed 2022-05-25 by github-actions[bot]

pepsonEL, 2023-02-07:
any news in this thread ?