home-assistant/operating-system #1754: RPi 3B+ does not boot up with HassOS 7.3 or 7.4

dml33, 2022-02-17:
### Describe the issue you are experiencing

I have tried to installed HassOS 7.4 for Pi3 32 bits in a RPi 3 B+, but it does not boot up. 

When it is powered on, red and green leds light, but the green one turns off one second later and remains off. The red one (POWER) remains active. There is no image in the screen.

I have used Balena Etcher v1.7.3 with a uSD Kingston 8 GB and a SandDisk Extreme 64 GB A2/V30 with the same result. I have also tried the HassOS 7.4 Pi 3 64bit, HassOS 7.3 for Pi3 32 bit images and it does not boot either. I tried this latter with Raspberry Pi Imager v1.7.1 and the same.

However, I have installed Raspberry Pi OS BullsEye 2022-01-28 in the SandDisk 64 GB and it boots perfectly.

What could be happening?

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Use Balena Etcher v1.7.3 to flash a HassOS 7.4 Pi3 32 bit on a uSD
2. Insert the uSD in the Pi 3 B+
3. Power it on

### Anything in the Supervisor logs that might be useful for us?

```txt
It does not boot up, so I do not have any log.
```

### Anything in the Host logs that might be useful for us?

```txt
It does not boot up, so I do not have any log.
```

agners, 2022-02-19:
HAOS 7.4 64-bit works fine here on a Raspberry Pi 3 B+.

Not sure what the problem is, it sounds like the SD card somehow doesn't get flashed correctly. Can you try to use the Raspbery Pi Imager? It offers our official Home Assistant Operating System under "Other specific-purpose OS -> Home Assistant".

dml33, 2022-02-19:
Sorry, my bad!

I have several Raspberry Pi at home and the one I was trying with was a Pi 2 Model B... I tried again with 2 uSD and 2 laptops but it never booted up. I got so frustrated that I opened its case to check if something was broken. Then, I discovered that it was not a Pi 3, but a Pi 2...

I have just tried with the correct Pi 2 image and it boots up perfectly. I will play a little with it and I will buy a new Pi 4 at soon as they are available again in normal prices.

Sorry!

agners, 2022-02-19:
No worries, thanks for the update!

closed 2022-02-19 by agners