home-assistant/operating-system #1756: USB device crashing (Google Goral USB TPU)

Kurisutian, 2022-02-20:
### Describe the issue you are experiencing

Running HA directly from an NUC I3 after one of the latest OS upgrades my Coral USB device started crashing when being triggered from frigate. 
Eventually I am lucky and get one run but it will crash right after. I can rule out an hardware issue (testes other devices and they work, also testes all USB ports on the NUC and they all keep crashing with the Coral device). Also I tested the Coral on another NUC at a friends place who was running frigate and all also an an NUC (other model though) where the device worked flawless. 
Got in touch with the frigate developer but he could not help either. 
It was running fine for a while but as far as I can remember it seems like the issues started when upgrading the OS to one of the latest versions (somewhere at 7.1 or later). I am looking for different solutions and checked everything I can think of, but the device keeps crashing.
Restarting the NUC didn't work either, so I presume based on the error message in the frigate logs, the issue happens on the OS level. Here is the frigate log of the crash and the first line to me looked like it is somehow coming from an USB driver (so OS level if I'm not mistaken):

``` 
F driver/usb/usb_driver.cc:406] HandleEvent failed. USB transfer error 1 [LibUsbDataInCallback]
Fatal Python error: Aborted
Thread 0x00007f7ed794d700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/multiprocessing/queues.py", line 227 in _feed
  File "/usr/lib/python3.8/threading.py", line 870 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
Thread 0x00007f7ee851e740 (most recent call first):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 833 in invoke
  File "/opt/frigate/frigate/edgetpu.py", line 84 in detect_raw
  File "/opt/frigate/frigate/edgetpu.py", line 160 in run_detector
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108 in run
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315 in _bootstrap
  File "/usr/lib/python3.8/multiprocessing/popen_fork.py", line 75 in _launch
  File "/usr/lib/python3.8/multiprocessing/popen_fork.py", line 19 in __init__
  File "/usr/lib/python3.8/multiprocessing/context.py", line 277 in _Popen
  File "/usr/lib/python3.8/multiprocessing/context.py", line 224 in _Popen
  File "/usr/lib/python3.8/multiprocessing/process.py", line 121 in start
  File "/opt/frigate/frigate/edgetpu.py", line 221 in start_or_restart
  File "/opt/frigate/frigate/edgetpu.py", line 190 in __init__
  File "/opt/frigate/frigate/app.py", line 197 in start_detectors
  File "/opt/frigate/frigate/app.py", line 340 in start
  File "/opt/frigate/frigate/__main__.py", line 17 in <module>
  File "/usr/lib/python3.8/runpy.py", line 87 in _run_code
  File "/usr/lib/python3.8/runpy.py", line 194 in _run_module_as_main
``` 

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Set up frigate
2. Attach Coral USB
3. Wait for detection triggering TPU
4. Check frigate logs for the crash showing up

### Anything in the Supervisor logs that might be useful for us?

```txt
nothing useful
```

### Anything in the Host logs that might be useful for us?

```txt
nothing useful
```

### System Health information

## System Health

version | core-2022.2.9
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Berlin

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
Github API Calls Remaining | 4993
Installed Version | 1.22.0
Stage | running
Available Repositories | 1040
Downloaded Repositories | 32

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 116.7 GB
disk_used | 32.9 GB
healthy | true
supported | true
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Terminal & SSH (9.3.0), Node-RED (11.0.3), Mosquitto broker (6.0.1), Studio Code Server (4.1.0), InfluxDB (4.3.0), Grafana (7.4.1), Grocy (0.16.0), Barcode Buddy for Grocy (dev), Frigate NVR (3.0), Z-Wave JS (0.1.54), Vaultwarden (Bitwarden) (0.15.0), Caddy 2 (1.1.0), AdGuard Home (4.4.4), UniFi Network Application (1.1.4), Network UPS Tools (0.9.1), chrony (2.2.2), Samba Backup (5.0.0), evcc (0.30), Genie - Edge (edge-3d0c903), ESPHome (2022.2.3)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 20
views | 22
mode | storage

</details>

### Additional information

``` 
F driver/usb/usb_driver.cc:406] HandleEvent failed. USB transfer error 1 [LibUsbDataInCallback]
Fatal Python error: Aborted
Thread 0x00007f7ed794d700 (most recent call first):
  File "/usr/lib/python3.8/threading.py", line 302 in wait
  File "/usr/lib/python3.8/multiprocessing/queues.py", line 227 in _feed
  File "/usr/lib/python3.8/threading.py", line 870 in run
  File "/usr/lib/python3.8/threading.py", line 932 in _bootstrap_inner
  File "/usr/lib/python3.8/threading.py", line 890 in _bootstrap
Thread 0x00007f7ee851e740 (most recent call first):
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 833 in invoke
  File "/opt/frigate/frigate/edgetpu.py", line 84 in detect_raw
  File "/opt/frigate/frigate/edgetpu.py", line 160 in run_detector
  File "/usr/lib/python3.8/multiprocessing/process.py", line 108 in run
  File "/usr/lib/python3.8/multiprocessing/process.py", line 315 in _bootstrap
  File "/usr/lib/python3.8/multiprocessing/popen_fork.py", line 75 in _launch
  File "/usr/lib/python3.8/multiprocessing/popen_fork.py", line 19 in __init__
  File "/usr/lib/python3.8/multiprocessing/context.py", line 277 in _Popen
  File "/usr/lib/python3.8/multiprocessing/context.py", line 224 in _Popen
  File "/usr/lib/python3.8/multiprocessing/process.py", line 121 in start
  File "/opt/frigate/frigate/edgetpu.py", line 221 in start_or_restart
  File "/opt/frigate/frigate/edgetpu.py", line 190 in __init__
  File "/opt/frigate/frigate/app.py", line 197 in start_detectors
  File "/opt/frigate/frigate/app.py", line 340 in start
  File "/opt/frigate/frigate/__main__.py", line 17 in <module>
  File "/usr/lib/python3.8/runpy.py", line 87 in _run_code
  File "/usr/lib/python3.8/runpy.py", line 194 in _run_module_as_main
```

agners, 2022-02-21:
Can you try to downgrade the OS, by using:

```
ha os update --version 7.2
```

Kurisutian, 2022-02-21:
> Can you try to downgrade the OS, by using:
> 
> ```
> ha os update --version 7.2
> ```

How far can I downgrade without anything going wrong? I might have to go all the way down to 7.0 and check from there. Do you know if I can do that without killing my installation by any chance?

agners, 2022-02-21:
It is pretty safe.

Downgrading within a release series should be no problem.

Across releases we sometimes make adjustments to the bootloader which can make the fallback fail (e.g. downgrade works, but you won't have a working fallback to the previous release).

That said, taking a backup and downloading is always a good idea :)

Kurisutian, 2022-02-21:
Well I checked all the way down to 7.0 but the error remains. Maybe the OS upgrades were just a co-incident. Anyhow I still don't know what to do to fix the error or where it actually comes from... 🤔

agners, 2022-02-21:
Unfortunately I don't have such a device here right now, so I can test. However, I thought it should work. Issue #1698 sounds like there is only a problem with PCIe?

Anyways, it is on my todo to integrate the latest drivers from Google. I'll let you know once I have development builds.

agners, 2022-02-22:
Is the host log (dmesg) showing something when the above Python error happens?

Kurisutian, 2022-02-23:
Nothing in dmesg... but somehow it seems to work now... I suspect a hardware "defect" somewhere since I took apart the setup and rebuild it which then seemed to have done the trick. So somewhere it seemed there is some loose wire which now is in place again or something. At least I cannot think of anything else what might be responsible as the software now runs fine without any error. 🤷‍♂️

closed 2022-02-23 by Kurisutian

agners, 2022-02-23:
Thanks for the update!

thesebastianf, 2022-10-08:
Well I have a similar issue actually.
I added a Coral USB today.

After only a few minutes it Crashes somehow and is Not listed lsusb in Ubuntu anymore. Also it Crashes my zigbee and zwave stick at this Moment.

Its a nuc .... So No loose cable Here.

Aryankids, 2023-03-07:
I have exactly the same issue.  Running HA on WSL2 on an i7 NUC. Within a couple of hours after attaching the coral I get an unexpected crash with no useful information !!