home-assistant/operating-system #1758: qemu-guest-agent using wrong port -> not starting

wardwouts, 2022-02-22:
### Describe the issue you are experiencing

qemu-guest-agent is via systemd configured to use `/dev/virtio-ports/org.qemu.guest_agent.0`. This does not exist. Instead `/dev/virtio-ports/com.redhat.spice.0` should be used. To change this `/usr/lib/systemd/system/qemu-guest.service` needs to be updated.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. `login` on the VM console
2. `systemctl status qemu-guest.service`
   (it shows not running)
3. try to start it again:
    `systemctl start qemu-guest.service`
4. check status again `systemctl status qemu-guest.service` it shows it can't open a channel `No such file or directory`
5. check `/usr/lib/systemd/system/qemu-guest.service` see non existing path in `/dev`

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
no
```

agners, 2022-02-22:
That depends what you configure. I don't think it makes sense to change that as we might have users out there having it set to `org.qemu.guest_agent.0`.

[image]

wardwouts, 2022-02-22:
Aaaah, I did not know this was configurable. Thanks! This is perfect.

For future lookups, in libvirt it's this bit of the XML:
```
    <channel type='spicevmc'>
      <target type='virtio' name='com.redhat.spice.0'/>
      <address type='virtio-serial' controller='0' bus='0' port='1'/>
    </channel>
```

closed 2022-02-22 by wardwouts