home-assistant/operating-system #1759: Pi Zero 2 W bluetooth not working

holygrolli, 2022-02-22:
### Describe the issue you are experiencing

trying to use/enable bluetooth is not possible.
```
# hciconfig -a
hci0:	Type: Primary  Bus: UART
	BD Address: B8:27:EB:A3:EC:F5  ACL MTU: 1021:8  SCO MTU: 64:1
	DOWN
	RX bytes:1250 acl:0 sco:0 events:66 errors:0
	TX bytes:810 acl:0 sco:0 commands:66 errors:0
	Features: 0xbf 0xfe 0xcf 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
	Link policy: RSWITCH SNIFF
	Link mode: PERIPHERAL ACCEPT

# hciconfig hci0 up
Can't init device hci0: Invalid request code (56)

# systemctl status bluetooth-bcm43xx.service
● bluetooth-bcm43xx.service - Bluetooth for BCM43xx
     Loaded: loaded (/usr/lib/systemd/system/bluetooth-bcm43xx.service; enabled; vendor preset: enabled)
     Active: active (running) since Mon 2022-02-21 19:46:58 UTC; 3min 42s ago
    Process: 335 ExecStart=/usr/bin/btuart (code=exited, status=0/SUCCESS)
   Main PID: 379 (hciattach)
      Tasks: 1 (limit: 866)
     Memory: 240.0K
     CGroup: /system.slice/bluetooth-bcm43xx.service
             └─379 /usr/bin/hciattach /dev/serial1 bcm43xx 3000000 flow - b8:27:eb:a3:ec:f5

Feb 21 19:46:57 homeassistant systemd[1]: Starting Bluetooth for BCM43xx...
Feb 21 19:46:57 homeassistant btuart[369]: Patch not found, continue anyway
Feb 21 19:46:58 homeassistant btuart[369]: bcm43xx_init
Feb 21 19:46:58 homeassistant btuart[369]: Set BDADDR UART: b8:27:eb:a3:ec:f5
Feb 21 19:46:58 homeassistant btuart[369]: Set Controller UART speed to 3000000 bit/s
Feb 21 19:46:58 homeassistant btuart[369]: Device setup complete
Feb 21 19:46:58 homeassistant systemd[1]: Started Bluetooth for BCM43xx.
```
In the [forum](https://community.home-assistant.io/t/bluetooth-not-working-on-pi-zero-2-w/395346/7) there was the idea of a wrong bluetooth firmware for Pi Zero 2 W

### What operating system image do you use?

rpi3 (Raspberry Pi 3 32-bit OS)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. `hciconfig hci0 up`

### Anything in the Supervisor logs that might be useful for us?

```txt
no
```

### Anything in the Host logs that might be useful for us?

```txt
[   11.190646] Bluetooth: Core ver 2.22
[   11.190755] NET: Registered protocol family 31
[   11.190764] Bluetooth: HCI device and connection manager initialized
[   11.190787] Bluetooth: HCI socket layer initialized
[   11.190801] Bluetooth: L2CAP socket layer initialized
[   11.190824] Bluetooth: SCO socket layer initialized
[   11.213827] Bluetooth: HCI UART driver ver 2.3
[   11.213845] Bluetooth: HCI UART protocol H4 registered
[   11.213942] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   11.214136] Bluetooth: HCI UART protocol Broadcom registered
[   11.627113] audit: type=1334 audit(1645472818.552:6): prog-id=9 op=LOAD
[   11.627203] audit: type=1334 audit(1645472818.552:7): prog-id=10 op=LOAD
[   11.719996] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   11.720012] Bluetooth: BNEP filters: protocol multicast
[   11.720036] Bluetooth: BNEP socket layer initialized
```

### System Health information

## System Health

version | core-2022.2.9
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.92-v7
arch | armv7l
timezone | UTC

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4990
Installed Version | 1.23.0
Stage | running
Available Repositories | 1062
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 58.9 GB
disk_used | 3.3 GB
healthy | true
supported | true
board | rpi3
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.0.1), Terminal & SSH (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

### Additional information

The Pi Zero 2 W has attached Waveshares' Ethernet / USB HUB HAT and is connected via ethernet. Wifi is not in use.
The scenario was also tested with a fresh install of RaspberryPiOS (lite). Bluetooth is working there.

GitPetri, 2022-03-10:
BT and GPIO don't work very well in co-operation with default config. Please try setting these in [/boot/config.txt](https://community.home-assistant.io/t/using-bluetooth-and-dtoverlay-pi3-miniuart-bt/63971/8):
```
dtoverlay=miniuart-bt
core_freq=250
```

closed 2022-06-15 by github-actions[bot]

kpouget, 2023-10-19:
> BT and GPIO don't work very well in co-operation with default config. Please try setting these in [/boot/config.txt](https://community.home-assistant.io/t/using-bluetooth-and-dtoverlay-pi3-miniuart-bt/63971/8):
> 
> ```
> dtoverlay=miniuart-bt
> core_freq=250
> ```

late answer to this issue, but for information these ^^^ changes resolved my BT issue on the `Raspberry Pi Zero W Rev 1.1`
thanks :)

GitPetri, 2023-10-20:
> late answer to this issue, but for information these ^^^ changes resolved my BT issue on the `Raspberry Pi Zero W Rev 1.1` thanks :)

@kpouget Please be aware that for some reason those lines config.txt make OS upgrades fail (not limited to HA-OS, also seen on 'normal' RasPi Linux upgrades) . So when doing OS upgrade:
- Shutdown HA
- Modify config.txt and comment out the lines above
- Start HA, initialize OS upgrade
- After OS upgrade shutdown HA
- Uncomment the lines above
- Start HA

kpouget, 2023-10-20:
thanks @GitPetri , I'll pay attention to it!