home-assistant/operating-system #1760: Same exact SSD imaged with 7.4 can boot if connected via external USB but not from internal SATA

phretor, 2022-02-22:
### Describe the issue you are experiencing

I verified with multiple experiments that if I image an SSD with HA OS 7.4 (also tried 8.0-dev and 6.6) and connect it via the SATA port, the disk seems not imaged in a way that the BIOS recognizes its UEFI partition. So, it doesn't boot: in the best case I see a black screen.

It can boot if I disconnect the SSD from the SATA slot and connect it to the external USB3 port: in that case, the BIOS sees its UEFI partition correctly.

After a long discussion (and support: thanks folks!) in the #installation Discord channel, I decided to file an issue.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.4, 8.0-dev, 6.6

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Point BalenaEtcher to https://github.com/home-assistant/operating-system/releases/download/7.4/haos_generic-x86-64-7.4.img.xz (tried with 8.0-dev and 6.6 as well)
2. Burn the image onto a SATA SSD
3. Try to boot from SATA SSD on a Fujitsu Esprimo q920 (i5-4590t) with default BIOS settings (no Secure Boot, UEFI boot, etc., was able to boot a standard Debian 11)
4. Try to boot while the *same* SATA SSD drive is connected via the external USB3 port (via a SATA-to-USB connector).

### Anything in the Supervisor logs that might be useful for us?

```txt
Black screen
```

### Anything in the Host logs that might be useful for us?

```txt
Black screen
```

agners, 2022-02-22:
Hm, that is weird. If a BIOS is able to read a GPT via USB 3.0, it really should be able to also read it via S-ATA.

Here some things I would check:
- Replace/verify S-ATA cable (just in case)
- Check BIOS settings (make sure that not RAID mode is enabled, S-ATA AHCI mode should be preferred, also make sure that no HDD encryption/password or similar options are set).
- Reset BIOS to default and check settings again
- Maybe try with a BIOS update

phretor, 2022-02-22:
Found another user having my very same problem: https://community.home-assistant.io/t/can-boot-from-ssd-hdd-via-usb-but-not-internal-sata-on-x86-laptop/366600/5

agners, 2022-02-23:
Oh ok, so the system is able to read the GPT and load the boot loader from the disk. The problem is that the bootloader (which is part of Home Assistant OS) then has troubles to load the kernel, it seems. The culprit are those two messages:

```
mount: Invalid argument
could not open /mnt/system/bzImage: No such file or directory
```

Not sure what causes this, it could be still a BIOS setting or just a bug in barebox.

Can you maybe try disabling CSM/legacy mode? HAOS should boot in pure UEFI mode.

I am actually considering switching to Grub for all UEFI platforms, which likely would help solve this.

phretor, 2022-02-23:
> 

> Oh ok, so the system is able to read the GPT and load the boot loader from the disk. The problem is that the bootloader (which is part of Home Assistant OS) then has troubles to load the kernel, it seems. The culprit are those two messages:

It depends. From the external USB thumbdrive, that is what I see. But if I use the SSD via SATA, the BIOS doesn't even see it as a disk with a UEFI partition. The *only* way I found working is to plug the SSD via USB (with a SATA-to-USB converter).

> ```
> mount: Invalid argument
> could not open /mnt/system/bzImage: No such file or directory
> ```
> 
> Not sure what causes this, it could be still a BIOS setting or just a bug in barebox.
> 
> Can you maybe try disabling CSM/legacy mode? HAOS should boot in pure UEFI mode.

If I disable it, I don't see the SSD anymore (not even as a legacy drive).

> I am actually considering switching to Grub for all UEFI platforms, which likely would help solve this.

Anything else I could try?

agners, 2022-02-23:
> It depends. From the external USB thumbdrive, that is what I see.

Oh I see that log/screenshot was with a USB thumbdrive.

> But if I use the SSD via SATA, the BIOS doesn't even see it as a disk with a UEFI partition.

Do other operating system using UEFI/GPT work on that machine?

Is it maybe that S-ATA drive which cannot talk with that controller (properly) somehow?

> > Can you maybe try disabling CSM/legacy mode? HAOS should boot in pure UEFI mode.

> If I disable it, I don't see the SSD anymore (not even as a legacy drive).

Hm, that could be the root of the problem. I am pretty sure that CSM off should really work with GPT and Barebox.

phretor, 2022-02-23:
> > It depends. From the external USB thumbdrive, that is what I see.
> 
> Oh I see that log/screenshot was with a USB thumbdrive.

I've ordered a new USB3 thumbdrive and I'll do more tests today, to rule out the possibility that this may be due to a bad memory.

> > But if I use the SSD via SATA, the BIOS doesn't even see it as a disk with a UEFI partition.
> 
> Do other operating system using UEFI/GPT work on that machine?

I used [this procedure](https://alexlubbock.com/bootable-windows-usb-on-mac) to create a bootable USB3 thumbdrive containing the Win10x64 installer. I installed it onto the SSD connected via the internal S-ATA, and at boot this was properly recognized as UEFI by the BIOS.

I also installed the latest Debian 11 using a similar method, using the installer's default settings.

> Is it maybe that S-ATA drive which cannot talk with that controller (properly) somehow?
>
>
>
> Hm, that could be the root of the problem. I am pretty sure that CSM off should really work with GPT and [...]

Yes, I feel this has to do with it, but the fact that Win10 and Debian 11 were able to boot...confuses me.

agners, 2022-02-23:
> Yes, I feel this has to do with it, but the fact that Win10 and Debian 11 were able to boot...confuses me.

Hm, our partition table isn't really anything special, its being created with `sfdisk`, nothing unusual. We do have some more partitions to store a redundant OS (A/B scheme), but the main partition which should be relevant for the UEFI BIOS is the ESP (EFI System Partition), which is pretty standard:

```
label: gpt
label-id: 45AF3392-95D6-440E-A59B-F407122A754A
device: haos_generic-x86-64-8.0.dev20220208.img
unit: sectors
first-lba: 34
last-lba: 12582878
sector-size: 512

haos_generic-x86-64-8.0.dev20220208.img1 : start=        2048, size=       65536, type=C12A7328-F81F-11D2-BA4B-00A0C93EC93B, uuid=B3DD0952-733C-4C88-8CBA-CAB9B8B4377F, name="hassos-boot"
haos_generic-x86-64-8.0.dev20220208.img2 : start=       67584, size=       49152, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=26700FC6-B0BC-4CCF-9837-EA1A4CBA3E65, name="hassos-kernel0"
haos_generic-x86-64-8.0.dev20220208.img3 : start=      116736, size=      524288, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=8D3D53E3-6D49-4C38-8349-AFF6859E82FD, name="hassos-system0"
haos_generic-x86-64-8.0.dev20220208.img4 : start=      641024, size=       49152, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=FC02A4F0-5350-406F-93A2-56CBED636B5F, name="hassos-kernel1"
haos_generic-x86-64-8.0.dev20220208.img5 : start=      690176, size=      524288, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=A3EC664E-32CE-4665-95EA-7AE90CE9AA20, name="hassos-system1"
haos_generic-x86-64-8.0.dev20220208.img6 : start=     1214464, size=       16384, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=33236519-7F32-4DFF-8002-3390B62C309D, name="hassos-bootstate"
haos_generic-x86-64-8.0.dev20220208.img7 : start=     1230848, size=      196608, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=F1326040-5236-40EB-B683-AAA100A9AFCF, name="hassos-overlay"
haos_generic-x86-64-8.0.dev20220208.img8 : start=     1427456, size=     2621440, type=0FC63DAF-8483-4772-8E79-3D69D8477DE4, uuid=A52A4597-FA3A-4851-AEFD-2FBE9F849079, name="hassos-data"
```

phretor, 2022-02-23:
@agners I've just flashed a brand-new USB3 stick with HAOS Generic x86-64 7.4 and I can see a white screen with `Autoboot`. If I choose "shell" I enter the Barebox shell, and if I type `boot` I get to this menu.

```
Enter exit to get back to the menu

HAOS-boot:/ boot
Booting entry
"bootchoosen
bootchooser: No valid targets found:
A
id: 1
priority: 20
default priority:
remaining attempts: 0
default attempts: 3
boot: "system0 disabled due to remaining attempts = 0

B
id: 2
priority: 10
default _priority: 1
remaining attempts:
default attemots: 3
boot: 'system1'
disabled due to remaining attempts =0
Booting entry "bootchooser failed
Nothing bootable found
HAOS-boot:/
```

[image]

agners, 2022-02-23:
@phretor the USB boot support for x86-64 with the Barebox boot loader is relatively new and got added with #1575. 

That said, I am working on a patch to replace Barebox with Grub as we speak. We should have some development builds tomorrow.

phretor, 2022-02-23:
Awesome news @agners 

Since I was able to boot the SSD only via USB, I've always (wrongly) assumed that booting from USB was the most stable and reliable way.

Now I learn it's the opposite 🤭

I'll stay tuned here and will test immediately as the build is ready.

phretor, 2022-02-24:
@agners I just noticed your latest 2 commits: they should have been part of the latest dev build, right?

https://os-builds.home-assistant.io/8.0.dev20220224/haos_generic-x86-64-8.0.dev20220224.img.xz

agners, 2022-02-25:
Yes, the latest dev build should now include GRUB as boot loader.

phretor, 2022-02-25:
@agners I tried this morning with https://os-builds.home-assistant.io/8.0.dev20220224/ but I think I'm gonna wait for https://os-builds.home-assistant.io/8.0.dev20220225, because yesterday's built still had Barebox.

agners, 2022-02-25:
No, [8.0.dev20220224.img.xz](https://os-builds.home-assistant.io/8.0.dev20220224/haos_generic-x86-64-8.0.dev20220224.img.xz) already has Grub 2 as boot loader.

phretor, 2022-02-25:
Still nothing. The SSD isn't recognized as UEFI. I suspect it's the BIOS' fault, but I wonder how other UEFI OSs manage to boot. Is there any other UEFI OS image (Linux based, hopefully) that is not HA OS, which I can use to flash the SSD and see how it behaves?

phretor, 2022-03-04:
Gentle bump @agners :-)

luciotorelli, 2022-03-18:
Same issue here, trying to run on an ATIV Book 6 670Z5E-XD1 which supports UEFI boot.

phretor, 2022-03-18:
@luciotorelli have you tried with a dev build?

luciotorelli, 2022-03-18:
> @luciotorelli have you tried with a dev build?

Yes, I tried on [8.0.dev20220224.img.xz](https://os-builds.home-assistant.io/8.0.dev20220224/haos_generic-x86-64-8.0.dev20220224.img.xz) linked above as well as the production one. All sorts of combination as possible from Bios xD. The same machine already ran Linux, Debian, ChromeOS, Windows 7, 10 and 11 at some point so I am nearly certain it isn't the machine. I will try with [dev20220225](https://os-builds.home-assistant.io/8.0.dev20220225/haos_generic-x86-64-8.0.dev20220225.img.xz) and report here.

phretor, 2022-03-18:
@luciotorelli I also tried Debian and Windows 10, but I can't be certain the installer used pure UEFI or some UEFI+legacy fallback trick to maximize compatibility.

Are you aware of a minimal working OS that uses pure UEFI? That would be a good test case.

This doesn't mean that HA OS shouldn't use the same legacy fallback tricks to broaden its compatibility. At the end of the day, it's still a (stripped) Linux distro.

hami89, 2022-03-20:
I have the exact same problem.
I have an Asrock 1800b-itx motherboard with built in 64bit intel atom CPU. Bios is UEFI capable, upgraded to latest version, SATA AHCI mode enabled, BIOS/compatibility modes disabled (I also tried all linear combinations of the available settings.).

I use a 250 Gb m2 sata SSD in an m2-->sata converter, attached to sata port no. 1.
UEFI Bios can see the SSD as a SATA disk correctly, but does not recognize it as a UEFI boot target.

If I use a USB stick with ubuntu, I can boot the machine with UEFI just fine, and I can also see the SSD. (Also, ubuntu recognizes the SSD as an EFI bootable OS disk).

I tried to flash HAOS 7.5 image to the SSD with another machine running WIN 10 and using a USB-m2sata adapter, same result.
I also tried to flash the image on the machine using live ubuntu and balena etcher, same result.

At the end, I ran a boot-repair utility from the ubuntu live system. The result looks fine for me, but maybe somebody else can find something interesting in it:
[boot_repair_dump.txt](https://github.com/home-assistant/operating-system/files/8311046/boot_repair_dump.txt)
.

christoph00, 2022-03-24:
I fixed the issue by running a live linux distro, mounting part 1 of the haos disk and added a EFI entry:
efibootmgr --create --disk /dev/sda --part 1 --label "HAOS" --loader \\EFI\\BOOT\\bootx64.efi

phretor, 2022-03-30:
@agners I confirm that @christoph00 fixes the issue. I guess that this could be added to the docs and/or in the builder.

closed 2022-03-30 by phretor

agners, 2022-03-30:
Hm, those are interesting findings. This means that this UEFI BIOS behaves differently depending on media type (removable e.g. USB drive vs. internal disk). Most systems just boot from a file `\EFI\BOOT\bootx64.efi` if it is present on any media.

Regular Linux distribution have an installer which creates the EFI entry at installation time. We don't have such a installer (yet). This issue we can only really resolve if we have an installer of some kind :worried:

DonKracho, 2022-04-23:
I have the same issue with an ASRock J1900-ITX board (BIOS is the latest available version 2.0). The SSD won't boot when attached to a SATA port.   When disabling CSM in the BIOS there will be no boot device at all.  Inserting the same SSD into an external SATA to USB3.0 enclosure there are two boot devices USB: EnclsureName (won't boot too) and UEFI: EnclsureName (this does boot into HassOS fine). Somehow I have to tell the ASRock J1900-ITX BIOS that it is an UEFI boot device. 
Unfortunately there is no such option when atached to SATA. B. t. w.: With an much older HP Thin Client T610 this disk will boot, But that's is no option to me because the HP draws 3 times more power than the ASRock board. Therefore the boot fail should be most likely a BIOS issue of the ASRock J1900-ITX.

With an USB thumbdrive flashed with an UEFI version of gparted live I did the command line recommended by @christoph00 with 'sudo' in front of it. This still did not help. I get an additional boot menu entry HAOS, but none ot the boot menu entries does boot. 

What am I doing wrong? Is there any way to check if the efibootmgr command did succed?

DonKracho, 2022-04-23:
OK, I'm a noob to this UEFI boot options and just learned more about the efibootmgr.
The path to the loader is wrong. A simplified command working for me is:

sudo efibootmgr -c -l /EFI/BOOT/BOOTx64.EFI -L HomeAssistant

agners, 2022-04-25:
Hm, it seems that Barebox (used in OS 7.x and before) uses `/EFI/BOOT/BOOTx64.EFI`, whereas 8.x uses `/EFI/BOOT/bootx64.efi`. I don't think that UEFI is case sensitive, but it might be that the `efibootmgr` command itself is case sensitive.

agners, 2022-04-25:
There is some related discussion in this Ubuntu issue: https://bugs.launchpad.net/ubuntu/+source/grub2/+bug/1366546

It seems that mostly older systems affected, presumably newer systems use a UEFI bios which read from /EFI/BOOT also on internal disks.

Calimerorulez, 2022-07-07:
I still have this problem on my Futro S920. I've managed to make a workaround, by booting a Linux USB stick, starting grub. Then set root to (hd0,gpt1). Next typing chainloader /EFI/BOOT/bootx64.efi. Then typing boot.

Atti992, 2022-08-25:
Hello all,

I have the same issue. I bought a Fujitu S740 thin client and I'd like to install HAOS on the inner SSD (16GB).
I do the same steps:
https://www.home-assistant.io/installation/generic-x86-64
Symptoms:
IF I put the M.2 SSD in a M.2 USB adapter and connenct to the client via USB the HAOS start and very fast and working without any error.
IF I put the M.2 SSD inside the thin client ( to the original place) there is no bootable media...
Could you help please how can I solve this issue?

Thanks a lot for reading this message

NCO3, 2022-09-07:
@Calimerorulez 
Is this a permanent fix or do you need to repeat this on every boot cycle?
Thanks

smith113311, 2022-09-07:
I am in the same spot as @Atti992 .

agners, 2022-09-07:
@Atti992 @smith113311 did you execute the commands mentioned in the Note at https://www.home-assistant.io/installation/generic-x86-64#start-up-your-generic-x86-64?

[image]

NCO3, 2022-09-07:
I did, but did not succeed.
I started with a brand new image ion the internal disk and restored a backup from the Ian drive.
Thanks for your quick response, though!

Calimerorulez, 2022-09-08:
> @Calimerorulez Is this a permanent fix or do you need to repeat this on every boot cycle? Thanks

@NCO3 
Hi,
If I remember correctly, I made a grub.conf with the commands in it so it executes at boot. But I'm not using the Futro anymore, because it was super slow for my purposes. So I've moved on to another PC, I can't give you my grub config then...

rpsh1919, 2022-10-11:
> OK, I'm a noob to this UEFI boot options and just learned more about the efibootmgr. The path to the loader is wrong. A simplified command working for me is:
> 
> **sudo efibootmgr -c -l /EFI/BOOT/BOOTx64.EFI -L HomeAssistant**
> 
> Before I did a **sudo mount /dev/sda1 /mnt** and used **ls /mnt** to verify the path (may be it is case sensitive ?)

OMG! Thank you very much!! This solved my problem with Fujitsu Q920!

RenEdi, 2022-10-27:
Please, how do I do this? Where to enter that command "sudo efibootmgr -c -l /EFI/BOOT/BOOTx64.EFI -L HomeAssistant"?

DonKracho, 2022-10-27:
Yes, prepare a bootable device (e.g. USB Stick) with the gparted live image and boot from it. Then open a terminal and execute the command. The trick is that the gparted live image has the efibootmgr included. gparted itself isn't used. I just downloaded the iso image from https://gparted.org/download.php and installe it tu the stick with rufus https://rufus.ie/de/

RenEdi, 2022-10-27:
thanks, but 
I made bootable USB stick as you wrote, enter that command in Terminal "sudo efibootmgr -c -l /EFI/BOOT/BOOTx64.EFI -L HomeAssistant" and the result
"bash: /efi/boot/bootx64.efi: No such file or directory  EFI variables are not supported on this system"

RenEdi, 2022-10-29:
thank you, now everything works
I had to first  "sudo apt-get install efibootmgr"
and then...
sudo efibootmgr -c -l /EFI/BOOT/BOOTx64.EFI -L HomeAssistant

dudududodododedede, 2022-12-09:
Thanks, this worked for me on the Lenovo M82e. I already had a USB key with Ubuntu 22.04 lying around, which supports Live Boot (no install of ubuntu necessary) and it already comes with efibootmgr. Didn't have to mount my main drive or anything, just ran that efibootmgr command in terminal. Then I removed the USB key, rebooted, and HA boots without issue now

kapciuch, 2023-01-07:
> Yes, prepare a bootable device (e.g. USB Stick) with the gparted live image and boot from it. Then open a terminal and execute the command. The trick is that the gparted live image has the [...]

Hi, I'm probably very thick. But I managed to flash USB with gdparted, I have small success that for the first time  booted Home Assistant from SSD drive by using gdparted and selecting operating system. But how to make it to boot that as default. I can't use command sudo because it doesn't recognise the command?

KESchnitzler, 2023-01-16:
hi, having the same probs with my ESPRIMO Q920, the internal SSD w. HASSOS doesn't boot up, connected to usb it works fine, SATA none. Tried it with ubuntu AND gparted to solve it with efibootmgr with .../bootx64,efi AND .../BOOTx64.EFI none of the trials made the SSD boot. how shall I proceed??? usb-connection is not an option... maybe someone may help me out??!! regs Mike

luciotorelli, 2023-01-16:
> hi, having the same probs with my ESPRIMO Q920, the internal SSD w. HASSOS doesn't boot up, connected to usb it works fine, SATA none. Tried it with ubuntu AND gparted to solve it with efibootmgr [...]

Hey Kes,

I was having the same issue using the code they provided here (although I see some people were successful). I reset my SD bootable card and the hard driver, I followed [this YouTube video](https://www.youtube.com/watch?v=nKvIYKCg8I4).

After HA was installed and I couldn't boot, I created a USB stick with Gparted as [DonKracho](https://github.com/DonKracho) mentioned. Logged in > when asked to login, exit or use command, I selected use command (2). but instead of issuing the code mentioned here, I used the command on the home assistant generic documentation:

> 
> If the machine complains about not being able to find a bootable medium, you might need to specify the EFI entry in your BIOS. This can be accomplished either by using a live operating system (e.g. Ubuntu) and running the following command (replace <drivename> with the appropriate drive name assigned by Linux, typically this will be sda or nvme0n1 on NVMe SSDs):
> 
```
efibootmgr --create --disk /dev/<drivename> --part 1 --label "HAOS" \
   --loader '\EFI\BOOT\bootx64.efi'
```

After months trying I was finally able to make my old Laptop HA. Wish you best of luck!

kapciuch, 2023-01-16:
Hi All,
After few days fight I found the solution. Flash any usb memory stick you don't need with https://www.plop.at/en/pbm6/efi.html boot manager using https://rufus.ie/en/ .
Then you stick that usb stick into usb slot on your server PC, and make sure it boots into that usb stick. Then you can set in pbm6 boot manager (the one on usb stick) where to boot it from (SSD connected to SATA) and also how long it should wait until it boots from there (I set 1 sec).
Hopefully it will work for many of you!

KESchnitzler, 2023-01-16:
to make a long story short: the problem was, that you have to boot the usb-stick in uefi mode, if not the changes on the boot partition wil not be made. so if you start the esprimo q920 with F12 into the boot selection menue you have to make shure, that the sitck shows up with an UEFI: remark in the list. choosing this option and executing the given efibootmgr cmdline wil result in a bootable internal SSD. maybe it helps for future installers... ;-) Regs K-E

kopfsick, 2023-01-21:
I'm trying to do this with GParted, but I'm just getting "EFI varibles are not supported on this system" when trying to run the commands in terminal. Any ideas?

artpanek, 2023-02-17:
> OK, I'm a noob to this UEFI boot options and just learned more about the efibootmgr. The path to the loader is wrong. A simplified command working for me is:
>
> **sudo efibootmgr -c -l [...]

YES, thats worked also on my Fujitsu ESPRIMO Q956 (BIOS was updated to V.5.0.0.11 – R1.33.0 (27.01.2023) )
Thank you!

GKG-ctrl, 2023-05-23:
> OK, I'm a noob to this UEFI boot options and just learned more about the efibootmgr. The path to the loader is wrong. A simplified command working for me is:
>
> **sudo efibootmgr -c -l [...]

@DonKracho  
Thank you very much!, I just repurposed my 2009 MACBOOK PRO!