home-assistant/operating-system #1764: Debugging random hangs, might have stumbled over a memory leak

sgofferj, 2022-02-24:
### Describe the issue you are experiencing

I'm trying to debug random freezes of my HA-OS running on an RPi3B. I'm trying to document my findings in the forum here: https://community.home-assistant.io/t/debugging-random-slowdowns-and-system-freezes-on-an-rpi3b/394778/4

In any case, from the latest freeze I finally got some reading from telegraf, one of which was the HA container allocating 2.67 Exabytes of RAM. Obviously that can't be serious, so I'm thinking, maybe there's a memory leak somewhere? I never had any issues running HA with the same configuration in a docker container on the same RPi. I just recently moved to HA-OS, hoping for a better experience.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Unknown

### Anything in the Supervisor logs that might be useful for us?

```txt
Rebooted system since last freeze for configuration changes. Will supply logs if freeze occurs again.
```

### Anything in the Host logs that might be useful for us?

```txt
Rebooted system since last freeze for configuration changes. Will supply logs if freeze occurs again.
```

### System Health information

## System Health

version | core-2022.2.9
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.7
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Helsinki

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.23.0
Stage | running
Available Repositories | 1060
Downloaded Repositories | 35

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.4
-- | --
update_channel | stable
supervisor_version | supervisor-2022.01.1
docker_version | 20.10.9
disk_total | 116.7 GB
disk_used | 15.7 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Samba share (9.5.1), Terminal & SSH (9.3.0), Node-RED (10.1.1), File editor (5.3.3), Telegraf (0.6.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 7
-- | --
resources | 24
views | 16
mode | storage

</details>

### Additional information

As I said, I'm trying to debug this at the moment. Not an easy task, because I can't reproduce it and when it happens, the system is not accessible.

How about you add a remote logging option to the system config? I have a log server in my home lab where I could send the logs via syslog. THAT would increase observability massively...

agners, 2022-02-24:
2.67 exabyte is not caused by a memory leak, that must be a wrong reading of some type.

I like to use the system monitor integration (needs yaml though), as it is more lightweight: https://www.home-assistant.io/integrations/systemmonitor/.

From what I can tell from your description, you run quite a bit of stuff on a Raspberry Pi 3. The Pi 3 has only 1GB of RAM, running, and from your graph  it seems that all container summed up come pretty close to that. I assume that your system just runs too close at its limit, and that under some circumstances (Core and Supervisor processes something and need memory at the same time) the system ends up in OOM situation.

Granted, the OOM handling could be better in Home Assistant. Ideally we should inform the user when it happened, and make recommendations.

sgofferj, 2022-02-24:
I'll have a look at this integration, thanks. And yes, the OOM handling could be improved... For example, automatic coredumping could be disabled. If a coredump freezes my Ryzen 7 desktop for minutes, I don't wanna imagine what it does to a little RPi.
I don't actually run that much stuff on the Pi. HA, NodeRED, File editor and telegraf. RAM usage in normal operatins is around 80%.
[image]

The thing that irritates me most is that I had HA running in a plain docker setup in Ubuntu Server on exactly the same PI before without any issue.

I'll continue monitoring and debugging. Feel free to close the issue. I just wanted it to exist. I can always ask to reopen it if I have new results.

agners, 2022-02-24:
> If a coredump freezes my Ryzen 7 desktop for minutes, I don't wanna imagine what it does to a little RPi.

Afaik, just a dump should be a rather quick thing as its just writing the contents of memory to a location on disk. Why is it taking so long? Is what distribution are you using/what kind of coredumping?

That said, what I noticed in Linux in general that sometimes when memory get scarce, just before the OOM killer comes in, the system can hang quite a bit (as it scrambles to try to free up memory). But this is unrelated to coredumping itself. Although, doing coredumping can of course require more memory, which in turn can lead to memory management to scramble for memory (and hence lengthen the process).

> RAM usage in normal operatins is around 80%.

Effectively used memory at 80% is rather high. Keep in mind that buffer/caches are not included, and to have a responsive system you want some memory free for that as well.

> The thing that irritates me most is that I had HA running in a plain docker setup in Ubuntu Server on exactly the same PI before without any issue.

Was that with swap enabled? What version of Ubuntu (32-bit/64-bit?)

sgofferj, 2022-02-25:
After removing now also the geo_location entity and rebooting, the system seems to run a little smoother again. Is there anything scheduled to running every 90 mins or so in the system? I don't think I have anything configured in this interval.
[image]

agners, 2022-03-01:
On OS level I am not aware of a service running every 90 minutes.

sgofferj, 2022-03-04:
Last night it hung again, although no OOM killer involved. As far as I could tell, it was just swapping hell. Now I disabled meteoalarm (which doesn't work anymore anyways), Blitzortung and the stock weather app. My observation is that the memory is filling up slowly over time, so I guess, something is either caching stuff or not cleaning up properly after itself.
But kicking those 3 apps reduced after-boot memory usage by almost 10%.

sgofferj, 2022-03-05:
Well, I tend to declare this unfixable due to platform limitations. I managed to score an RPi4 8GB which I will move the system to.
@agners I'll close this but I'd like to leave some suggestions:
- Improve observability, e.g. allow configuration of a remote log server.
- Provide a way to get remote access to bare metal root. If the supervisor is choked by IO or load, supervised ssh access doesn't help much.
- Disable core dumps on OS level by default. Nobody really needs them except hardcore developers working on very specific problems.  Coredumps could be a developer option in the system config.

closed 2022-03-05 by sgofferj