home-assistant/operating-system #1775: Unable to change keymap when directly connected to host

friggeri, 2022-03-01:
### Describe the issue you are experiencing

I have a keyboard and small monitor plugged into the NUC on which I run Home Assistant OS (that way I can directly use the CLI in the event I have a networking issue). Unfortunately there seems to be no way of selecting the keyboard layout.

The OS is built with `loadkmap` [enabled](https://github.com/home-assistant/operating-system/blob/ae0aeb84f513d877e402d38bd547555bd67f0b22/buildroot-external/busybox.config#L378) and I have confirmed that I can manually load a new keymap by logging into the host system and running `loadkmap < mykeymap.bmap`. 

Unfortunately this does not persist across reboots. Is there any way of making this configurable? Alternatively, as a short term fix, is there a way to get the OS to execute the above command on startup?

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Plug AZERTY keyboard, 
2. Type A
3. See  Q

### Anything in the Supervisor logs that might be useful for us?

```txt
n/a
```

### Anything in the Host logs that might be useful for us?

```txt
n/a
```

agners, 2022-03-01:
Don't think it was the intention to make that possible really. Usually the console is only meant for debugging.

friggeri, 2022-03-01:
Totally get that, it's pretty hard to debug when what you type on the keyboard does not match what shows up on the screen though

friggeri, 2022-03-03:
@agners an easy workaround would be to enable users to customize `/root/.profile` in a way similar to the [way](https://github.com/home-assistant/operating-system/blob/ae0aeb84f513d877e402d38bd547555bd67f0b22/Documentation/configuration.md
) it works for `/root/.ssh/authorized_keys` 

Would you be open to that?

agners, 2022-03-03:
Is the busybox shell evaluating that by default?

I guess that would be one of the simplest approach to archive your goal. 

I still feel its not worth the effort. How many times do you need to use the console, and even when, is it that big of a deal to use the US keyboard layout?

friggeri, 2022-03-03:
Yes, ash is run as a login shell [here](https://github.com/home-assistant/operating-system/blob/40b4d5ca2e6f59a7ada953c58c66c0d236aa98c2/buildroot-external/rootfs-overlay/usr/sbin/hassos-cli#L24) so it will read `/etc/profile/ and `/root/.profile`.

And sure, it's not one of those things that happens a lot, but it would really increase my quality of life (and I guess at least [one other person](https://github.com/home-assistant/operating-system/issues/517)). 

I'm happy to send over a PR!

warlordattack, 2022-07-16:
I am fighting with Hyper-V HA OS.
I swithed from docker container to hyper-v because no Add-ons store in docker

I try to debug DNS problems :
Logger: async_upnp_client.ssdp
Source: /usr/local/lib/python3.10/site-packages/async_upnp_client/ssdp.py:255
First occurred: 14:34:52 (20 occurrences)
Last logged: 14:52:52
Received error: [Errno 101] Network unreachable, transport: <_SelectorDatagramTransport fd=19 read=polling write=<idle, bufsize=0>>

My Addon-store is infinite spin.

And i need add trusted proxies in configuration.yaml (HA OS acces from the web via proxy) :
https://community.home-assistant.io/t/unable-to-connect-to-home-assistant-via-nginx-reverse-proxy/382937/3

No way to install vs-code becose no store access,
no way to edit configuration.yaml

I am stuck in command line, and i am french, so i have fu**ing french keyboard,
and no way to change keyboard to french !

All non english planet needs this !
Please do something.

And no way to copy-paste text from main PC to hyperv HA OS command line.

Fredyy90, 2022-11-29:
any update on this issue? 
I just ran into the same issue, can't debug, as keyboad input and on screen characters don't match

precchia, 2022-12-14:
Pinging to see if there is any update on this issue?
I do not own an US keyboard either, and it is pretty frustrating when trying to access the console.

frederickjh, 2022-12-15:
> is it that big of a deal to use the US keyboard layout?

To make a touch typist to have to "hen peck" the keyboard, is a big deal, especially in a situation where you just want to concentrate on debugging not finding the next key to hit in the correct order.

We have a menu item to configure audio and several other items. There should really by a keyboard or keymap menu item in the cil to configure the keyboard layout. This issue is not the only one that has been opened for this. Others have been closed.

frederickjh, 2022-12-15:
If the command to change the keyboard was to get added to the cli then this issues should get moved to [home-assistant/cli](https://github.com/home-assistant/cli).

n4w4k, 2023-02-24:
What would be the simplest way to fix that? I do not want to create an insane amount of work for that, so I don't need a "pretty" way to do it, I'd be happy to manually edit files to make it work!

But as of now, not only is there no install-time option or GUI option for this, but since the filesystem is read-only for the most part, we can't even upload our own keymap and add a .profile command...

Those saying that it's not a big deal should really try using a CLI with another layout, and tell us their experience afterwards !
When you are in the CLI, it's because you are already struggling with something, there is no need to add another layer of frustration !

bartplessers, 2023-04-12:
+1
ability to change keyboard layout will be much appreciated!

CypherConjured, 2023-05-26:
I agree, key remapping is crucial for accessibility and personal efficiency for users familiar with alternative layouts like Dvorak. That's why modern OS's often prioritize keyboard layout configuration during setup.

scls19fr, 2023-06-24:
+1
I'm also looking for this feature (especially to be able to connect HA behind a proxy).
Ideally a configuration file which could be stored on micro SD card of RaspberryPi could help to store default keymap settings and probably some other default settings

gsiscotty, 2023-07-02:
+1
this is mandatory, i'm also stuck on cli, and can't edit anything correctly...

erik3rik, 2023-07-24:
There should be an EASY way to change the keyboard layout of the OS.
Arguments like "well it is only for debugging, you should not need that" as seen above is totally wrong.

Try debugging when single and double quotes, slash, underscore, equals sign and asterisk is NOT on the right keys.

`cd some/path` (ohh no, not possible, can't use slash!)
`ls -la` (ohh no, not possible, can't use dash!)
`cat *.log` (ohh no, not possible, cant use asterisk)

So, please make a way to change (and persist!) the keyboard layout to the layout preferred by the user. Preferably also without having to type special characters to do so. E.g. `ha keyboard set sv` for Swedish, `ha keyboard set en-us` etc...

mscott370, 2023-09-10:
Agreed! This has been a tiresome issue for everyone who doesn't have a QWERTY keyboard, and the complaints have kept bouncing back since 2019. Hope one of these suggestions gets implemented.

mdeweerd, 2023-10-01:
Maybe @jlpouffier could help give some more priority to this as a French Product Manager for HA.

https://docs.alpinelinux.org/user-handbook/0.1a/Installing/manual.html

```
apk update
apk add -s kbd-bkeymaps alpine-conf busybox-openrc
setup-keymap
# or
setup-keymap fr fr
```

I will try that with a 'shell_script' when I can.

lenamariahackl, 2023-12-05:
+1
ability to change the keyboard layout will be highly appreciated !

Krolow93, 2023-12-12:
+1
It's a pain having to type with the onscreen keyboard since i'm unable to change keymap to match the real keyboard, runing haos as a VM in proxmox.

OlwinFroon, 2023-12-20:
Pretty pretty pleeeeeeease with cream and a cherry on top?

This is a real PITA for anyone who doesn't use an US keyboard.

I'm aware the console is only for debugging, but debugging with a wrong keymap adds a really annoying level of complexity to it.

Baxxy13, 2023-12-20:
+1
Debugging with non QWERTY Keyboard is really annoying.

stex66, 2024-01-06:
+1
please fix it

agrimal, 2024-01-07:
+1 please this is really needed

Einar-S, 2024-01-07:
I cannot find the required " anywhere on my norwegian keyboard. So I am seriously painted into a corner! Please tell me how I can find it.

Impact123, 2024-01-07:
You can use something like this as reference but this is not the right place to ask for help with this.
[image]

OlwinFroon, 2024-01-08:
Considering the age of this issue, you may have a point and this is not the right place.

What would be?

Nickduino, 2024-01-08:
> is it that big of a deal to use the US keyboard layout?

Yes it is when you don't own a QWERTY keyboard. You get used to the moved letter pretty easily but all the special characters are an issue. And since linux commands are full of -/#" etc., that's a real hassle. Worse for me: the Tiny French keyboard I'm using on the console doesn't even have a proper French layout (they put the special characters where they had space), so I can't really rely on a image of the US vs FR layouts.

mdeweerd, 2024-01-08:
> is it that big of a deal to use the US keyboard layout?

+ It's the number of users taking the time to search and then report here kind of a deal.  It also happens when you're already in trouble.  The least that could be done with minimum effort is to show a message "The keyboard layout is US and can not be changed" to avoid that users spend time looking for a solution - it's that big of a deal as well.

MagicJF, 2024-02-12:
requesting the same here, layout change is option is a must

ouinouin, 2024-05-04:
+1 , if there is OS  in the name , it should support keymap :-)

chron-isch, 2024-06-15:
I ran into the same issue trying to debug network issues.

Wouldn't it be realativly easy to just make /etc/vconsole.conf a softlink to something user-writeable?

mscott370, 2024-06-16:
I think the basic issue is the conflict between keyboard support and an OS which aspires to be headless+fingerless+toeless.

OlwinFroon, 2024-06-17:
The console is a last resort kind of tool. 
Which means, you're already in panic if you need to use it.

Of course the whole system is supposed to be headless and fingerless under normal circumstances,
but without proper keyboard support this tool is also leaning towards pointless.

I had to make a cheat sheet to translate the needed characters to my keyboard layout.
Not user friendly.

WhiteWolf84, 2024-07-21:
I'm also need to switch layout, any news?

Loic691, 2024-08-25:
+1
Actually it's impossible for French users to debug with bad mapping keyboard...
I think it's a MUST HAVE small evolution to get ;-)

spaggio80, 2024-08-25:
I installed Home Assistant on a Synology NAS Virtual Machine and spent a lot of time looking for a solution, thinking it was a Virtual Machine Manager issue. Finally I got here and realized that it is the standard behavior of the host.
It is really absurd that you cannot change the keyboard layout, please fix it! Or, as someone suggested, write explicitly in the terminal session that the keyboard layout is "US" and cannot be changed!