home-assistant/operating-system #1783: KVM image growing

Sodium-Hydrogen, 2022-03-04:
### Describe the issue you are experiencing

My `.qcow2` file is growing in size overtime. I only noticed when it grew beyond the recommended 32gb image size.

[image]

As can be seen by used space I am only using about 8gb so the image should not be this large.

I used `login` and ran `systemctl start fstrim.service` and it reported clearing 22gb but this doesn't seem to matter. I made sure to `sync` the changes and shutdown the vm. I used `qemu-img convert -O qcow2 haos_ova-6.6.qcow2 shrunk.qcow2` which did cut off about 10gb but didn't shrink down to the actual 8gb I am seeing elsewhere.

Even `du -hs` on the root of the vm still only shows 12gb so I don't know where the extra space is coming from and why I cannot get rid of it.

Here is the before and after attempting to shrink the `.qcow2` file:
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

I do not know if it is reproducible on a standard proxmox setup since the hypervisor isn't a dedicated proxmox device.

I installed proxmox on openmediavault via their omv-extras plugin and used the kvm plugin to get it running. It didn't manually install all the proxmox tools and as such the guest fails to communicate via the qemu guest agent. 

It has also been running since November without any check-in on the image size so I do not know how long the issue has been ongoing. 

I do install updates fairly frequently so perhaps an update fragmented the filesystem? I'm not sure if `qcow` can be fragmented like that, but I have run into a similar issue on a different virtualization platform where a windows vm was bloated and I couldn't reclaim the space until I ran a defrag tool.

### Anything in the Supervisor logs that might be useful for us?

```txt
na
```

### Anything in the Host logs that might be useful for us?

```txt
na
```

### Additional information

Here is the fstrim logs from haos
```
○ fstrim.service - Discard unused blocks on filesystems from /etc/fstab
     Loaded: loaded (/usr/lib/systemd/system/fstrim.service; static)
     Active: inactive (dead) since Fri 2022-03-04 23:08:10 UTC; 7min ago
TriggeredBy: ● fstrim.timer
       Docs: man:fstrim(8)
    Process: 5280 ExecStart=/usr/sbin/fstrim --listed-in /etc/fstab:/proc/self/mountinfo --verbose --quiet-unsupported (code=exited, status=0/SUCCESS)
   Main PID: 5280 (code=exited, status=0/SUCCESS)

Mar 04 23:08:10 homeassistant systemd[1]: Starting Discard unused blocks on filesystems from /etc/fstab...
Mar 04 23:08:10 homeassistant fstrim[5280]: /var: 29.9 MiB (31305728 bytes) trimmed on /dev/zram1
Mar 04 23:08:10 homeassistant fstrim[5280]: /tmp: 14.9 MiB (15638528 bytes) trimmed on /dev/zram2
Mar 04 23:08:10 homeassistant fstrim[5280]: /mnt/data: 22.2 GiB (23825342464 bytes) trimmed on /dev/vda8
Mar 04 23:08:10 homeassistant fstrim[5280]: /mnt/overlay: 87.3 MiB (91497472 bytes) trimmed on /dev/vda7
Mar 04 23:08:10 homeassistant fstrim[5280]: /mnt/boot: 31.5 MiB (33040384 bytes) trimmed on /dev/vda1
Mar 04 23:08:10 homeassistant systemd[1]: fstrim.service: Deactivated successfully.
Mar 04 23:08:10 homeassistant systemd[1]: Finished Discard unused blocks on filesystems from /etc/fstab.
```

Sodium-Hydrogen, 2022-03-05:
I've looked into it further and it does look like the file system has fragmented somehow.

`e2fsck -nf /dev/vda8`

[image]

I haven't messed with anything outside of the web-ui until I noticed this issue so I don't believe it was something I've done to break the file system.

agners, 2022-03-17:
The qcow2 file could be sparse as well, can you check the file size using `du -sh <path-to-haos-qcow2>`.

What virtual disk type are you using? From what I remember the virtio-scsi type should support fstrim command best.

Sodium-Hydrogen, 2022-03-17:
The default is listed as virtio in openmediavault and that is what I used. I believe it maps to virtio block in proxmox.

And here are the qcow file sizes from the original post:

>Here is the before and after attempting to shrink the .qcow2 file:
>[image]

However I recently was able to get an actual virtualization server so I am moving all my vms from my do-everything server so I am running in a more supported format now using scsi.

Thanks for looking into this.

closed 2022-03-17 by Sodium-Hydrogen