home-assistant/operating-system #1786: Unable to upgrade to latest operation system

PrimaryInc, 2022-03-09:
### Describe the issue you are experiencing

I have issues upgrading from 8.0.dev20220223, to version 8.0.dev20220302.
HA reboots without any further information.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

 	Home Assistant OS 8.0.dev20220223

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Configuration show a OS update
2. Press upgrade
3. Host reboots
4. Upgrade not applied
5. Back to step 1.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
NA
```

### Anything in the Host logs that might be useful for us?

```txt
NA
```

### System Health information

## System Health

version | core-2022.4.0.dev20220309
-- | --
installation_type | Home Assistant OS
dev | true
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.98
arch | x86_64
timezone | Europe/Oslo

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4883
Installed Version | 1.23.0
Stage | running
Available Repositories | 1009
Downloaded Repositories | 20

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 8.0.dev20220223
-- | --
update_channel | dev
supervisor_version | supervisor-2022.03.dev0901
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 19.0 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | SSH & Web Terminal (10.0.2), File editor (5.3.3), Node-RED (11.0.4), Samba share (9.5.1), Studio Code Server (4.1.0), Grafana (7.4.1), InfluxDB (4.3.0), Z-Wave JS (0.1.54)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 11
views | 12
mode | storage

</details>

agners, 2022-03-17:
This is likely fixed by #1793. Can you try updating again?

PrimaryInc, 2022-03-17:
Hi @agners,

Perfect that worked :)
Thank you!!

closed 2022-03-17 by PrimaryInc