home-assistant/operating-system #1787: Unable to start HomeAssistant OS

pst11790, 2022-03-09:
### Describe the issue you are experiencing

HomeAssistant will not fully initialize after fresh build
PC is a Dell OptiPlex MFF 7050
SecureBoot disabled & Legacy Boot

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

Generic x86-64: Attempted using all 7 builds

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

PC is a Dell OptiPlex MFF 7050
SecureBoot disabled & Legacy Boot
My previous HomeAssistant build had issues that I was unable to resolve so I attempted to start from scratch (z-wave add-on issues)
1. Use balenaEtcher to write image for:
https://github.com/home-assistant/operating-system/releases/download/7.4/haos_generic-x86-64-7.4.img.xz to SSD
3. Boot PC
4. Attached screen boots to CLI
5. Web Server hangs at Preparing Home Assistant
6. Logs continuously cycle through same process

### Anything in the Supervisor logs that might be useful for us?

```txt
[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[23:15:18] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[23:15:18] INFO: Starting local supervisor watchdog...
22-03-09 23:15:20 INFO (MainThread) [__main__] Initializing Supervisor setup
22-03-09 23:15:21 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-03-09 18:15:21 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: generic-x86-64
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/amd64-hassio-supervisor with version 2022.03.2
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/amd64-hassio-supervisor:2021.10.6']
22-03-09 18:15:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
22-03-09 18:15:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-03-09 18:15:22 INFO (MainThread) [__main__] Setting up Supervisor
22-03-09 18:15:22 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
22-03-09 18:15:22 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.resolve1
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.info] Updating local host information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.services] Updating service information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.network] Updating local network information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
[... 182 lines cut ...]
22-03-09 18:16:54 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.RUNNING
22-03-09 18:16:55 WARNING (MainThread) [supervisor.utils.codenotary] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 35.168.17.115:443: i/o timeout"
22-03-09 18:17:25 ERROR (MainThread) [supervisor.utils.codenotary] Timeout while processing CodeNotary
22-03-09 18:17:25 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-03-09 18:17:25 INFO (MainThread) [supervisor.resolution.fixup] Starting system autofix at state CoreState.RUNNING
22-03-09 18:17:25 INFO (MainThread) [supervisor.resolution.fixup] System autofix complete
22-03-09 18:19:25 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-09 18:19:35 ERROR (MainThread) [supervisor.utils.codenotary] Timeout while processing CodeNotary
22-03-09 18:19:55 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:landingpage to ghcr.io/home-assistant/generic-x86-64-homeassistant:2022.3.3
22-03-09 18:19:55 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2022.3.3.
22-03-09 18:20:16 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-09 18:20:16 WARNING (MainThread) [supervisor.utils.codenotary] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 35.168.17.115:443: i/o timeout"
22-03-09 18:20:46 INFO (SyncWorker_0) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:landingpage to ghcr.io/home-assistant/generic-x86-64-homeassistant:2022.3.3
22-03-09 18:20:46 INFO (SyncWorker_0) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2022.3.3.
22-03-09 18:21:06 WARNING (MainThread) [supervisor.homeassistant.core] Error on Home Assistant installation. Retry in 30sec
22-03-09 18:21:16 ERROR (MainThread) [supervisor.utils.codenotary] Timeout while processing CodeNotary
22-03-09 18:21:36 INFO (SyncWorker_2) [supervisor.docker.interface] Updating image ghcr.io/home-assistant/generic-x86-64-homeassistant:landingpage to ghcr.io/home-assistant/generic-x86-64-homeassistant:2022.3.3
22-03-09 18:21:36 INFO (SyncWorker_2) [supervisor.docker.interface] Downloading docker image ghcr.io/home-assistant/generic-x86-64-homeassistant with tag 2022.3.3.
(this can take up to 20 minutes)
While waiting, some suggestions:
Read our founder's vision for the perfect home automation.
Install the official Home Assistant apps
Get it on the Apple App Store Get it on Google Play
Join our community on the forums or on the chat.
```

pst11790, 2022-03-10:
Re-attaching logs:

[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] udev.sh: executing...
[23:15:18] INFO: Using udev information from host
[cont-init.d] udev.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[23:15:18] INFO: Starting local supervisor watchdog...
22-03-09 23:15:20 INFO (MainThread) [__main__] Initializing Supervisor setup
22-03-09 23:15:21 INFO (MainThread) [supervisor.bootstrap] Initializing Supervisor Sentry
22-03-09 18:15:21 INFO (MainThread) [supervisor.bootstrap] Seting up coresys for machine: generic-x86-64
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.supervisor] Attaching to Supervisor homeassistant/amd64-hassio-supervisor with version 2022.03.2
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.supervisor] Connecting Supervisor to hassio-network
22-03-09 18:15:21 INFO (SyncWorker_0) [supervisor.docker.interface] Cleanup images: ['ghcr.io/home-assistant/amd64-hassio-supervisor:2021.10.6']
22-03-09 18:15:22 INFO (MainThread) [supervisor.resolution.evaluate] Starting system evaluation with state CoreState.INITIALIZE
22-03-09 18:15:22 INFO (MainThread) [supervisor.resolution.evaluate] System evaluation complete
22-03-09 18:15:22 INFO (MainThread) [__main__] Setting up Supervisor
22-03-09 18:15:22 INFO (MainThread) [supervisor.api] Starting API on 172.30.32.2
22-03-09 18:15:22 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
22-03-09 18:15:22 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
22-03-09 18:15:23 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.resolve1
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.info] Updating local host information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.services] Updating service information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.network] Updating local network information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-03-09 18:15:23 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
22-03-09 18:15:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-cli with version 2021.12.0
22-03-09 18:15:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-dns with version 2021.06.0
22-03-09 18:15:23 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
22-03-09 18:15:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-audio with version 2021.07.0
22-03-09 18:15:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-observer with version 2021.10.0
22-03-09 18:15:23 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/amd64-hassio-multicast with version 2022.02.0
22-03-09 18:15:23 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-03-09 18:15:53 ERROR (MainThread) [supervisor.utils.codenotary] Timeout while processing CodeNotary
22-03-09 18:15:53 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-03-09 18:16:13 WARNING (MainThread) [supervisor.utils.codenotary] rpc error: code = Unavailable desc = connection error: desc = "transport: Error while dialing dial tcp 35.168.17.115:443: i/o timeout"
22-03-09 18:16:13 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 0 Home Assistant secrets
22-03-09 18:16:13 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/generic-x86-64-homeassistant with version landingpage
22-03-09 18:16:13 INFO (MainThread) [supervisor.homeassistant.core] Starting HomeAssistant landingpage
22-03-09 18:16:14 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 6.6 / BootSlot A
22-03-09 18:16:14 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
22-03-09 18:16:14 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
22-03-09 18:16:14 INFO (MainThread) [supervisor.store] Loading add-ons from store: 62 all - 62 new - 0 remove
22-03-09 18:16:14 INFO (MainThread) [supervisor.addons] Found 0 installed add-ons
22-03-09 18:16:14 INFO (MainThread) [supervisor.backups.manager] Found 0 backup files
[... cut ...]
22-03-09 19:20:15 INFO (SyncWorker_3) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/generic-x86-64-homeassistant
22-03-09 19:22:18 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant
22-03-09 19:24:18 ERROR (MainThread) [supervisor.misc.tasks] Watchdog found a problem with Home Assistant API!
22-03-09 19:24:18 INFO (SyncWorker_2) [supervisor.docker.interface] Restarting ghcr.io/home-assistant/generic-x86-64-homeassistant
22-03-09 19:26:22 WARNING (MainThread) [supervisor.misc.tasks] Watchdog miss API response from Home Assistant

frenck, 2022-03-10:
duplicate of https://github.com/home-assistant/supervisor/issues/3495

closed 2022-03-10 by frenck