home-assistant/operating-system #1792: Home Assistant OS 7.5 - unable to connect after update

Twanne, 2022-03-16:
### Describe the issue you are experiencing

After the update from 7.4 to 7.5, I'm getting an error:
Unable to connect to Home Assistant.
Retry

I've tried connecting via DNS url and via local ip address directly.
No difference.

The Samba and SSH connections are down too, so i'm thinking the OS has not booted correctly.

### What operating system image do you use?

generic-x86-64 (Generic UEFI capable x86-64 systems)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. upgrade HASSOS from 7.4 to 7.5
2. fail to connect...
3.
...

### Anything in the Supervisor logs that might be useful for us?

```txt
unable to access logs
```

### Anything in the Host logs that might be useful for us?

```txt
unable to access logs
```

Twanne, 2022-03-16:
Issue resolved: HASSOS VM in Proxmox was running into a CPU limit.
Stopped the vm, increased number of cores and rebooted.
It now works.

agners, 2022-03-16:
Thanks for the update!