home-assistant/operating-system #1795: Conbee II ZHA broke after HA OS update 7.5

satyambaba, 2022-03-16:
### Describe the issue you are experiencing

ZHA using Conbee II has stopped working after the home assistant OS update 7.5. Using HA Blue (core-2022.3.5). I unplugged and re-plugged the stick (with and without the USB extension chord), rebooted the HA Blue machine several times, reloaded the ZHA integration, but to no avail.

Error is -
Retrying setup: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2440792-if00: [Errno 2] No such file or directory: ‘/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2440792-if00’

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Just after the update to HA OS 7.5

### Anything in the Supervisor logs that might be useful for us?

```txt
.
```

### Anything in the Host logs that might be useful for us?

```txt
.
```

satyambaba, 2022-03-17:
It started working when I rolled back to HA OS 7.4

agners, 2022-03-17:
Does un-/replugging the device help? Can you check if you see the device under Supervisor System -> Host -> 3 dot menu -> Hardware? Also, can you share the host logs (dmesg)?

In a quick try, a Conbee II seems to be correctly recognized here

```
# ha os info
board: odroid-n2
boot: B
data_disk: /dev/mmcblk1p4
update_available: true
version: "7.5"
version_latest: 7.5
# dmesg | tail
[249417.811651] audit: type=1334 audit(1647496672.914:431): prog-id=147 op=UNLOAD
[249418.135745] audit: type=1334 audit(1647496673.238:432): prog-id=150 op=UNLOAD
[249418.135756] audit: type=1334 audit(1647496673.238:433): prog-id=149 op=UNLOAD
[256773.164879] usb 1-1.4: new full-speed USB device number 3 using xhci-hcd
[256773.334195] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
[256773.334595] usbcore: registered new interface driver cdc_acm
[256773.334597] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[256780.812873] usb 1-1.4: USB disconnect, device number 3
[256781.092554] usb 1-1.4: new full-speed USB device number 4 using xhci-hcd
[256781.260898] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device
```

satyambaba, 2022-03-17:
> Does un-/replugging the device help? Can you check if you see the device under Supervisor System -> Host -> 3 dot menu -> Hardware? Also, can you share the host logs (dmesg)?
>
> In a quick try, a [...]

No, un-/replugging the device doesn't help. I can't see the Conbee II stick under the hardware menu. Below is the host logs -

2022-03-17 05:24:59 DEBUG (SyncWorker_3) [miio.miioprotocol] 10.0.0.70:54321 >>: {'id': 41, 'method': 'set_properties', 'params': [{'did': 'aqi_realtime_update_duration', 'siid': 13, 'piid': 9, 'value': 5}]}
2022-03-17 05:24:59 ERROR (MainThread) [zigpy.application] Couldn't start application
2022-03-17 05:24:59 ERROR (MainThread) [homeassistant.components.zha.core.gateway] Couldn't start deCONZ = dresden elektronik deCONZ protocol: ConBee I/II, RaspBee I/II coordinator
Traceback (most recent call last):
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 322, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
FileNotFoundError: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436216-if00'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/zha/core/gateway.py", line 152, in async_initialize
    self.application_controller = await app_controller_cls.new(
  File "/usr/local/lib/python3.9/site-packages/zigpy/application.py", line 69, in new
    await app.startup(auto_form)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/zigbee/application.py", line 66, in startup
    await self._api.connect()
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/api.py", line 253, in connect
    self._uart = await zigpy_deconz.uart.connect(self._config, self)
  File "/usr/local/lib/python3.9/site-packages/zigpy_deconz/uart.py", line 142, in connect
    _, protocol = await serial_asyncio.create_serial_connection(
  File "/usr/local/lib/python3.9/site-packages/serial_asyncio/__init__.py", line 448, in create_serial_connection
    serial_instance = serial.serial_for_url(*args, **kwargs)
  File "/usr/local/lib/python3.9/site-packages/serial/__init__.py", line 90, in serial_for_url
    instance.open()
  File "/usr/local/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436216-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436216-if00'
2022-03-17 05:24:59 WARNING (MainThread) [homeassistant.config_entries] Config entry 'ConBee II, s/n: DE2436216 - dresden elektronik ingenieurtechnik GmbH' for zha integration not ready yet: [Errno 2] could not open port /dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436216-if00: [Errno 2] No such file or directory: '/dev/serial/by-id/usb-dresden_elektronik_ingenieurtechnik_GmbH_ConBee_II_DE2436216-if00'; Retrying in background

agners, 2022-03-17:
Do you use an extension cable? The front plate of the Blue is known to be a bit too thick, and some USB devices don't make proper connection. See also https://github.com/home-assistant/operating-system/issues/1602#issuecomment-958994083.

satyambaba, 2022-03-17:
> Do you use an extension cable? The front plate of the Blue is known to be a bit too thick, and some USB devices don't make proper connection. See also [#1602 (comment)](https://github.com/home-assistant/operating-system/issues/1602#issuecomment-958994083).

Yes, I use extension cable. However everything was working fine, until the update. Also, when I rolled back to 7.4 it started working again.

agners, 2022-03-17:
> Yes, I use extension cable. However everything was working fine, until the update. Also, when I rolled back to 7.4 it started working again.

Yeah, that is weird. However, I can't reproduce this on my ODROID-N2, so it might be just a bit unreliable, or really a weird interaction with your device <-> OS 7.5...

Can you share logs from Configuration -> Add-ons, Backup & Supervisor -> System -> Log Provider "Host"? Ideally after upgrade to 7.5, and also after unplugging/re-plugging the device.

mbrennan, 2022-05-19:
The same issue happened to me after upgrading to Home Assistant Operating System to 8.0.  My Conbee II doesn't seem to be recognized.  I can't see any other devices in the hardware section (recently moved around in Home Assistant Core), under Settings -> System -> Hardware.  The only thing that's listed there is "Raspberry Pi 4 rpi4-64".  

I've tried rebooting the host and unplugging and replugging in my Conbee II stick.  I'm not using an extension cable.

How can I roll back the Home Assistant OS to the previous version?

agners, 2022-05-19:
@mbrennan use the following command in the Terminal add-on:

```
ha os update --version 7.6
```

That said, I doubt that this is related to the OS version. Can you post the Host logs (dmesg)?

mbrennan, 2022-05-19:
This is copy/pasted from System -> Logs with the 'Host' drop down:

`[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 5.15.32-v8 (builder@6a275e347639) (aarch64-buildroot-linux-gnu-gcc.br_real (Buildroot -gee5615946) 9.4.0, GNU ld (GNU Binutils) 2.36.1) #1 SMP PREEMPT Wed May 11 12:09:58 UTC 2022
[    0.000000] random: fast init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at 0x000000002ac00000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   0: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x00000001ffffffff]
[    0.000000] percpu: Embedded 28 pages/cpu s77272 r8192 d29224 u114688
[    0.000000] pcpu-alloc: s77272 r8192 d29224 u114688 alloc=28*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Detected PIPT I-cache on CPU0
[    0.000000] CPU features: detected: Spectre-v2
[    0.000000] CPU features: detected: Spectre-v4
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: detected: ARM errata 1165522, 1319367, or 1530923
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 2028848
[    0.000000] Kernel command line: zram.enabled=1 zram.num_devices=3 apparmor=1 security=apparmor rootwait systemd.machine_id=3cc6162a61744f4a836fdb5e2e401e41 cgroup_enable=memory fsck.repair=yes coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 bcm2708_fb.fbwidth=0 bcm2708_fb.fbheight=0 bcm2708_fb.fbswap=1 smsc95xx.macaddr=E4:5F:01:5E:2A:53 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 root=PARTUUID=8d3d53e3-6d49-4c38-8349-aff6859e82fd rootfstype=squashfs ro rauc.slot=A
[    0.000000] cgroup: Enabling memory control group subsystem
[    0.000000] Dentry cache hash table entries: 1048576 (order: 11, 8388608 bytes, linear)
[    0.000000] Inode-cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: mapped [mem 0x0000000037400000-0x000000003b400000] (64MB)
[    0.000000] Memory: 7929060K/8245248K available (12608K kernel code, 2008K rwdata, 4388K rodata, 3968K init, 985K bss, 250652K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 40879 entries in 160 pages
[    0.000000] ftrace: allocated 160 pages with 2 groups
[    0.000000] trace event string verifier disabled
[    0.000000] rcu: Preemptible hierarchical RCU implementation.
[    0.000000] rcu: 	RCU event tracing is enabled.
[    0.000000] rcu: 	RCU restricting CPUs from NR_CPUS=256 to nr_cpu_ids=4.
[    0.000000] 	Trampoline variant of Tasks RCU enabled.
[    0.000000] 	Rude variant of Tasks RCU enabled.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 25 jiffies.
[    0.000000] rcu: Adjusting geometry for rcu_fanout_leaf=16, nr_cpu_ids=4
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] random: get_random_bytes called from start_kernel+0x4fc/0x6e8 with crng_init=1
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0xc743ce346, max_idle_ns: 440795203123 ns
[    0.000001] sched_clock: 56 bits at 54MHz, resolution 18ns, wraps every 4398046511102ns
[    0.000257] Console: colour dummy device 80x25
[    0.000841] printk: console [tty1] enabled
[    0.000928] Calibrating delay loop (skipped), value calculated using timer frequency.. 108.00 BogoMIPS (lpj=216000)
[    0.000968] pid_max: default: 32768 minimum: 301
[    0.001104] LSM: Security Framework initializing
[    0.001247] AppArmor: AppArmor initialized
[    0.001480] Mount-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.001609] Mountpoint-cache hash table entries: 16384 (order: 5, 131072 bytes, linear)
[    0.004936] rcu: Hierarchical SRCU implementation.
[    0.006066] EFI services will not be available.
[    0.006622] smp: Bringing up secondary CPUs ...
[    0.007665] Detected PIPT I-cache on CPU1
[    0.007745] CPU1: Booted secondary processor 0x0000000001 [0x410fd083]
[    0.008945] Detected PIPT I-cache on CPU2
[    0.008995] CPU2: Booted secondary processor 0x0000000002 [0x410fd083]
[    0.010133] Detected PIPT I-cache on CPU3
[... 545 lines cut ...]
[  159.163357] audit: type=1300 audit(1652961716.678:187): arch=c00000b7 syscall=206 success=yes exit=40 a0=f a1=400155be00 a2=28 a3=0 items=0 ppid=1 pid=448 auid=4294967295 uid=0 gid=0 euid=0 suid=0 fsuid=0 egid=0 sgid=0 fsgid=0 tty=(no [... cut ...]
[  166.782027] audit: type=1334 audit(1652961724.297:199): prog-id=16 op=LOAD
[  167.584955] audit: type=1334 audit(1652961725.101:200): prog-id=17 op=LOAD
[  167.585268] audit: type=1334 audit(1652961725.101:201): prog-id=18 op=LOAD
[  197.644588] audit: type=1334 audit(1652961755.161:202): prog-id=0 op=UNLOAD
[  197.644627] audit: type=1334 audit(1652961755.161:203): prog-id=0 op=UNLOAD
[  198.424763] audit: type=1334 audit(1652961755.941:204): prog-id=0 op=UNLOAD
[  198.424834] audit: type=1334 audit(1652961755.941:205): prog-id=0 op=UNLOAD`

agners, 2022-05-19:
Hm, a Conbee II should enumerate with something like this:

```
[196811.525811] usb 5-4.3.4: new full-speed USB device number 117 using xhci_hcd
[196811.651043] usb 5-4.3.4: New USB device found, idVendor=1cf1, idProduct=0030, bcdDevice= 1.00
[196811.651051] usb 5-4.3.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[196811.651053] usb 5-4.3.4: Product: ConBee II
[196811.651054] usb 5-4.3.4: Manufacturer: dresden elektronik ingenieurtechnik GmbH
[196811.651056] usb 5-4.3.4: SerialNumber: DE1963366
[196811.690120] cdc_acm 5-4.3.4:1.0: ttyACM2: USB ACM device
```
I don't see such device in your kernel log, that means the Linux kernel itself does not enumerate/see that device. Can you try a different USB port? It seems as if that port or the device is dead.

mbrennan, 2022-05-19:
Sorry, I realized it's not a Conbee II stick, it's a Raspbee II integrated device onboard.  Everything was working fine prior to the upgrade.  After the upgrade it stopped working.  Downgrading to 7.6 doesn't seem to help either.

agners, 2022-05-19:
@mbrennan ok, this odes not belong here then. I created a new issue #1931, let's continue discussing there.

roessi, 2022-05-29:
I made some Research last night.
It looks like kernel version 5.15.41 is causing the problem
On my Odroid n2 the 5V of the USB port are switched off.
No idea if this can be fixed with a setting in the linux config, or if it's a driver problem.
I don't know enough about linux for that, but as a result, no USB device is recognized on the Odroid.

thomasklein, 2022-06-01:
+1 with the conbee II issue also running HA on odroid-n2 (Hardkernel ODROID-N2/N2+)
```
Home Assistant Core 2022.6.0
Home Assistant Supervisor 2022.05.3
Home Assistant OS 8.1
```

The supervisor logs show me:
```
22-06-01 23:01:28 WARNING (MainThread) [supervisor.addons.options] Option 'vnc_password' does not exist in the schema for deCONZ (core_deconz)
```

The host logs don't mention `Product: ConBee II`, however I found:
```
 1.423326] USB_PWR_EN: supplied by 5V
[    1.427009] VDDAO_1V8: supplied by VDDAO_3V3
[    5.046507] phy-g12a-usb3-pcie: probe of ff646000.phy failed with error -5
```
I wonder if that is related.

thomasklein, 2022-06-05:
@roessi were you able to advance on it?

roessi, 2022-06-06:
No
I changed my hardware to an intel nuc.

Mit freundlichen Grüßen
Rudi Rösler

closed 2022-09-11 by github-actions[bot]