home-assistant/operating-system #1796: HassOS 7.5 rflink USB device gone!

herwarth, 2022-03-17:
### Describe the issue you are experiencing

After upgrading to HassOS 7.5 my rflink usb device is not available anymore so integration in core is not working. Reboot and power down, replugging do not work!

### What operating system image do you use?

odroid-n2 (Hardkernel ODROID-N2/N2+)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1.
2.
3.
...

Update to 7.5. Reboot. Rflink dev file gone. 

### Anything in the Supervisor logs that might be useful for us?

```txt
None
```

### Anything in the Host logs that might be useful for us?

```txt
None
```

agners, 2022-03-17:
Did maybe the device path change? Can you check if you see the device under Supervisor System -> Host -> 3 dot menu -> Hardware?

agners, 2022-03-17:
> Is Linux with the latest kernels more picky with USB recognition?

Not that I am aware off. However, there are other reports related to ODROID-N2. Also, I had a ODROID-N2 which had troubles with USB, so maybe its device specific...