home-assistant/operating-system #1800: USB problem after last update

malosaa, 2022-03-17:
### Describe the issue you are experiencing

Well i did experience an issue with my rfxcom device, but now also with other usb devices.

When i restart homeassistant from the gui the usb disconnects and i get this error:

i have 3 usb devices connected: 
1. rfxtrx
2. plugwise usb
3. zigbee2mqtt (never disconnects)

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. reboot homeassistant from the gui
2. 2 usb devices get disconnected.

### Anything in the Supervisor logs that might be useful for us?

```txt
yes see above message
```

### Anything in the Host logs that might be useful for us?

```txt
No
```

agners, 2022-03-17:
Can you try downgrade again to see if the issue indeed disappears?

```
ha os update --version 7.4
```

malosaa, 2022-03-17:
Hi @agners 

Well i did restart 2x from gui with 1 hour between it, and it did not happen.
I dont know why it did happen before.

On 7.2 - 7.4 it did happen aswell but many times on the rfxcom device, same usb error.
Mostly it just stops working when there is no restart.

malosaa, 2022-03-18:
I always had this issue, but i thought it was the rfxcom, but Bert the creator of the rfxcom said, please run it into the rfxmgr for a couple of days and see if there is a disconnect.
But there is no disconnect for 3 days.

A long time ago i had the issue with the zigbee2mqtt usb, i always got that error like above and it stopped working, than after some new versions it got fixed and it never happened anymore.

But yesterday somehow after the first time i update to 7.5 2 usb failed with that error, but it did not happen again.
When there is another core update i let homeassistant restart again and see if it happens again.

I really don't know if this is a OS problem or just the RFXCOM/PLUGWISE problem.

regards

malosaa, 2022-03-19:
[image]

@agners
this happens after a full restart, this is my full restart in 2 months.
regards

malosaa, 2022-04-06:
Well, magically it still working, probably some new versions HASS dit fix it.
I think you maybe can close this issue, as i think its solved :)

closed 2022-04-07 by agners