home-assistant/operating-system #1802: Apply for a vendor url (example 0.home-assistant.pool.ntp.org) for using directly the NTP pools from ntp.org

redgryphon, 2022-03-18:
### Describe the issue you are experiencing

As discussed with @agners [here](https://github.com/home-assistant/operating-system/pull/1798/files/0a927ac8452858e0e3365d6cc54c90700241012c#r829226385) I'm creating a separated issue about asking the ntp.org project to create a [vendor url](https://www.ntppool.org/en/vendors.html) for NTP Doing so would be nice and would place Home Assistant in the right place.

Plus, at the moment the Home Assistant project is not following the guidelines requested by the ntp.org website since they ask vendors (if the Debian project falls into the definition of vendor then Home Assistant has to too) not to reference directly the general pools but to apply for a vendor pool.

Ideally the result would be changing [this line](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/rootfs-overlay/etc/systemd/timesyncd.conf#L3) into something like:
` FallbackNTP=0.home-assistant.pool.ntp.org 1.home-assistant..pool.ntp.org 2.home-assistant..pool.ntp.org 3.home-assistant..pool.ntp.org`

Pros:

- At the moment before merging #1798 Home Assistant is not abiding the ntp.org [requests](https://www.ntppool.org/en/vendors.html#vendor-zone) of not using the default pool.
- It makes the fallback not dependent on a single organization (after merging #1798 it would be Cloudflare), even if that organization provides a distributed pool
- It's more ethical since Home Assistant as an Open Source project would rely on a no-profit organization instead of a for profit one
- It just looks nicer

Cons:

- It takes some minutes of work at the beginning requiring writing [a single form](https://manage.ntppool.org/manage/vendor) on the ntp.org website.

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.4

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Do nothing, it is hardcoded [here](https://github.com/home-assistant/operating-system/blob/dev/buildroot-external/rootfs-overlay/etc/systemd/timesyncd.conf#L3).

### Anything in the Supervisor logs that might be useful for us?

```txt
No.
```

### Anything in the Host logs that might be useful for us?

```txt
No.
```

agners, 2022-03-21:
The first point doesn't seem to be a valid "Pro" argument, but it makes the list longer sure :wink: 

One advantage of Cloudflare is that they support NTS and also have a roughtime service. That said, we are not using any of the two currently, so :man_shrugging: 

Currently the form isn't working (it returns a 500 error). I'll retry in a couple of days.

It might be not free for us (depending on the definition of "commercial open source").

redgryphon, 2022-04-05:
Any news on the matter? I'd gladly do it myself but I suppose the request coming from an account like name@home-assistant.io is better.

agners, 2022-04-07:
I sent an email on March 21, didn't received an answer so far. I'll follow up.

redgryphon, 2022-04-07:
Thanks for your work. I appreciate it.

redgryphon, 2023-04-07:
I know that there are more important things to do but trying costs me nothing. So, any news?

Kagee, 2024-04-10:
Time to try again? This vendor prefix could then also possibly be re-used in esphome examples https://esphome.io/components/time/sntp.html