home-assistant/operating-system #1803: HASS unresponsive after a few hours, kernel panics

HangGlidersRule, 2022-03-19:
### Describe the issue you are experiencing

Brand new rpi, brand new SD card, latest OS image.  No peripherals connected other than the display.

After a few hours, sometimes more quickly, HASS locks up entirely.  HDMI display goes blank with nothing displayed, keyboard input does nothing, webUI is inaccessible.  The rpi4 is still pingable, and my port 22222 SSH session remains connected.  Only solution is a complete reboot of the rpi.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

None.  Leaving it powered on does it.

### Anything in the Supervisor logs that might be useful for us?

```txt
inaccessible, see journalctl

https://pastebin.com/C4CsQ7Yd
```

### Anything in the Host logs that might be useful for us?

```txt
inaccessible, see dmesg

https://pastebin.com/BGtwCxFk
```

thedead, 2022-03-19:
I'm getting the same thing, I actually restarted when this was happening and my qcow2 file got corrupted.

agners, 2022-03-19:
> I'm getting the same thing, I actually restarted when this was happening and my qcow2 file got corrupted.

That is not the same thing, this kernel crash is very Raspberry Pi specific. Please open a separate issue with details and logs.

agners, 2022-03-19:
@HangGlidersRule that sounds pretty serious. Did you do any modifications to `config.txt`? Can you run Raspberry Pi OS for a few hours and see if something similar is reported there as well? We use the same kernel as the latest Raspberry Pi OS release, so it should be the same behavior.

agners, 2022-03-19:
raspberrypi/linux#4047 seems to report a similar problem.

What add-ons are you using? Anything multimedia related?

HangGlidersRule, 2022-03-19:
@agners This is a vanilla HASS install, no addons.  Not doing any multimedia. I have about a dozen or so integrations now, no automations/scenes, but this was happening before any integrations were added. I’ll run it on vanilla rpi OS for a bit, and I’ll throw this SD card in my other brand new rpi4 to see if it’s a hardware issue.

HangGlidersRule, 2022-03-20:
~24 hours of running the same SD card in my other rpi and it’s been rock solid stable. I think it’s safe to assume that it is a hardware issue with the first unit. Appreciate the quick attention!  Possible that it has to do with some power saving mode as it only happens when it’s not doing anything. Not sure. I’ll get RPOS going on it and open a bug over there.  Thanks!!

agners, 2022-03-21:
Thanks for the update! Ok, then I am closing this for the moment. If it turns out something needs to be done in HAOS feel free to reopen.

Btw, I just updated HAOS to Raspberry Pi's latest firmware/kernel (20220308 tags). You might want to give those images a try, just in case they happen to fix it already: https://os-builds.home-assistant.io/8.0.dev20220321/

closed 2022-03-21 by agners