home-assistant/operating-system #1805: Failed to restart during OS 7.5 update

vandad51, 2022-03-20:
### Describe the issue you are experiencing

After updating from OS 7.4 to 7.5, HA failed to resume running state. Had to do a hard reset and then it started normally after that. 
[image]

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Initiated update from 7.4

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
See picture
```

### System Health information

## System Health

version | core-2022.3.5
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.103
arch | x86_64
timezone | America/Winnipeg

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4946
Installed Version | 1.23.0
Stage | running
Available Repositories | 1075
Downloaded Repositories | 9

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.5
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 30.8 GB
disk_used | 10.4 GB
healthy | true
supported | true
board | ova
supervisor_api | ok
version_api | ok
installed_addons | Node-RED (11.1.0), Samba share (9.5.1), Studio Code Server (4.1.0), Mosquitto broker (6.0.1), Terminal & SSH (9.3.0), Duck DNS (1.14.0), File editor (5.3.3), Log Viewer (0.12.2), Zigbee2mqtt (1.24.0-1), MariaDB (2.4.0), Nginx Proxy Manager (0.11.0), NGINX Home Assistant SSL proxy (3.1.1), Alsa & PulseAudio Fix (3.4), VLC (0.1.3), AdGuard Home (4.5.0), Home Assistant Google Drive Backup (0.105.2)

</details>

<details><summary>Lovelace</summary>

dashboards | 2
-- | --
resources | 4
views | 4
mode | storage

</details>

### Additional information

This is more of a FYI for the developers

agners, 2022-03-21:
What virtualization software and version are you using?

vandad51, 2022-03-21:
VirtualBox 6.1.30

agners, 2022-03-28:
This seems a similar problem to #1705. Can you try an earlier version of VirtualBox?

vandad51, 2022-03-30:
@agners Since last comment, I've had no issues with restarts and updates. I did not roll back vbox either

agners, 2022-03-30:
Uh, that is interesting... Maybe some other system changes (Windows update) or cosmic radiation changed it?

I am closing for now, feel free to reopen in case you experience the problem again.

closed 2022-03-30 by agners