home-assistant/operating-system #1808: unable to mount hassos-data partition on a freshly imaged SD card. Fails to Boot

MrSaatci, 2022-03-21:
### Describe the issue you are experiencing

I’ve been using HA for the past couple of months and have moved from a QEMU on Linux to a dedicated raspberry pi 4 and couldn’t be happier EXCEEEPT that I just got a couple of brand new larger SD cards and I wanted to transplant everything to one of them and in the process I convinced a friend of mine of getting his own pi4 with HA, the problem comes when trying to flash HA to the SD Cards, I see an odd result:

No matter what software I use it will always fail at verification, tools I’ve used:
Etcher (Linux & Mac)
Raspberry Imager (Linux & Mac)
Rufus (Win10)
Linux’ dd
fdisk
gparted
UNetbootin

I’ve used a total of 12 SD Cards from different vendors and 8 different adapters, I’ve also tried using the integrated ports.

It’s always the same result:
Fails
Shows up only 2 partitions as mountable (Kernel and System)
If I then go to Terminal and run fdisk I’ll get the warning that the partion table is corrupted and if I create a new table (dos) I then magically see the boot partition mountable, everything else is gone.

With all cards I’ve tried reformatting, 00’ed them with dd, fdisk, and everything. I do not understand what’s going on.

Could the img file be corrupted from source?

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

5.4

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

1. Flash HA img with any software on Mac or Linux, even with dd.
2. Try to boot it on the Pi4
...

### Anything in the Supervisor logs that might be useful for us?

```txt
never loads
```

### Anything in the Host logs that might be useful for us?

```txt
never loads
```

### System Health information

never loads

agners, 2022-03-21:
> What version of Home Assistant Operating System is installed?
> 5.4

That seems a rather old version, can you try the latest version available at:

https://github.com/home-assistant/operating-system/releases

> If I then go to Terminal and run fdisk I’ll get the warning that the partion table is corrupted and if I create a new table (dos) I then magically see the boot partition mountable, everything else is gone.

Obviously, the partition table should not be corrupted. We do write a hybrid partition table (it means we use MBR and GPT at the same time, this is required since on some Raspberry Pi's the ROM only knows about MBR). What fdisk version are you using?

For me the partition table looks like this:

```
# fdisk /dev/mmcblk0

Welcome to fdisk (util-linux 2.36.2).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').

Command (m for help): p

Disk /dev/mmcblk0: 58.88 GiB, 63218647040 bytes, 123473920 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 229E0A4B-E98F-43B3-8C6C-46E1E71DF3D7

Device           Start       End   Sectors  Size Type
/dev/mmcblk0p1    2048     67583     65536   32M Microsoft reserved
/dev/mmcblk0p2   67584    116735     49152   24M Linux filesystem
/dev/mmcblk0p3  116736    641023    524288  256M Linux filesystem
/dev/mmcblk0p4  641024    690175     49152   24M Linux filesystem
/dev/mmcblk0p5  690176   1214463    524288  256M Linux filesystem
/dev/mmcblk0p6 1214464   1230847     16384    8M Linux filesystem
/dev/mmcblk0p7 1230848   1427455    196608   96M Linux filesystem
/dev/mmcblk0p8 1427456 123473886 122046431 58.2G Linux filesystem

```

Note that the images are xz compressed. Etcher should decompress the image automatically.

Also can you check the sha256sum of the downloaded file? `haos_rpi4-64-7.5.img.xz` should be `fa0ff6f0c84dee0bfeff464f345605aa4d3ee8e2f066b2a85352a8ea3111345f`.

MrSaatci, 2022-03-21:
Thanks for replying so soon @agners 

I was in fact, using 7.5 img downloaded directly from the Home Assistant site, I just tried redownloaded the img file with same results, I've also tried using etcher using the URL as input.

on a side note:
Imager finishes writing the image but fails at verification:
Verifying write failed. Contents of SD card is different from what was written.
Etcher also fails but gives a little more light into what's going on: Checksum does not match for range [0, 2147483647]: "cff91b5dd1068876" != "6f7060e686220a25"

fsck returns:
``` root@MrSaatci-Kubuntu:/home/mrsaatci# fdisk /dev/sdc

Welcome to fdisk (util-linux 2.36.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

The backup GPT table is corrupt, but the primary appears OK, so that will be used.
The backup GPT table is not on the end of the device. This problem will be corrected by write.
A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').

Command (m for help): p

Disk /dev/sdc: 500.06 GiB, 536939069440 bytes, 1048709120 sectors
Disk model: Storage Device  
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 0C01CB21-A462-4605-B22C-187C3C2B77A2

Device       Start     End Sectors  Size Type
/dev/sdc1     2048   67583   65536   32M Microsoft reserved
/dev/sdc2    67584  116735   49152   24M Linux filesystem
/dev/sdc3   116736  641023  524288  256M Linux filesystem
/dev/sdc4   641024  690175   49152   24M Linux filesystem
/dev/sdc5   690176 1214463  524288  256M Linux filesystem
/dev/sdc6  1214464 1230847   16384    8M Linux filesystem
/dev/sdc7  1230848 1427455  196608   96M Linux filesystem
/dev/sdc8  1427456 4048895 2621440  1.3G Linux filesystem

Command (m for help): 

root@MrSaatci-Kubuntu:/home/mrsaatci# sha256sum /home/mrsaatci/Documents/haos_rpi4-64-7.5.img.xz
fa0ff6f0c84dee0bfeff464f345605aa4d3ee8e2f066b2a85352a8ea3111345f  /home/mrsaatci/Documents/haos_rpi4-64-7.5.img.xz 
```
´´´

I've tried decompressing it myself so I can use dd to flash my SD cards, I've also tried using Imager and picking HA from their menu and Balena Etcher with the link and both img and xz downloaded files. Nothing seems to work. I've been playing around with some cards and mkef2s

gdisk returns:
```
root@MrSaatci-Kubuntu:/home/mrsaatci# gdisk /dev/sdc
GPT fdisk (gdisk) version 1.0.8

Caution: invalid backup GPT header, but valid main header; regenerating
backup header from main header.

Warning! Main and backup partition tables differ! Use the 'c' and 'e' options
on the recovery & transformation menu to examine the two tables.

Warning! One or more CRCs don't match. You should repair the disk!
Main header: OK
Backup header: ERROR
Main partition table: OK
Backup partition table: ERROR

Partition table scan:
  MBR: hybrid
  BSD: not present
  APM: not present
  GPT: damaged

Found valid MBR and corrupt GPT. Which do you want to use? (Using the
GPT MAY permit recovery of GPT data.)
 1 - MBR
 2 - GPT
 3 - Create blank GPT

Your answer: 
```

agners, 2022-03-21:
> Verifying write failed. Contents of SD card is different from what was written.
> Etcher also fails but gives a little more light into what's going on: Checksum does not match for range [0, 2147483647]: "cff91b5dd1068876" != "6f7060e686220a25"

That should definitely not be the case. Can you try writing some other image? Is maybe the SD card reader/writer broken?

> The backup GPT table is corrupt, but the primary appears OK, so that will be used.
The backup GPT table is not on the end of the device. This problem will be corrected by write.
A hybrid GPT was detected. You have to sync the hybrid MBR manually (expert command 'M').

Yes, that is expected! The image does not know how big the SD card is when flashing. In fact, Home Assistant will fix that on first boot. But of course, for that it needs to boot first :smile: 

What happens when you plug-in the SD card to your Pi? Any output on HDMI? Any LEDs blinking? Can you double check you use the right image (e.g. `rpi4-64` is for Raspberry Pi 4 only).

MrSaatci, 2022-03-23:
Hi @agners 
Once it fails I cannot see the hassos-data and hassos-overlay partitions ready to mount (as I've seen on previously flashed, including my own) I only hassos-system0 and hassos-kernel0. I don't know if this helps in any way...
Booting the pi will start just as normal as my current card, but then gets stalled at mounting hassos-data, it shows some error on the kernel time syncing, but reading through the forums I've seen that that is related to the mounting process of the data and overlay partitions. After a couple of minutes shows failed dependencies and gets stuck on: Waiting for the CLI forever.

agners, 2022-03-28:
Ah ok, so the system is booting then... That's different from what you described earlier.

Do you have the system connected to Ethernet? The OS does not ship with Home Assistant Core pre-installed, instead it will get installed on first boot. Make sure to plug-in HA to a network via Ethernet with a working Internet connectivity.

MrSaatci, 2022-05-08:
It turned out to be a defective SSD card. Once I changed the card, it worked.

closed 2022-05-08 by MrSaatci