home-assistant/operating-system #1812: Add nftables to kernel modules

theseal, 2022-03-23:
### Describe the issue you are experiencing

Hi,

Using nftables in my Wireguard UI addon. The addon can't be started using the ova image due to lack of nftables support in kernel.

Is it possible to add nftables to the OS?
https://wiki.nftables.org/wiki-nftables/index.php/Building_and_installing_nftables_from_sources

### What operating system image do you use?

ova (for Virtual Machines)

### What version of Home Assistant Operating System is installed?

7.5

### Did you upgrade the Operating System.

No

### Steps to reproduce the issue

No steps, just the nftables module that is not present in the OS.

### Anything in the Supervisor logs that might be useful for us?

```txt
No
```

### Anything in the Host logs that might be useful for us?

```txt
time="2022-03-18T01:43:58+01:00" level=debug msg="Server initialized: /data"
time="2022-03-18T01:43:58+01:00" level=debug msg="Adding wireguard device: wg0"
time="2022-03-18T01:43:58+01:00" level=debug msg="Adding ip address to wireguard device: 172.31.255.0/24"
time="2022-03-18T01:43:58+01:00" level=debug msg="Bringing up wireguard device: wg0"
time="2022-03-18T01:43:58+01:00" level=debug msg="Adding NAT / IP masquerading using nftables"
time="2022-03-18T01:43:58+01:00" level=debug msg="Flushing nftable rulesets"
time="2022-03-18T01:43:58+01:00" level=debug msg="Setting up nftable rules for ip masquerading"
time="2022-03-18T01:43:58+01:00" level=fatal msg="Receive: netlink receive: operation not supported"
```

### Additional information

This is where it fails in the upstream project:
https://github.com/EmbarkStudios/wg-ui/blob/1397fdb0e10b3b35430389b0e3bbc8595a619828/server.go#L192-L226

agners, 2022-04-01:
@theseal the latest dev builds have nftables enabled. Can you give this a try? Either by enabling dev channel (`ha su options --channel=dev`, `ha su reload` and `ha os update` or download as a new image from here:

https://os-builds.home-assistant.io/8.0.dev20220331/

theseal, 2022-04-01:
> @theseal the latest dev builds have nftables enabled. Can you give this a try? Either by enabling dev channel (`ha su options --channel=dev`, `ha su reload` and `ha os update` or download as a new image from here:
> 
> 
> 
> https://os-builds.home-assistant.io/8.0.dev20220331/

I'm sorry but I don't have any machines to run it on. 

Maybe @DFS-90 who created the related issue on the add-on could give it a try.

DFS-90, 2022-04-01:
Just gave it a try - here is the result:

`[s6-init] making user provided files available at /var/run/s6/etc...exited 0.
[s6-init] ensuring user provided files have correct perms...exited 0.
[fix-attrs.d] applying ownership & permissions fixes...
[fix-attrs.d] done.
[cont-init.d] executing container initialization scripts...
[cont-init.d] 00-banner.sh: executing... 
-----------------------------------------------------------
 Add-on: Wireguard UI
 Web UI for provisioning and managing device keys and configurations for Wireguard.
-----------------------------------------------------------
 Add-on version: 1.0.4
 You are running the latest version of this add-on.
 System: Home Assistant OS 8.0.dev20220331  (amd64 / qemux86-64)
 Home Assistant Core: 2022.3.8
 Home Assistant Supervisor: 2022.03.dev3104
-----------------------------------------------------------
 Please, share the above information when looking for help
 or support in, e.g., GitHub, forums or the Discord chat.
-----------------------------------------------------------
[cont-init.d] 00-banner.sh: exited 0.
[cont-init.d] 01-log-level.sh: executing... 
[cont-init.d] 01-log-level.sh: exited 0.
[cont-init.d] nginx.sh: executing... 
[cont-init.d] nginx.sh: exited 0.
[cont-init.d] done.
[services.d] starting services
[services.d] done.
[19:06:43] INFO: Starting NGinx...
[19:06:43] INFO: Starting Wireguard UI...
time="2022-04-01T19:06:44+02:00" level=info msg=Starting
time="2022-04-01T19:06:44+02:00" level=debug msg="ipAddr: 172.31.255.0  ipNet: 172.31.255.0/24"
time="2022-04-01T19:06:44+02:00" level=debug msg="Read server config from file: /data/config.json"
time="2022-04-01T19:06:44+02:00" level=debug msg="Configuration loaded with public key: <key removed>"
[favicon.png index.html main.css main.css.map main.js main.js.map vendors~main.1.js vendors~main.1.js.map vendors~main.css vendors~main.css.map] <nil>
time="2022-04-01T19:06:44+02:00" level=debug msg="Server initialized: /data"
time="2022-04-01T19:06:44+02:00" level=debug msg="Adding wireguard device: wg0"
time="2022-04-01T19:06:44+02:00" level=debug msg="Adding ip address to wireguard device: 172.31.255.0/24"
time="2022-04-01T19:06:44+02:00" level=debug msg="Bringing up wireguard device: wg0"
time="2022-04-01T19:06:44+02:00" level=debug msg="Adding NAT / IP masquerading using nftables"
time="2022-04-01T19:06:44+02:00" level=debug msg="Flushing nftable rulesets"
time="2022-04-01T19:06:44+02:00" level=debug msg="Setting up nftable rules for ip masquerading"
time="2022-04-01T19:06:44+02:00" level=debug msg="Reconfiguring wireguard interface wg0"
time="2022-04-01T19:06:44+02:00" level=debug msg="Adding wireguard private key"
time="2022-04-01T19:06:44+02:00" level=debug msg="Getting current Wireguard config"
time="2022-04-01T19:06:44+02:00" level=debug msg="Serving static assets embedded in binary"
time="2022-04-01T19:06:44+02:00" level=info msg="Starting server" listenAddr="127.0.0.1:8080"`

Guess it works - will give it a more detailed test in the following days!
**Thanks a lot @theseal @agners ! :-)**

agners, 2022-04-02:
Perfect, so this will be addressed with HAOS 8.0. In case something else is needed reopen or create a new issue.

closed 2022-04-02 by agners