home-assistant/operating-system #1814: Raspberry Pi 3 with Home Assistant crashes after couple of days.

gregersn, 2022-03-24:
### Describe the issue you are experiencing

The system is running fine for a day or two, then it seems to hang. The network activity lamps are still blinking, but the system is unresponsive. SSH access is also unresponsive.

### What operating system image do you use?

rpi3-64 (Raspberry Pi 3 64-bit OS)

### What version of Home Assistant Operating System is installed?

Home Assistant OS 7.5

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

Boot system
Wait for a day or two

### Anything in the Supervisor logs that might be useful for us?

```txt
22-03-24 22:07:28 INFO (MainThread) [supervisor.hardware.monitor] Started Supervisor hardware monitor
22-03-24 22:07:28 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface io.hass.os
22-03-24 22:07:28 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.systemd1
22-03-24 22:07:28 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.login1
22-03-24 22:07:29 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.hostname1
22-03-24 22:07:29 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.timedate1
22-03-24 22:07:30 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.NetworkManager
22-03-24 22:07:30 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface de.pengutronix.rauc
22-03-24 22:07:30 INFO (MainThread) [supervisor.dbus.manager] Load dbus interface org.freedesktop.resolve1
22-03-24 22:07:30 INFO (MainThread) [supervisor.host.info] Updating local host information
22-03-24 22:07:30 INFO (MainThread) [supervisor.host.services] Updating service information
22-03-24 22:07:30 INFO (MainThread) [supervisor.host.network] Updating local network information
22-03-24 22:07:31 INFO (MainThread) [supervisor.host.sound] Updating PulseAudio information
22-03-24 22:07:31 INFO (MainThread) [supervisor.host.manager] Host information reload completed
22-03-24 22:07:31 INFO (MainThread) [supervisor.host.apparmor] Loading AppArmor Profiles: {'hassio-supervisor'}
22-03-24 22:07:31 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-cli with version 2021.12.0
22-03-24 22:07:31 INFO (MainThread) [supervisor.plugins.cli] Starting CLI plugin
22-03-24 22:07:31 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_cli application
22-03-24 22:07:34 INFO (SyncWorker_0) [supervisor.docker.cli] Starting CLI ghcr.io/home-assistant/aarch64-hassio-cli with version 2021.12.0 - 172.30.32.5
22-03-24 22:07:34 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-dns with version 2021.06.0
22-03-24 22:07:34 INFO (MainThread) [supervisor.plugins.dns] Starting CoreDNS plugin
22-03-24 22:07:34 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_dns application
22-03-24 22:07:36 INFO (SyncWorker_1) [supervisor.docker.dns] Starting DNS ghcr.io/home-assistant/aarch64-hassio-dns with version 2021.06.0 - 172.30.32.3
22-03-24 22:07:36 INFO (MainThread) [supervisor.plugins.dns] Updated /etc/resolv.conf
22-03-24 22:07:36 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-audio with version 2021.07.0
22-03-24 22:07:37 INFO (MainThread) [supervisor.plugins.audio] Starting Audio plugin
22-03-24 22:07:37 INFO (SyncWorker_0) [supervisor.docker.interface] Cleaning hassio_audio application
22-03-24 22:07:39 INFO (SyncWorker_0) [supervisor.docker.audio] Starting Audio ghcr.io/home-assistant/aarch64-hassio-audio with version 2021.07.0 - 172.30.32.4
22-03-24 22:07:39 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-observer with version 2021.10.0
22-03-24 22:07:39 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0
22-03-24 22:07:40 INFO (MainThread) [supervisor.plugins.multicast] Starting Multicast plugin
22-03-24 22:07:40 INFO (SyncWorker_1) [supervisor.docker.interface] Cleaning hassio_multicast application
22-03-24 22:07:41 INFO (SyncWorker_1) [supervisor.docker.multicast] Starting Multicast ghcr.io/home-assistant/aarch64-hassio-multicast with version 2022.02.0 - Host
22-03-24 22:07:42 INFO (MainThread) [supervisor.updater] Fetching update data from https://version.home-assistant.io/stable.json
22-03-24 22:07:42 INFO (MainThread) [supervisor.homeassistant.secrets] Loaded 1 Home Assistant secrets
22-03-24 22:07:42 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to ghcr.io/home-assistant/raspberrypi3-64-homeassistant with version 2022.3.6
22-03-24 22:07:42 INFO (MainThread) [supervisor.os.manager] Detect Home Assistant Operating System 7.5 / BootSlot B
22-03-24 22:07:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/a0d7b954 repository
22-03-24 22:07:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/git/45df7312 repository
22-03-24 22:07:44 INFO (MainThread) [supervisor.store.git] Loading add-on /data/addons/core repository
22-03-24 22:07:47 INFO (MainThread) [supervisor.store] Loading add-ons from store: 64 all - 64 new - 0 remove
22-03-24 22:07:47 INFO (MainThread) [supervisor.addons] Found 4 installed add-ons
22-03-24 22:07:47 INFO (SyncWorker_3) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-mosquitto with version 6.0.1
22-03-24 22:07:47 INFO (SyncWorker_1) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-configurator with version 5.3.3
22-03-24 22:07:47 INFO (SyncWorker_0) [supervisor.docker.interface] Attaching to zigbee2mqtt/zigbee2mqtt-aarch64 with version 1.24.0-1
22-03-24 22:07:47 INFO (SyncWorker_2) [supervisor.docker.interface] Attaching to homeassistant/aarch64-addon-ssh with version 9.3.0
[... 715 lines cut ...]
docker_version | 20.10.9
disk_total | 29.1 GB
disk_used | 5.7 GB
healthy | true
supported | true
board | rpi3-64
supervisor_api | ok
version_api | ok
installed_addons | Mosquitto broker (6.0.1), File editor (5.3.3), Zigbee2mqtt (1.24.0-1), Terminal & SSH (9.3.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 0
mode | auto-gen

</details>

### Additional information

There is a Sonof Zigbee adapter connected on the USB. 
Zigbee2Mqtt is running.

agners, 2022-03-30:
Did that same system work stable before? What version of the operating system have you been using before?

If the system did work stable before, its probably worth downgrading to see if the problem indeed disappears again. You can downgrade the operating system using

```
ha os update --version 7.x
```

gregersn, 2022-05-19:
After running for a while on 7.0 again, and observing the same issue, and after having gotten in some other updates, and now being back at 7.6 with seemingly better stability, my estimate is that it was not the OS. 
Closing this issue.

closed 2022-05-19 by gregersn