home-assistant/operating-system #1815: Raspberry Pi 4 crashes after some days with HASSOS 7.3

auanasgheps, 2022-03-28:
### Describe the issue you are experiencing

Hi, my Raspberry Pi 4 crashes after some days when running HASSOS 7.3 and above.

This looks like a come back of the infamous #1119.
Symptoms are exactly the same: the device runs fine for some days, then goes offline. It cannot be accessed via SSH, WebGUI or anything similar. Leds stop blinking. 
I haven't tried to connect a monitor, but since a while HASSOS doesn't output logs anymore on screen.

HASSOS 7.2 runs perfectly.

By looking at the [7.3 release](https://github.com/home-assistant/operating-system/releases/tag/7.3) the Raspberry Pi4 firmware has been updated: what a coincidence! ;)

I have a Raspberry Pi 4 4GB, running off an SSD with a RaspbeeII connected.

### What operating system image do you use?

rpi4-64  (Raspberry Pi 4/400 64-bit OS)

### What version of Home Assistant Operating System is installed?

7.3

### Did you upgrade the Operating System.

Yes

### Steps to reproduce the issue

1. Upgrade to 7.3
2. Wait for some days
3. System crashes
...

### Anything in the Supervisor logs that might be useful for us?

```txt
I haven't collected any logs yet. Current supervisor logs look good.
```

### Anything in the Host logs that might be useful for us?

```txt
Nope. It's the same as above.
```

### System Health information

## System Health

version | core-2022.3.7
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.9.9
os_name | Linux
os_version | 5.10.92-v8
arch | aarch64
timezone | Europe/Rome

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 4898
Installed Version | 1.23.0
Stage | running
Available Repositories | 1009
Downloaded Repositories | 17

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 7.3
-- | --
update_channel | stable
supervisor_version | supervisor-2022.03.5
docker_version | 20.10.9
disk_total | 109.3 GB
disk_used | 8.4 GB
healthy | true
supported | true
board | rpi4-64
supervisor_api | ok
version_api | ok
installed_addons | Home Assistant Google Drive Backup (0.106.2), SSH & Web Terminal (10.1.1), Studio Code Server (4.2.0), File editor (5.3.3), Check Home Assistant configuration (3.10.0), Let's Encrypt (4.12.0), NGINX Home Assistant SSL proxy (3.1.1), Vaultwarden (Bitwarden) (0.16.0), Mosquitto broker (6.0.1), Zigbee2mqtt (1.24.0-1), deCONZ (6.12.0)

</details>

<details><summary>Lovelace</summary>

dashboards | 1
-- | --
resources | 8
views | 5
mode | storage

</details>

<details><summary>Spotify</summary>

api_endpoint_reachable | ok
-- | --

</details>

<details><summary>Xiaomi Miot Auto</summary>

component_version | 0.5.4
-- | --
can_reach_server | ok
can_reach_spec | ok
logged_accounts | 0
total_devices | 0

</details>

agners, 2022-03-28:
:cry: Maybe we have to reactivate that issue then? Let's check a few things on our end: Do others which did see the problem back then see it again as well? I will release 7.6 on the beta channel soon with their latest firmware release. You should definitely also test that release.

bschatzow, 2022-03-28:
I have used every update since the January fix.i have not experienced the freeze issue which I had on all updates after 5.4.. Currently on 7.5.

bschatzow, 2022-03-28:
@auanasgheps As you were instrumental in getting 1119 fixed and have a good understanding of the issue, is there any information I can get for you on my system to see why mine is still working and yours is not?

jcsouthworth, 2022-03-28:
I have been running current versions since the #1119 fix and I haven't had any crashes. I don't think I rebooted at all on version 7.4 - over 4 weeks of up time. Same on 7.5 - over two weeks and counting. it has been so nice. I'm sorry you're experiencing it again! I'll stay in the loop to help how I can.

auanasgheps, 2022-03-28:
Thank you guys.

So I think the plan for now is test 7.6 Beta with the new firmware, I can see is available.

If the problem is not solved, ask the Raspberry team what is going on. Might be a subtle change noticed only by some systems.

auanasgheps, 2022-04-07:
After ten days I can happily confirm that HASSOS 7.6 is stable on my Pi4! 🎉

Thank you :)

closed 2022-04-07 by auanasgheps